MarlinFirmware/Marlin #18051: I am not sure if i put in the data correctly..for my ender3 i am compiling it from a default marlin firmware... need to change the offset

Dp4610yz, 2020-05-20:
//file Configuration.h Release:tags/2.0.5.3
//section machine
#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2
#define CUSTOM_MACHINE_NAME "Ender3"
#define X_BED_SIZE 235
#define Y_BED_SIZE 235
#define Z_MAX_POS 250
//section extruder
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define SINGLENOZZLE 
#define HOTEND_OFFSET_X { 0.0, 20.00 }
#define HOTEND_OFFSET_Y { 0.0, 5.00 }
//section temperature
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_2_LABEL "PETG"
#define PREHEAT_2_TEMP_BED 80
//section homing
#define X_DRIVER_TYPE TMC2209
#define Y_DRIVER_TYPE TMC2209
#define Z_DRIVER_TYPE TMC2209
#define E0_DRIVER_TYPE TMC2209
#define Z_HOMING_HEIGHT 2
#define Z_AFTER_HOMING 4
#define MANUAL_X_HOME_POS 0
#define Z_SAFE_HOMING 
//section probes
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
#define BLTOUCH 
#define NOZZLE_TO_PROBE_OFFSET { -51,-20, 0 }
//section calibrate
#define AUTO_BED_LEVELING_UBL 
#define GRID_MAX_POINTS_X.0 5

Dp4610yz, 2020-05-20:
can u all please comment...if there is anything wrong..

ellensp, 2020-05-20:
Anything wrong, like posing support requests in a area that is for Marlin Bugs and feature requests, when other areas are designed for support and even listed for you? 
Please read https://github.com/MarlinFirmware/Marlin/issues/14694 and use one of those services

ruggb, 2020-05-23:
#define HOTEND_OFFSET_X { 0.0, 20.00 }
#define HOTEND_OFFSET_Y { 0.0, 5.00 }
not needed since you have a single hotend

closed 2020-06-21 by boelle