MarlinFirmware/Marlin #18052: [BUG] (Power Loss 2.0.5.3 no show recovery menu)

campy, 2020-05-20:
### Bug Description

Hello, i compile marlin 2.0.5.3 and i use Mini UPS connected to motherboard (SKR 1.3), so in this version eveything working fine except power loss feature

When i cut the power of 3d printer, printing from sd, nozzle raise 2mm filament retract, and marlin detect have a power loss, normal behavior, but when turn on again the machine now don´t show recovery menu to continue printing or cancel printing.

I think the problem is marlin don´t save last position file in SD

### My Configurations

```
#define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    #define BACKUP_POWER_SUPPLY       // Backup power / UPS to move the steppers on power loss
    #define POWER_LOSS_ZRAISE       2 // (mm) Z axis raise on resume (on power loss with UPS)
    #define POWER_LOSS_PIN         P1_24 // Pin to detect power loss. Set to -1 to disable default pin on boards without module.
    #define POWER_LOSS_STATE     HIGH // State of pin indicating power loss
    #define POWER_LOSS_PULL           // Set pullup / pulldown as appropriate
    #define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    #define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.
```

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data
  #endif

Configuration files: https://gofile.io/d/9X8dRx

### Steps to Reproduce

**Expected behavior:**
Turn on again the machine and recovery menu appears

**Actual behavior:** 
When turn on the machine is like no power loss before

boelle, 2020-06-20:
@campy still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle