MarlinFirmware/Marlin #18053: [BUG] SKR 1.4 (non turbo) BLtouch not working

jamesrooke54, 2020-05-20:
### Bug Description

ok, im trying to get my BLtouch working on the SKR 1.4, the 1.4 has dedicated pins for the BLtouch so im using those, the BLtouch self tests when the machine starts and im able to deploy the probe and retract. but when I home all the X and Y move to 0 then the Z moves up and X and Y move to maximum

if I go into the auto bed level menu then z moves up by maybe 10 X and Y move back to 0 the bltouch pin deploys and z starts to move down but if I trigger the pin it doesn't stop and I have to turn off the machine to prevent a crash

I do not have z endstops install only the bltouch, I thought i would be able to figure this out but im just not experienced enough to know where to start

### My Configurations
config file attached
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4658055/Configuration.zip)

unixb0y, 2020-05-20:
I have a kind of related issue, but in my case, when trying to use the pins that are specifically put there for the BLTouch, leveling works but not Z homing. When leveling, the printer moves to the leveling position (center of the bed) and then just doesn't deploy the BLTouch and just says that that's Z=0.

That's with this configuration:
```
//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
//#define Z_MIN_PROBE_PIN 32 // Pin 32 is the RAMPS default
#define BLTOUCH
#define AUTO_BED_LEVELING_BILINEAR
```

If I try to use the Z_MIN_ENDSTOP pin instead, the BLTouch _does_ deploy, but when I manually trigger it with my finger, the Z axis just continues going down until I cut the power.

jamesrooke54, 2020-05-21:
I already have USE_PROBE_FOR_Z_HOMING enabled so that doesn't solve my problem im afraid

paukstelis, 2020-05-21:
This has been reported quite a few times. The simplest solution is to define `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and then change the Z endstop pin in the SKR pin defintions.

https://github.com/MarlinFirmware/Marlin/issues/16971

unixb0y, 2020-05-21:
> This has been reported quite a few times. The simplest solution is to define `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and then change the Z endstop pin in the SKR pin defintions.
> 
> #16971

TBH with the latest version of bugfix-2.x there is this feature explicitly for that reason (`USE_PROBE_FOR_Z_HOMING`), I read through all the issues and read the code of the PR that introduced the feature. 

I'll now look at @jamesrooke54's config. If I don't find the issues, I could send him my config, it's working now under the latest version of Marlin with TMC2130 drivers, the RepRapDiscount LCD and SD card also works.

unixb0y, 2020-05-21:
Okay, it looks like you really don't have the latest version @jamesrooke54! You have something from the BTT repo. I'd highly suggest you download / clone this repo instead, on the bugfix-2.0.x branch, port your config over and try again! 

You did add the `USE_PROBE_FOR_Z_HOMING` config item, but that doesn't help if the project doesn't know what to do with it. You basically don't have the new code that handles this define.

jamesrooke54, 2020-05-21:
Ok I'll try again with the latest marlin and see how I get on, apologise I didn't know the BTT one was that far behind your version

unixb0y, 2020-05-21:
> Ok I'll try again with the latest marlin and see how I get on, apologise I didn't know the BTT one was that far behind your version

Great, good luck! No problem 👍

Matdos, 2020-06-12:
Hi,

my 3DTouch is working.

I have modified

Configuration .h

//#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_MIN_PROBE_PIN P0_10 // Pin 32 is the RAMPS default

And this is the special Problem:

you need to go to Folder  Marlin - scr - inc - Conditionals_LCD.h

in Line 586  

original           #if Z_HOME_DIR < 0 && !HAS_CUSTOM_PROBE_PIN

its modified        #if Z_HOME_DIR < 0 //&& !HAS_CUSTOM_PROBE_PIN

boelle, 2020-06-21:
@jamesrooke54 still an issue?

jamesrooke54, 2020-06-21:
To be honest I've just removed it, the fix is to use the z stop plus and  it use the bltouch port but doing that makes me itch lol if there is a port for it then you should be able to use it

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle