MarlinFirmware/Marlin #18054: Suddenly restart during homeing - BLTouch &  ENDSTOP_INTERRUPTS_FEATURE

JozsefKatus, 2020-05-20:
Hallo guys,

I have a long week behind me and I want to tell a bug. I´m setting up my new own-design cartesian printer with following freatures:
-"Prusa-type" mechanic with dual Z-spindle (TR8x8 lead screws with anti-backlash nuts)
-Arduino ATmega2560 main board
-RAMPS 1.6 Plus
-5xDRV8825 steppers in 32 microstep mode
-4xNEMA17 49Ncm/2A and 1xNEMA17 13Ncm/0,9A, 4-wire steppers
-genuine BLTouch V3.0 sensor
-ATX PSU with PS_ON function (additional switched 5V)
-REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
-5xoptical endstops
-Dotbit Dual drive extruder
-external breakout for switching the heat bed
-travel: 200x200x310
-Marlin 2.0.5.3

About the issue:
I installed the printer with the settings according to the machine elements. Auto bed leveling and other features for BLTouch needed acc. to the datasheet.
By switching the printer on from the menu and homing the axes suddenly and randomly restarts, freeze the Arduino board. By freezing no response neithier to the reset switch on the RAMPS nor plugging into the USB of the computer.
Sometimes I get the axes homed, sometimes even more times after each other, but suddenly happens. Nearly before it the axes begins not to move (typically the Y (no mechanic failure or high friction, etc.)) or making some noise, moving slowly (Y and Z) as they would have thermal or insufficient power problem. The noise is not loud, but it feels sick.
I researched a lot about issues like this without success. I tried/checked followings:
-PSU: I added a switched 5V to avoid overloading 5VSB line with the BLTouch (300mA)
-reducing/disabling microstepping
-reducing speed/acceleration of the axes generally and by homing/probing too
-LCD/SD issue with restarting
-wiring loom
-increased heatsink on the drivers
-reducing the output current of the drivers
-Z_LATE option
-other RAMPS (1.6)
-new drivers
-reducing tension of the belts
-dividing the Z axis for two steppers
-increasing MINIMUM_STEPPER_PULSE and decreasing MAXIMUM_STEPPER_RATE
-etc.
Anything without success! Being not to long: any hardware issues kept out (checked with oscilloscope too). It tried Marlin 1.1.x-bugfix, because I use it on my old printer without any issues. By compiling I get the error message that Y_MIN_PIN, Y_MAX_PIN, X_MIN_PIN, X_MAX_PIN are not interrupt capable.
To mention: I don´t get any error message by compiling the 2.0.5.3. and ANTCLAB suggest to enable the interrupting feature for both firmware versions!
Making -at least- the end short: If I disable ENDSTOP_INTERRUPTS_FEATURE any problems seems gone!
I hope that I could help for other users/developer. If this issue already known, please let me know and I try to delete the topic.

Best regards,
József

ManuelMcLure, 2020-05-20:
``ENDSTOP_INTERRUPTS_FEATURE`` is almost never necessary. By default, end stops are only checked when homing, and in that case they are checked on every move so end stop interrupts do not provide any advantage.
The only case where ``ENDSTOP_INTERRUPTS_FEATURE`` makes any sense is if ``ENDSTOPS_ALWAYS_ON_DEFAULT`` or they were explicitly enabled with ``M120``.

JozsefKatus, 2020-05-20:
Thanks for your reply. This is an option which is suggested by ANTCLABS (see BLTouch manual). I think it shouldn't generate such a phenomenon or it shouldn´t be suggested to enable it by the sensor manufacturer.

ManuelMcLure, 2020-05-20:
If ANTCLABS is recommending turning it on they're wrong. There is absolutely no reason for this option on most setups.
If they are concerned that the short pulse that is generated by the BLTouch might be missed, a better option is to enable ``BLTOUCH_FORCE_SW_MODE`` instead.

GMagician, 2020-05-20:
@ManuelMcLure I think a little difference exists. Now endstops are checked ever 1ms when not using interrupts, checked in temperature isr, all depends on speed. Originally check was made into stepper irs, but not now

ManuelMcLure, 2020-05-20:
@GMagician I was unaware of that change - thanks for setting me straight. I do see that ``endstops.update()`` is called in ``Stepper::block_phase_isr()`` but that isn't every step like I was thinking.

GMagician, 2020-05-20:
Stepper::block_phase_isr() ?!?!? then something changed again...recently I think

GMagician, 2020-05-21:
@ManuelMcLure ok...I rechecked code. Is not changed. The call you see is called when movement queue is empty. Is used to prevent start any move when endstops are pressed. Once the "home" movement is inside queue, only temperature isr will stop

GMagician, 2020-05-21:
@JozsefKatus I read your post and one thing sound a little bit weird...

> By freezing no response neithier to the reset switch on the RAMPS nor plugging into the USB of the computer.

AFAIK reset button on ramps is a physical reset and is almost impossible it doesn't reset the board

Also 32 steps on a mega2560 is a little bit "too much" if you want to use high speeds. I have your same stepstick cnf and I had to lower down to 80/90mm/s otherwise... missing steps or even board reset.

JozsefKatus, 2020-05-21:
Hi guys,

first of all many thanks taking care about this topic.
I double checked that the issue goes with disabling only this feature and it´s confirmed.
However freezing is really rare, most of the time it restarts only. I changed in between the Arduino board too and with the new one I had only one time freezing. I´m not sure that the old one was a genuine.
If I find some time I try to make a video about freezing if you are interested.
Thank you for the mention with microstepping. I use this config with my old printer -with reduced print speed due to the worn mechanic- without any issues. Lost steps I experienced by overheating the Z stepper and it has been solve by reducing the phase current.
I´ll pay attention to that next time by the first test prints and change it if needed.

Best regards,
József

GMagician, 2020-05-21:
Freezing when interrupts are on may also be related to electrical noise. If you generate a lot of io changes in a short time, you may "hang" the system.
To check it you may enable interrupts, disconnect endstops (even bltouch) and start homing...BE READY TO SWITCH OFF BEFORE CRASHING

boelle, 2020-06-20:
@JozsefKatus still an issue?

JozsefKatus, 2020-06-21:
Hallo,

currently I´m living with the work-around I described in my first post.
Many thanks!

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle