MarlinFirmware/Marlin #18056: Not blinking need homing warning when using "HOME_AFTER_DEACTIVATE"

BugsonPL, 2020-05-20:
https://github.com/MarlinFirmware/Marlin/blob/e4d8336175ee09b28d23220f446d07451ca3de43/Marlin/src/lcd/dogm/status_screen_DOGM.cpp#L323

I think the only condition should be (DISABLE_REDUCED_ACCURACY_WARNING)

GMagician, 2020-05-21:
Theoretically when `HOME_AFTER_DEACTIVATE` is enabled you are forced to home axes when stepper are disabled, then blinking to warning position is not accurate may be not so important...

BugsonPL, 2020-05-21:
After power up you also need to home but it blinks :)
Just it would be more consistent and useful- you look ant screen and you know right a way you need to home.

boelle, 2020-06-21:
@BugsonPL still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle