MarlinFirmware/Marlin #18057: Controll -> Temperature -> Fan Speed 2

AnDrEaZaNoLeTtI, 2020-05-20:
Controll -> Temperature -> Fan Speed 2
### Bug Description

Dear all, 

If I change the speed of fan 2 in the temperature submenu, the value is set to fan 3.

thanks to all

### My Configurations

BOARD BTT SKR PRO V1.1
ZONESTAR LCD
[image]
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4658832/Configuration.zip)

boelle, 2020-06-20:
@AnDrEaZaNoLeTtI still an issue?

AnDrEaZaNoLeTtI, 2020-06-21:
Yes

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

AnDrEaZaNoLeTtI, 2020-06-24:
I download yesterday last version on "bugfix-2.0.x branch" and i see that the problem still remain

but i see that same issue is already opened [https://github.com/MarlinFirmware/Marlin/issues/17570](url)

sorry.
i close that one....

closed 2020-06-24 by AnDrEaZaNoLeTtI