MarlinFirmware/Marlin #18058: [GLITCH] Progress bar 1 pixel a tad too low after commit

Lord-Quake, 2020-05-20:
[image]

After a certain commit the progress bar fell by 1 pixel :-( 

Lets give it a hand and move it up to where it rightly belongs :-)

Where: Marlin/src/lcd/dogm/status_screen_DOGM.cpp 

Change to: `#define PROGRESS_BAR_Y (EXTRAS_BASELINE + 1)`

thisiskeithb, 2020-05-21:
You should put a PR in with the fix 🙂

ellensp, 2020-05-21:
Yes it is an ideal simple change to learn to make PR's 
Though I feel this change is subjective... It doesn't interfere with the text below or above. 
I suspect your OCD is playing up..  It doesn't have to be on the same level as the bottom of the SD card icon.

Lord-Quake, 2020-05-21:
OK @thisiskeithb, you win :-) You got me to install github desktop and make my first pull request. Hope I did it correctly.

ellensp, 2020-05-21:
PR Looks good.

swilkens, 2020-05-21:
Possibly caused here?

https://github.com/MarlinFirmware/Marlin/issues/17940

closed 2020-05-23 by Lord-Quake