MarlinFirmware/Marlin #18061: [BUG] Unable to activate StealthChop

adams79, 2020-05-21:
On my E3 Mini board also enabling stealth chop

#define STEALTHCHOP_XY
#define STEALTHCHOP_Z

by issuing M569 seems that all motor stays in sreadCycle:

Recv: X driver mode:	spreadCycle
Recv: Y driver mode:	spreadCycle
Recv: Z driver mode:	spreadCycle
Recv: E driver mode:	spreadCycle

Additional after updating the firmware with those config I've a strange noise from extruder motor: a swish very similar to a train :)

I've tried enabling and disabling StealthChop but nothing changed in M569

[config.zip](https://github.com/MarlinFirmware/Marlin/files/4661272/config.zip)

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

connect to the board and issue M569

**Expected behavior:** 

I should see X Y and Z motor in StealthChop

**Actual behavior:** [What actually happens]

Driver stays in SpreadCycle mode

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

thisiskeithb, 2020-05-21:
Try moving the red SPREAD jumper ([located near the X driver](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V1.2/BTT%20SKR%20MINI%20E3%20V1.2top.pdf)) to the opposite position it’s in now.

adams79, 2020-05-21:
Ok seems that the best solution is to remove the jumper ta have consistent behavior with firmware settings. In same case the jumper will cause reversed behaviour to what specified in firmware config

closed 2020-05-21 by adams79