MarlinFirmware/Marlin #18064: [BUG] M851 ignores negative Z-Values

dcbo, 2020-05-21:
### Bug Description

When setting negative Z-Values with `M851` e.g.: `M851 z-5`
the response to `M851` is correct, but the printer acts like it was set to `0`

### My Configurations
- Marlin 2.0.5.3
- Ender 3 Pro
- SKR 1.4 TURBO
- Physical Z-Enstop removed
- BLTouch connectet to P0_10
- All ABL DISABLED (to isolate the error)

### Steps to Reproduce
1. Flash `firmware.bin`
2. `m502`
3. `m500`
4. `m851 z0`
5. `g28`
6. `m114` returns `Z:10.00` (At the End it moves to Z=10)
7. Measure Hight of Nozzle with calipper = **H0**
8. `m851 z-5`
9. `g28`
10. `m114` returns `Z:10.00`
11. Measure Hight of Nozzle 

**Expected behavior:** 
Hight of Nozzle should be **H0 - 5mm**

**Actual behavior:** 
Hight of Nozzle is identical to H0 from step 7

**To verify that `m851` works for positive values** 
12. `m851 z10`
13. `g28`
14. `m114` returns `Z:10.00`
15. Measure Hight of Nozzle should be H0 + 10mm
**This is correct, the Nozzle is at H0 + 10mm**

#### Additional Information
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4663717/Configuration.zip)

boelle, 2020-06-20:
@dcbo still an issue?

sjasonsmith, 2020-06-21:
@dcbo, I cannot reproduce this with the latest bugfix. When I use a negative offset, the nozzle parks in the same physical position, but the Z position returned by M114 (your step 10) has the offset value applies to it.

With a Z-5 offset, the M114 result after probing was Z15.00.

Somebody may very well have fixed this. Please try using the latest bugfix-2.0.x branch to see if the issue is resolved for you.

dcbo, 2020-06-21:
Will test it with Bugfix from Jun 9, 2020, give me some time.
I will generate new a complete new configuration manualy to eleminate possible errors in old Configuration.

sjasonsmith, 2020-06-21:
I tested it using the latest from just before I posted, so June 20th.

thinkyhead, 2020-06-21:
When testing, please enable `M114_DETAIL` and examine the output of `M114 D`.

dcbo, 2020-06-21:
Noting changed :-(

- got Marlin-bugfix-2.0.x today
- got Configurations-bugfix-2.0.x today
- extracted "\config\examples\Creality\Ender-3 Pro" from "Configurations-bugfix-2.0.x"
- manually merged my changes from Release 2.0.5 to configuration from previous step
- enabled M114_DETAIL in Configuration_adv.h

1. flashed
2. `m502`
3. `m500`
4. `m851 z0`
5. `g28`
6. `m114 d`

```
Recv: X:157.00 Y:127.00 Z:10.00 E:0.00 Count X:12560 Y:10160 Z:4000
Recv: 
Recv: Logical: X: 157.000 Y: 127.000 Z: 10.000
Recv: Raw:     X: 157.000 Y: 127.000 Z: 10.000
Recv: Stepper: X:12560 Y:10160 Z:4000 E:0
Recv: FromStp: X: 157.000 Y: 127.000 Z: 10.000 E: 0.000
Recv: Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.000
Recv: ok
```
7. Measure Hight of Nozzle with calipper = **H0**
8. `m851 z-5`
9. `g28 z`
10. `m114 d`
```
Recv: X:157.00 Y:127.00 Z:10.00 E:0.00 Count X:12560 Y:10160 Z:4000
Recv: 
Recv: Logical: X: 157.000 Y: 127.000 Z: 10.000
Recv: Raw:     X: 157.000 Y: 127.000 Z: 10.000
Recv: Stepper: X:12560 Y:10160 Z:4000 E:0
Recv: FromStp: X: 157.000 Y: 127.000 Z: 10.000 E: 0.000
Recv: Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.000
Recv: ok
```
11. Measure Hight of Nozzle

**Expected behavior:**
Hight of Nozzle should be H0 - 5mm

**Actual behavior:**
Hight of Nozzle is identical to H0 from step 7

**To verify that `m851` works for positive values**
12. `m851 z10`
13. `g28 z`
14. `m114 d`

```
Recv: X:157.00 Y:127.00 Z:10.00 E:0.00 Count X:12560 Y:10160 Z:4000
Recv: 
Recv: Logical: X: 157.000 Y: 127.000 Z: 10.000
Recv: Raw:     X: 157.000 Y: 127.000 Z: 10.000
Recv: Stepper: X:12560 Y:10160 Z:4000 E:0
Recv: FromStp: X: 157.000 Y: 127.000 Z: 10.000 E: 0.000
Recv: Diff:    X: 0.000 Y: 0.000 Z: 0.000 E: 0.000
Recv: ok
```

15. Measure Hight of Nozzle should be H0 + 10mm
**This is correct, the Nozzle is at H0 + 10mm**

In the Attachment the only files i changed 
- Configuration.h 
- Configuration_adv.h.
- platformio.ini

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4809601/configuration.zip)

I did not change anything else

dcbo, 2020-06-21:
For convenience I add DIFF-Files for the changes I made strating from 
"Configurations-bugfix-2.0.x\config\examples\Creality\Ender-3 Pro"

[configuration_diff.zip](https://github.com/MarlinFirmware/Marlin/files/4809628/configuration_diff.zip)

sjasonsmith, 2020-06-21:
I updated my firmware to use your configuration. I had to change a few things, since our hardware is not 100% identical. The main difference is that I have an SKR 1.3 in the printer, so I use `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` instead of `USE_PROBE_FOR_Z_HOMING`.

Now M114 always reports 10mm after homing, and the physical location is always adjusted for the offset.

M851 Z0: Measured Z = 12mm, M114 Z = 10mm
M851 Z5: Measured Z = 17mm, M114 Z = 10mm
M851 Z-5: Measured Z = 7mm, M114 Z = 10mm

I reworked my pins file so that I can disable `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. Everything is still working fine.

I see no problems. The only differences in our configuration files are now:
- MOTHERBOARD (SKR 1.3 vs. SKR 1.4 Turbo)
- Driver Models (2208 instead of 2130)
- Extruder configuration (steps/mm, AUTO_FAN, etc.)

sjasonsmith, 2020-06-21:
My most recent testing was done using bugfix-2.0.x, on top of commit 67db50c69950e96425005b5b165558323d41737f.

dcbo, 2020-06-21:
**Thank you for Your time in finding the Error.**

I have found the Option `DEBUG_LEVELING_FEATURE` and enabled it.
With `M111 S32` the Debugging is turned on I think I can help in finding the Error:

The Logfiles are in the attachment, here are the relevant lines.

When BLTouch has triggered, there the Z-Value is set correct in all three cases:
- M851 Z0  -> When triggeres Z=0
- M851 Z10 -> When triggeres Z=-10
- M851 Z-5 -> When triggeres Z=5

Than it should go to Z=10 Probe::move_z(10.00) but it results in 
- M851 Z0  -> do_blocking_move_to X157.00 Y127.00 Z10.00
- M851 Z10 -> do_blocking_move_to X157.00 Y127.00 Z10.00
- M851 Z-5 -> do_blocking_move_to X157.00 Y127.00 Z15.00 **(WTF)**

M851 Z0 
```
Recv: > probe.offset.z = 0.00
Recv:   current_position= X157.00 Y127.00 Z0.00 : > AFTER set_axis_is_at_home
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X157.00 Y127.00 Z10.00
```

M851 Z10 
```
Recv: > probe.offset.z = 10.00
Recv:   current_position= X157.00 Y127.00 Z-10.00 : > AFTER set_axis_is_at_home
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X157.00 Y127.00 Z10.00
```

M851 Z-5 
```
Recv: > probe.offset.z = -5.00
Recv:   current_position= X157.00 Y127.00 Z5.00 : > AFTER set_axis_is_at_home
Recv: Probe::move_z(10.00)
Recv: >>> do_blocking_move_to X157.00 Y127.00 Z15.00
```

Why the hell does a move_z(**10.00**) result in do_blocking_move_to X157.00 Y127.00 **Z15.00**

And that is just what I observe during G28, the Head moves to the same position as on M851 Z0 
after that he seems to reset the coordinates to Z=10 without any further movement.

Maybe this can help in finding the Error.

[M851 Z0.txt](https://github.com/MarlinFirmware/Marlin/files/4810307/M851.Z0.txt)
[M851 Z10.txt](https://github.com/MarlinFirmware/Marlin/files/4810308/M851.Z10.txt)
[M851 Z-5.txt](https://github.com/MarlinFirmware/Marlin/files/4810309/M851.Z-5.txt)

shitcreek, 2020-06-23:
**M851** is working as it is stated in the config.h file:
```
 * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
 */
#define Z_CLEARANCE_DEPLOY_PROBE    8 // Z Clearance for Deploy/Stow
```
You may want to read through these: 
https://github.com/MarlinFirmware/Marlin/issues/11046#issuecomment-397901009
https://github.com/MarlinFirmware/Marlin/issues/11046
https://github.com/MarlinFirmware/Marlin/issues/11299
https://github.com/MarlinFirmware/Marlin/issues/8479

Note: with leveling turned off, the machine will not go below the configured **Z_MIN_POS** value so the print head will remain in the same position for **M851 Z0** and **M851 Z-10**.

boelle, 2020-06-23:
@shitcreek so no bug here?

Desuuuu, 2020-07-06:
I've noticed the same issue with the bugfix-2.0.x branch (from July 4th) with a similar setup:

* SKR 1.4 Turbo
* TMC2209
* Z safe homing
* BLTouch wired to Z_MIN_PROBE_PIN
* `MULTIPLE_PROBING 2`
* `Z_AFTER_HOMING 10`
* `Z_CLEARANCE_DEPLOY_PROBE 10`

From what I've found, here's how it goes with my Z offset set to -2.4mm:

* Run G28.
* X/Y are homed then move to the safe homing position.
* The BLTouch probes the bed.
* After the 2nd probe, the Z axis is raised by 10mm (`Z_CLEARANCE_DEPLOY_PROBE`) and the BLTouch is stowed.
* The Z axis is now physically at 12.4mm from the bed and Marlin's position is correct at this point.
* The printer tries to lower the Z axis to 10mm (`Z_AFTER_HOMING`). The axis does not move at this point, but the current position in Marlin has changed to 10mm, negating the Z offset.

Here's the log from running G28Z (where the last move does not actually happen): 
[g28z.txt](https://github.com/MarlinFirmware/Marlin/files/4876634/g28z.txt)

closed 2020-08-11 by github-actions[bot]