MarlinFirmware/Marlin #18065: [BUG] Universal Bed Levelling Results in Null Values for

DanielDC88, 2020-05-21:
### Bug Description

When building a 4x4 cold mesh using UBL, the back row is probed but the data does not show up:
`Send: M420 V
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv:     ( 20,215)              (215,215)
Recv:         0       1       2       3
Recv:  3 |   .       .       .       .
Recv:    |
Recv:  2 | -0.200  -0.128  -0.192  -0.160
Recv:    |
Recv:  1 | -0.220  -0.223  -0.120  -0.230
Recv:    |
Recv:  0 |[-0.233] -0.225  -0.258  -0.118
Recv:         0       1       2       3
Recv:     ( 20, 20)              (215, 20)
Recv: 
Recv: Mesh is invalid
Recv: Storage slot: 0
Recv: echo:Invalid mesh.
Recv: echo:Bed Leveling OFF
Recv: echo:Fade Height OFF`

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4664005/Configuration_adv.zip)

### Steps to Reproduce

1. Build cold mesh using ubl
2. Send M420 V command
3. observe row of .

**Expected behavior:** back row has numerical values in it

**Actual behavior:** back row has .'s in it

#### Additional Information

Ender 3 using SKR mini 1.2 and bl touch clone

Lord-Quake, 2020-05-23:
Make sure UBL is testing all mesh points and not leaving the ones at a particular edge out.
This can happen if your
`#define MESH_INSET 20             // Set Mesh bounds as an inset region of the bed`
is not set correctly and the probe is unable to reach the edge.

Tannoo, 2020-05-26:
Have you tried `G29 P3` a few times to populate any blank points?

DanielDC88, 2020-05-26:
> I tested it with Marlin 2.0.5.3 and am not seeing your issue.
> 
> What is G29 T showing?
> 
> Make sure UBL is actually testing all mesh points and not leaving the ones at a particular edge out.
> This can happen if your
> 
> `#define MESH_INSET 20 // Set Mesh bounds as an inset region of the bed`
> 
> is not set correctly and the probe is unable to reach the edge.

This was the issue, my bad.

closed 2020-05-26 by DanielDC88