MarlinFirmware/Marlin #18066: Marlin video  - need feedback

Bob-the-Kuhn, 2020-05-21:
After a year hiatus I'm once again working on a video aimed at helping new Marlin users upload  firmware.

A lot has changed in the last year so I need to do some major changes to the video.  Before I do that I need feedback on some items:
- Is this video still something good to do?  I'm not aware of any videos that show how to use the **Auto Build** feature.
- The current Marlin web page says that PlatformIO will be dropping support for Atom.  I can't find any confirmation of that and it is still listed as one of the active options.  A link to the announcement would be appreciated.
- Should this video be based on Atom or VScode?  I personally prefer Atom for the build/upload but others like VScode.  Maybe one for each is needed.
- How detailed should the upload section be?  I found myself going into much more detail than I expected in order to provide enough info to be useful.
- How universal are the STM32 upload options?  My understanding is all STM32 controllers can be uploaded via the SD card but Auto Build's **PIO Upload** doesn't support it.  I also see that ST-Link is supported.  Do all STM32 controllers have a ST-Link port? On the boards that do have a port, does **PIO Upload** support ST-Link for every board?

I'd also like feedback on the current content.  

Here's a link to the video:  [Marlin_video 2020-05-21.mp4](https://www.dropbox.com/s/vmilcnp68zdctkc/Marlin_video%202020-05-21.mp4?dl=0)

Bob-the-Kuhn, 2020-05-21:
Another question.

Is there a better format to use than MP4?

I had originally started out using FLASH's FLV because I can set it up so that it pauses at important slides until the mouse is clicked.  Unfortunately FLASH is dead after 2020.

ManuelMcLure, 2020-05-21:
The following comment from someone on the PlatformIO team implies the deprecation: https://community.platformio.org/t/atom-installation-problem/4859

It's possible that this has changed since then.

EvilGremlin, 2020-05-21:
1. Existing Autobuild is kinda abandoned and broken, for now. IIRC, Scott said somewhere that he want to make new version.
2. PIO not dropping Atom per se, but it's not recommended IDE anymore due to it's many problems.
3. Yep, VScode is easiest compile button of them all. Arduino not worth mentioning anymore, i think. Or mention that devs don't really care about it, so don't expect it to work.
4. Whatever, just don't forget to point out where resulting binary is.
Encoding and stuff - just upload to youtube i think? So make H264 1920*1080 60fps

swilkens, 2020-05-21:
There was a while where PIO was using different build flags relative to the arduino IDE, resulting in larger binaries. I'm not sure if that is still the case, but might be relevant for 8-bit boards with small memory capacities.

EvilGremlin, 2020-05-21:
Must be looooong time ago, last year i've only seen mentions of pio binary being smaller.

thisiskeithb, 2020-05-21:
> Existing Autobuild is kinda abandoned and broken, for now.

Autobuild is not abandoned. It was updated [three days ago](https://github.com/MarlinFirmware/AutoBuildMarlin/commits/master) and is being actively worked on and improved so it can do more than just build firmware. Stay tuned 😉

EvilGremlin, 2020-05-21:
Wow, so Scott finally got his hands on it, good to know! But is that out in the wild yet? All i know is people unable to build anything with the version accessible from vscode.

thisiskeithb, 2020-05-21:
> But is that out in the wild yet? All i know is people unable to build anything with the version accessible from vscode.

Sure is and compiles fine (used it this morning).

Search VSCode’s Extensions for Auto Build Marlin. It has [24k installs and counting](https://marketplace.visualstudio.com/items?itemName=MarlinFirmware.auto-build).

EvilGremlin, 2020-05-21:
Gonna check it out, but just about a week ago there were bad reports about it...

Roxy-3D, 2020-05-21:
Bob,    Does it make sense to do a short video getting the build environment set up for 2 or 3 different processors.   And perhaps an Atom and VSCode flavor of each?   I'm not talking super in depth.  But maybe go here, download this, configure it this way, put the source code here, do a little more, do this and here is your image?  (With AVR being one of the processors!)

And what would be really good is a 'Developer Version' where the the tail end of each video shows exactly what to do to get the J-Tag debuggers going...

thisiskeithb, 2020-05-21:
It'll take me a bit to find the tweet, but Scott/Marlin's official stance is to use VSCode + PlatformIO. I'd skip anything with Atom since it causes too many questions when it doesn't work.

Bob-the-Kuhn, 2020-05-21:
I only use Atom with the Auto Build script.  I like Notepad ++ as the editor.

Time to play with VSCode.  Last I used it there were problems switching project folders.

@Roxy-3D - what J-TAG debuggers are you thinking of?  I have a J-Link, a real ST-Link, some ST-Link clones and several 8 bit debuggers.

thisiskeithb, 2020-05-21:
> On the boards that do have a port, does PIO Upload support ST-Link for every board?

I’ve seen at least one related PR, so someone will have to double check that start addresses are after bootloaders since we don’t want to accidentally overwrite them 🙂

Roxy-3D, 2020-05-21:
> Roxy-3D - what J-TAG debuggers are you thinking of? I have a J-Link, a real ST-Link, some ST-Link clones and several 8 bit debuggers.

I don't think it really matters as long as they are cheap and easy to get.   I have a knock-off J-Link so it would be nice to see a cheat sheet on getting that going.    BUT...   I have an LPC processor so hopefully that will be one of your videos with the tail end of it dedicated to J-Links.

ellensp, 2020-05-22:
LPC based boards are the ones that most users overwrite their bootloader by accident when they first use a J-LINK to upload...  the linker script LD1768.ld is part of the framework... so they dont think to change the address...

Bob-the-Kuhn, 2020-05-22:
> knock-off J-Link

Can you point me to where I can get one?

Bob-the-Kuhn, 2020-05-22:
@ellensp - can you elaborate on how the LPC1768 bootloader gets blown away?

I've only used the J-Link with Ozone and the ELF file.  Lots & lots of uploads with no bootloader issues.

sjasonsmith, 2020-05-22:
> @ellensp - can you elaborate on how the LPC1768 bootloader gets blown away?
> 
> I've only used the J-Link with Ozone and the ELF file. Lots & lots of uploads with no bootloader issues.

I can confirm that I blew away my bootloader the first time I used upload inside platformio after naively selected jlink as the upload tool.

You have to provide an offset for it so it know where to program, but the framework ignores the normal way to do it in pio. @p3p merged a change I submitted to allow specifying that in the pio file, but I don’t think the framework has released since then.

Roxy-3D, 2020-05-22:
>> knock-off J-Link

> Can you point me to where I can get one?

Under $15 with shipping.    I've been a little bit scared to try to use it because some people killed their boards.   But I REALLY would love to be able to set break points and look at variable values when debugging new features!

https://www.ebay.com/itm/V9-High-Speed-USB-ARM-JTAG-Emulator-Debugger-J-Link-V9-Emulator-Converter/264479270303?hash=item3d9431659f:m:m1bgRFvY_HTDH6gHcMy1ygQ

Your videos have the potential of enabling much better debugging capability and faster code development for Marlin.

Bob-the-Kuhn, 2020-05-22:
Lately I've been using the ST-Link portion of a Nucleo board as a J-Link.  Segger has a utility that will re-program a ST-Link to act a a J-Link.  The utility can also restore it to ST-Link functionality.

I had a nucleo board laying around that I had no use for.  I cut all the traces between the ST-Link and main portion of the nucleo and then ran the utility.

Bob-the-Kuhn, 2020-05-23:
The only free GUI based source line debugger I found was Segger's Ozone.  The trial/free version has a once daily license reminder.  It only works with J-Link modules.  

Anyone know of other reasonable cost debug software?

The lowest cost J-Link module can be obtained by using Segger's utility to reprogram an ST-Link V2.  The utility can also restore it to ST-Link function.  A ST-Link clone can not be reprogrammed.  The lowest cost ST-Link V2 I found were some $11 Nucleo boards at Digikey. Postage is $5 for one.

The next lowest cost J-Link module is a J-Link EDU Mini.  It's $21 from Digikey.  It's probably $5 postage for it.  The nice thing is it includes the JTAG/SWD cable needed for DUE.

---

Looks like the lowest cost way to get into hardware debugging are:

**STM32**:
- Segger's Ozone debug software 
- ST-Link V2 - $11 + $5 postage from Digikey.  Needs to be re-programmed as a J-Link using the Segger utility
- adapter cable - STM32 targets need just a 5 pin to 5 pin cable. 

 **DUE**
- Segger's Ozone debug software 
- J-Link EDU MINI - $21 + $5 postage from Digikey.  
- adapter cable - comes with J-Link EDU MINI

**LPC1768/9**
Setup for this one is difficult because I'm not aware of any that have the JTAG/SWD port available on a connector.  I ended up soldering directly to the pins on the CPU.  Not for the faint of heart.
- Segger's Ozone debug software 
- J-Link EDU MINI - $21 + $5 postage from Digikey.  
- adapter cable - $10 in parts

sjasonsmith, 2020-05-23:
@Bob-the-Kuhn Source-level debugging seems to work well inside VS Code with PlatformIO and an ST-Link adapter. No extra software needed that PIO doesn't pull down automatically.

I've had less success inside VS Code with LPC. I've been able to connect and step through a few lines with a J-Link, but it's prone to lose connection. The SKR 1.4 has SWD pins, but unfortunately has no Boot jumper to recover the board when you accidentally erase your bootloader!

Bob-the-Kuhn, 2020-05-24:
I'm getting a headache stumbling around the PlatformIO docs trying to figure out how to point the debugger at the source code and the compiled code.

I've done an ST-Link upload via the Auto Build Upload option.    As best I can tell the PlatformIO debugger wants to use the default ENV in platformio.ini rather than the one used in the upload.

HELP!

sjasonsmith, 2020-05-24:
> HELP!

I _probably_ had the default environment set to the SKR Pro environment. From there I think all I did is select 'Run->Start Debugging". It takes a couple minutes to get started, but after that is works really well!

GMagician, 2020-05-24:
Adafruit Grand Central M4 (SAMD51)
    Sagger J-Link EDU MINI
    adapter cable - comes with J-Link EDU MINI

And in answer to @Roxy-3D about danger to brick something, it litterally saved my board (I bricked it twice while developping the SEE manager) and I was able to reprogram senitive flash area with it before WD intervention

GMagician, 2020-05-24:
> I'm getting a headache stumbling around the PlatformIO docs trying to figure out how to point the debugger at the source code and the compiled code.

I simply selected start debug (F5) from ide, it opens a window with standard trace stop etc buttons. Sagger uses GDB api. I had to launch server because it wasn't found but I think thjis is my win10 installation issue (on an old sagger version it worked)

Bob-the-Kuhn, 2020-05-27:
Here's my first attempt at a VSCode based video.

Feedback is encouraged.

[Easy Marlin Upload - STM32 - VSCode & Auto Build Marlin.mp4](https://www.dropbox.com/s/gm3o7mplilpltc6/Easy%20Marlin%20Upload%20-%20STM32%20-%20VSCode%20%26%20Auto%20Build%20Marlin.mp4?dl=0)

---

Should there be a sound track with the video?

If yes, I'll need suggestions.

Bob-the-Kuhn, 2020-05-27:
I've been able to set breakpoints and single step source code.  What other features should I include?

I'm not sure if I like downloading a debug image.  Keeping execution in the order of the source file is nice.  Not debugging the final executable is bothersome.

Bob-the-Kuhn, 2020-05-30:
Here's my first pass at a code debug overview.

[Marlin code debug overview - STM32, VSCode, PlatformIO Unified Debugger, Auto Build Marlin.mp4](https://www.dropbox.com/s/j9raefichod0okm/Marlin%20code%20debug%20overview%20-%20STM32%2C%20VSCode%2C%20PlatformIO%20Unified%20Debugger%2C%20Auto%20Build%20Marlin.mp4?dl=0)

The only major feature I couldn't find was setting a breakpoint on a memory access.  The STM32F4xx data sheet says the same cell that does the instruction breakpoints also does memory breakpoints.  I didn't find anything in the PlatformIO docs or online.

Roxy-3D, 2020-05-30:
> The only major feature I couldn't find was setting a breakpoint on a memory access. The STM32F4xx data sheet says the same cell that does the instruction breakpoints also does memory breakpoints. I didn't find anything in the PlatformIO docs or online.

Very Good!   For me, being able to set a break point at a particular line of code is the most important thing.    But being able to set a break point when accessing a certain memory location would be very nice.    We should be able to figure out how to get that capability if it can be done.

GMagician, 2020-05-31:
@Roxy-3D  j-link gdb server support this (breakpoint on memory read / write), don't know if PIO has some way to do that.
I saw PIO let you set a breakpoint if some "expression" is true.
Not sure j-link may stop when memory is written to a value (that would be interesting too).

Bob-the-Kuhn, 2020-05-31:
Here's the latest STM32 upload video.  

[Easy Marlin Upload - STM32 - VSCode & Auto Build Marlin 4](https://www.dropbox.com/s/8yt3gaiovsin9yu/Easy%20Marlin%20Upload%20-%20STM32%20-%20VSCode%20%26%20Auto%20Build%20Marlin%204.fbr?dl=0)

I think it's ready for external viewers.  If you agree then I'll try to put it on YouTube.

Bob-the-Kuhn, 2020-06-01:
I looked into setting  breakpoint on accessing a variable.  It's like a dog walking on it's hind legs ...

I went to the PlatformIO community Unified Debugger topic and asked if it had the capability.  The only answer so far pointed me a GDB debugging commends.

Using the Debug Console I was able to issue GDB commands.  The following resulted in a functional breakpoint on any access of the beginning of the TransmitQueue.
```
info address TransmitQueue
  Symbol "TransmitQueue" is static storage at address 0x20001abc.
  {"token":1069,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
x 0x20001abc
  0x20001abc <TransmitQueue>:	0x2e0a303a
  {"token":1541,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
awatch *0x20001abc
  Hardware access (read/write) watchpoint 1: *0x20001abc
  {"token":2485,"outOfBandRecord":[],"resultRecords":{"resultClass":"done","results":[]}}
```

The **0x2e0a303a** agreed with what the global variable window showed for the first four elements.

The Unified Debugger didn't show the new breakpoint.

Bob-the-Kuhn, 2020-06-02:
Here's the latest debug video.

[Marlin code debug overview - STM32, VSCode, PlatformIO Unified Debugger, Auto Build Marlin](https://www.dropbox.com/s/j9raefichod0okm/Marlin%20code%20debug%20overview%20-%20STM32%2C%20VSCode%2C%20PlatformIO%20Unified%20Debugger%2C%20Auto%20Build%20Marlin.mp4?dl=0)

I think it's ready for external viewers. If you agree then I'll try to put it on YouTube.

Bob-the-Kuhn, 2020-06-02:
I've decided it makes sense to create one upload video covering both the STM32 boards and the LPC176x boards.

Let me know what you think.

[Easy Marlin Upload - LPC176x & STM32 - VSCode & Auto Build Marlin](https://www.dropbox.com/s/14mbkiyhf8m9mvq/Easy%20Marlin%20Upload%20-%20LPC176x%20%26%20STM32%20-%20VSCode%20%26%20Auto%20Build%20Marlin%204.mp4?dl=0)

GMagician, 2020-06-02:
good job

Roxy-3D, 2020-06-02:
> I went to the PlatformIO community Unified Debugger topic and asked if it had the capability. The only answer so far pointed me a GDB debugging commends.

It would be nice to have that capability!   But being able to set break points at a particular C++ statement is good enough.      

I did some poking around...   I think we have to have the DWT (Data WatchPoint & Trace) capability to do this.   I haven't gone deep enough to figure out if the J-Link's have that capability.   But if yoiu jump ahead to page 10 of this document...   You can see how the hardware component of it is setup:

https://www.st.com/content/ccc/resource/training/technical/product_training/group0/c9/b9/a2/df/3e/80/45/59/STM32G4-System-Debug_DBG/files/STM32G4-System-Debug_DBG.pdf/_jcr_content/translations/en.STM32G4-System-Debug_DBG.pdf

PS.  Thanks for all the effort Bob!!!!

thinkyhead, 2020-06-02:
Here’s the message from Ivan Kravets Sept 2018. So for the last 15 months I have been passing on this information to users in the various Marlin forums I manage. https://github.com/MarlinFirmware/MarlinDocumentation/issues/176

thinkyhead, 2020-06-02:
> I'm not aware of any videos that show how to use the Auto Build feature.

Chris Riley (“Chris’s Basement”) did an introductory video about “Auto Build Marlin” a few weeks ago and has been using it in his videos ever since.

thinkyhead, 2020-06-02:
> I personally prefer Atom for the build/upload but others like VScode.

You may want to mention Atom, but I will be telling people to ignore that and use VSCode. Atom is not going to be supported unless there is a conjoined effort to make Auto Build Marlin –and the configuration tool that I’m working on within it– work in the same way across both editors.

thinkyhead, 2020-06-02:
> Let me know what you think.

It's good! but a couple of minor issues. Comments posted at Dropbox.

Bob-the-Kuhn, 2020-06-03:
@thinkyhead  - thanks for reviewing it.

> 6:49 — Note that if you click on the folder icon next to "Last Build" it should open the build folder. No need to hunt for it.

Nothing happened when I clicked on it or when I right clicked on it.

This is a nice feature.  I'll switch to it when it's working again.

> 7:36 — I see your note that not all bootloaders will rename firmware.bin to firmware.cur. For example the Creality 4.x board leaves the file unchanged. When you want to upload new firmware you **must use a different name** from the last time.

Are you saying that manually renaming **fimware.bin** to **FIRMWARE.CUR** won't work for the Creality 4.x boards?

> I can publish it on that channel. I have some good microphones and I don't mind adding narration, if needed.

I'd be honored if you would publish it.

Please add narration if you think it'll be a plus.   

I tried a couple of different mics but returned them because I didn't like the results.  

---

I'm using FlashBack Pro 5 to record the screen and edit the video.  What are you using?

Should the editable FlashBack file be made available in case someone wants to modify it?

thinkyhead, 2020-06-04:
> Nothing happened when I clicked on it or when I right clicked on it.

Well, I never did test it myself on Windows. It opens a hidden Terminal and uses shell commands to open the folder (`cd path`, `open .`, `exit`). If you open a terminal inside of VSCode, what commands can you use from the command line to open a folder in Explorer? Here's what Auto-Build is doing…

```js
//
// Use a native shell command to open the build folder
//
function reveal_build(env) {
  var aterm = vw.createTerminal({ name:'reveal', env:process.env });
  const relpath = path.join('.', '.pio', 'build', env);
  if (process.platform == 'win32') {
    command_with_ping(aterm, 'cd ' + relpath, false);
    command_with_ping(aterm, 'open .', false);
    command_with_ping(aterm, 'exit', false);
  }
  else
    command_with_ping(aterm
      , '`which xdg-open open other-open | grep -v found | head -n1` ' + relpath + ' ; exit 0'
      , false
    );
}
```

> Are you saying that manually renaming fimware.bin to FIRMWARE.CUR won't work for the Creality 4.x boards?

Your existing blurb is generally right. The way this particular board works is that it installs the first file it finds ending in ".bin" (unless it installed that file last time). There's no need to delete or rename the file, since the bootloader remembers which file it last flashed, probably using the file's unique id and not its name.

> I'm using FlashBack Pro 5 to record the screen and edit the video. What are you using?

I've got DaVinci Pro but haven't dived into it yet. This might be a good time. Otherwise I've done a couple of edits using iMovie.

I'll grab your latest export and try and do some editing with it over the weekend.

Bob-the-Kuhn, 2020-06-05:
Here's a link to the Ozone version of the debugging video.

[Marlin code debug overview - STM32, Ozone, J-Link](https://www.dropbox.com/s/objb2qm5rolrbzm/Marlin%20code%20debug%20overview%20-%20STM32%2C%20Ozone%2C%20J-Link.mp4?dl=0)

---

I'll play with the ABM code changes later today.

Bob-the-Kuhn, 2020-06-05:
@thinkyhead - if I replace `open . ` with `explorer . ` then an explorer window opens to the build directory.  

Is that the behavior you want?

Bob-the-Kuhn, 2020-06-05:
Here's the upload video with all the comments addressed.

[Easy Marlin Upload - LPC176x & STM32 - VSCode & Auto Build Marlin](https://www.dropbox.com/s/14mbkiyhf8m9mvq/Easy%20Marlin%20Upload%20-%20LPC176x%20%26%20STM32%20-%20VSCode%20%26%20Auto%20Build%20Marlin.mp4?dl=0)

Bob-the-Kuhn, 2020-06-06:
Here's a debug video using Ozone and J-Link probes.

[Marlin code debug overview - STM32, Ozone, J-Link](https://www.dropbox.com/s/objb2qm5rolrbzm/Marlin%20code%20debug%20overview%20-%20STM32%2C%20Ozone%2C%20J-Link.mp4?dl=0)

thinkyhead, 2020-06-14:
> Is that the behavior you want?

Thanks for the fix! I figured it had to be something like `iexplode` or `exploder`. Ideally I want the `firmware.bin` file to be highlighted too. On macOS I can (hopefully) do this by commanding the Finder with `osascript`. Maybe `explorer` has a flag for this…?

thinkyhead, 2020-06-14:
This will (supposedly) work on Windows…
```js
    var fname = 'firmware.bin'; // just for testing
    command_with_ping(aterm, 'cd ' + relpath, false);
    command_with_ping(aterm, 'Explorer.exe /select,' + fname, false);
    command_with_ping(aterm, 'exit', false);
```

Bob-the-Kuhn, 2020-06-16:
Both result in an explorer window with **firmware.bin** highlighted.

It was my editing process that resulted in the highlight disappearing.

Here's the video with the correction made.
[Easy Marlin Upload - LPC176x & STM32 - VSCode & Auto Build Marlin](https://www.dropbox.com/s/14mbkiyhf8m9mvq/Easy%20Marlin%20Upload%20-%20LPC176x%20%26%20STM32%20-%20VSCode%20%26%20Auto%20Build%20Marlin.mp4?dl=0)

thinkyhead, 2020-06-16:
Thanks! As time permits, I'll pull that video down for some editing, add titles and things, and publish it privately for review when it's in good shape.

Bob-the-Kuhn, 2020-06-17:
> when it's in good shape

I just can't resist ...

**😉**  What!!??!  Are you implying that something done by the **GREAT** and **WONDERFUL** Bob isn't **PERFECT** the first time?

thinkyhead, 2020-06-21:
LOL. I'm more worried my editing will be terrible the first few tries. It's been awhile……

I'm still on the case, keeping this tab open. But I'm busy getting all the pieces together for a 2.0.6 release. 🤞  And, I will be locking interaction with the repo for the next few days to clear up issues.

boelle, 2020-06-23:
@Bob-the-Kuhn @thinkyhead could we maybe move this one to the discord server instead? just to keep the issue list for bugs only

Bob-the-Kuhn, 2020-06-23:
Bo - I want to keep it here.  I find Discord makes it much harder to follow a thread.

boelle, 2020-06-23:
no problemo, just a thought

Roxy-3D, 2020-07-26:
> Bo - I want to keep it here. I find Discord makes it much harder to follow a thread.

For sure....  I want to see this video.   Not to critique it.   But to learn from it.    I am hoping all my questions about J-Link will be answered and I'll know how to proceed!

closed 2020-11-06 by github-actions[bot]