MarlinFirmware/Marlin #18068: ?? [BUG] ABL does not work properly??

Xenato, 2020-05-22:
hi all,
I have a problem. I used a SKR v1.4 turbo
and have updated marlin to version 2.0.5.3 to use the eeprom as memory. Now the problem is that when I start AUTO_BED_LEVELING_BILINEAR on the leveln it starts on the print bed in the right front corner and not in the left front corner as usual. Is this normal or a bug?
i will hang up my marlin file - maybe you notice something 

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4666376/Marlin.zip)

sjasonsmith, 2020-05-29:
The starting direction is determined by this line of code:

`bool zig = PR_OUTER_END & 1;  // Always end at RIGHT and BACK_PROBE_BED_POSITION`

Any configuration with `GRID_MAX_POINTS_Y` set to an even number (such as the `12` from your configs` will start on the right side.

If leveling is working and this was your only question, I think you can go ahead and close this issue.

boelle, 2020-06-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-21 by boelle