MarlinFirmware/Marlin #18070: [BUG] RADDS 1.5 EEPROM error

ShadowOfTheDamn, 2020-05-22:
### Bug Description

After compiling and uploading the new bugfix #18059 the printer will boot with 
echo:EEPROM version mismatch (EEPROM=? Marlin=V79)
when trying to reset EEPROM via LCD prompt printer resets. using pronterface M502 M500
the same behavior, Using M501 M500 the same behavior,
flashing back to and old marlin (what I got was about a year ago Marlin=V67)
there is no problem and all commands are working correctly including M500.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4667780/Configuration.zip)

### Steps to Reproduce

using arduino due with RADDS 1.5
1. compile and upload new bugfix #18059

**Expected behavior:** being able to write to eeprom or reset it as printer prompts to.

**Actual behavior:** printer resets by the watchdog after trying to write eeprom

#### Additional Information

see the pronterface output with marlin debug activated.

Connecting...
echo:Unknown command: ""
Printer is now online.
echo:start
echo:[2] HAL_init()
echo:[3] runout.setup()
echo:[4] setup_killpin()
echo:[5] setup_powerhold()
echo:[6] SPI.begin()
echo:[7] tmc_init_cs_pins()
echo:[8] esp_wifi_init()
PowerUp
Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-22 | Author: (Nima, default config)
echo:Compiled: May 22 2020
echo: Free Memory: 81988  PlannerBufferBytes: 1728
echo:[16] ui.init()
echo:[133] ui.reset_status()
echo:[134] ui.show_bootscreen()
echo:[3261] settings.first_load()
echo:EEPROM version mismatch (EEPROM=? Marlin=V79)
echo:Hardcoded Default Settings Loaded
echo:[3908] thermalManager.init()
echo:[4163] print_job_timer.init()
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:[4811] endstops.init()
echo:[4812] stepper.init()
echo:[4814] servo_init()
echo:[4815] probe.servo_probe_init()
echo:[4866] endstops.enable_z_probe(0x0)
echo:[4867] fanmux_init()
echo:[4868] bltouch.init( 0x1)
echo:[6219] watchdog_init()
echo:[6220] test_tmc_connection(0x1, 0x1, 0x1, 0x1)
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
echo:[6227] setup() completed.
>>> M502
SENDING:M502
echo:Hardcoded Default Settings Loaded
>>> M500
SENDING:M500
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:start
echo:[1] HAL_init()
echo:[2] runout.setup()
echo:[3] setup_killpin()
echo:[4] setup_powerhold()
echo:[5] SPI.begin()
echo:[6] tmc_init_cs_pins()
echo:[7] esp_wifi_init()
 Watchdog Reset
Marlin bugfix-2.0.x
echo: Last Updated: 2020-05-22 | Author: (Nima, default config)
echo:Compiled: May 22 2020
echo: Free Memory: 81988  PlannerBufferBytes: 1728
echo:[15] ui.init()
echo:[133] ui.reset_status()
echo:[134] ui.show_bootscreen()
echo:[3261] settings.first_load()
echo:EEPROM version mismatch (EEPROM=? Marlin=V79)
echo:Hardcoded Default Settings Loaded
echo:[3908] thermalManager.init()
echo:[4163] print_job_timer.init()
echo:Error writing to EEPROM!
echo:Error writing to EEPROM!
echo:[4811] endstops.init()
echo:[4812] stepper.init()
echo:[4814] servo_init()
echo:[4815] probe.servo_probe_init()
echo:[4866] endstops.enable_z_probe(0x0)
echo:[4867] fanmux_init()
echo:[4868] bltouch.init( 0x1)
echo:[6219] watchdog_init()
echo:[6220] test_tmc_connection(0x1, 0x1, 0x1, 0x1)
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... OK
Testing E connection... OK
echo:[6227] setup() completed.

p3p, 2020-05-22:
In `Marlin/Marlin/src/HAL/DUE/eeprom_wired.cpp` on 
line 41 : `bool PersistentStore::access_start()  { return true; }` change it to:  
`bool PersistentStore::access_start()  { eeprom_init(); return true; }`  
and see if that helps.

ShadowOfTheDamn, 2020-05-22:
@p3p
That didn't correct the issue but thanks for reply.

closed 2020-05-22 by ShadowOfTheDamn

p3p, 2020-05-22:
Glad you updated your original reply or I would have been rather confused ^^

PR #18074 (2020-05-22): Fix DUE wired (i2c) eeprom init
Fix the wired eeprom implementation for the DUE platform.

### Related Issues
fix #18070