MarlinFirmware/Marlin #18072: [BUG] Auto bed leveling not correcting engraving depth

SonOfSardaar, 2020-05-22:
### Bug Description

I have modified my printer to mount V bit for PCB milling and my PCB is 70x100 mm in the center of the bed. I configured bilinear bed leveling with nozzle as probe. When i probe the pcb area i want mill and start the milling then some tracks are being cut at different depth and in some cases the cutting depth changes suddenly. See the pictures attached at the bottom

### My Configurations

[Configurations](https://www.dropbox.com/s/ttzq5xmcs0bgup1/Marlin.zip?dl=0)
### Steps to Reproduce
1. Home and then go to pcb lower left corner G1 X70 Y60
2. Home offsets at this position M206 X-70 Y-60
3. Probe PCB engraving area G29 L0 Y0 R50 B30 (only the rectangle i want to engrave)
4. Start print

**Expected behavior:** This should engrave the pcb tracks at same depth

**Actual behavior:** Pcb tracks are being engraved at different depth and in some cases abruptly changes engraving depth

#### Additional Information

[Image of the pcb](https://www.dropbox.com/s/hk2iii5ipatcbtc/IMG_20200522_121714.jpg?dl=0)

sjasonsmith, 2020-05-29:
My first thought is that you should be probing more points for something like PCB etching. Try increasing GRID_MAX_POINTS_X and GRID_MAX_POINTS_Y to see if that improves things. For engraving traces you probably want the largest number of probe points possible.

SonOfSardaar, 2020-05-30:
Thanks @sjasonsmith the pcb i was trying to mill in this case was only 25x40 mm and that area had 3x3 probes and i thought that would be enough but i will give it a go with more probing points.

boelle, 2020-06-21:
@SonOfSardaar still an issue?

SonOfSardaar, 2020-06-21:
Ya still an issue. I tried with 5x5 points with 3 samples each and it takes a lot of time but no luck. It is still being abrupt as shown in the images

sjasonsmith, 2020-07-22:
I removed the stale label, because those do look like extremely abrupt changes in the pictures provided. It seems like some investigation is warranted.

closed 2020-08-27 by github-actions[bot]