MarlinFirmware/Marlin #18073: [BUG] STM32F103RE composite USB not working

Bob-the-Kuhn, 2020-05-22:
Enabling USB and SDSUPPORT on a BTT_SKR_MINI_E3_V1_0 does not result in a useable virtual drive on the PC.

PR #17400 enabled the composite USB option for the STM32F103RE.

**SETUP:**
BTT_SKR_MINI_E3_V1_0 motherboard  (STM32F103RE with 512K FLASH)
Today's Bugfix-2.0.x
PlatformIO in Atom
Configuration.h changes:
- set SERIAL_PORT to -1
- set MOTHERBOARD to BOARD_BTT_SKR_MINI_E3_V1_0
- enable SDSUPPORT

Auto Build PIO Upload used to build the firmware.bin file
- selected 512K FLASH
- selected USB support

Copied firmware.bin to SD card
Inserted SD card into motherboard
Attached USB cable to motherboard

**RESULT:**
Firmware is updated.
A window pops up on the PC saying the disk needs to be formatted.  
Host USB link is up while this window is up.  Host shows proper SD card contents
Clicked on FORMAT.  
PC said it was unable to format the disk.
Clicked OK and then cancelled all PC windows concerning formatting the drive.
Host USB link stays up.  Host shows proper SD card contents.
A mapped/virtual drive is not present.

**THINGS I'VE TRIED:**
Different size SD cards (128MB & 16GB) - format window shows correct disk size.
Formatted SD card on PC with exFAT & NTFS - now get different errors.

---

I'm not familiar with the Maple environment but it appears to me that the composite USB software wants to use the SD card in SDIO mode (MSC read & write) while the Marlin host interfacing software wants to use the SD card in SPI mode.

sl1pkn07, 2020-05-23:
then if fail with RE chips, also fail with RC chips, right? in #17400 i only add the same things like RC chip for parity between envs

greetings

Bob-the-Kuhn, 2020-05-23:
Do you know of any STM32 controllers where the USB composite works?

sl1pkn07, 2020-05-23:
i'm not use that feature, so idk

greetings

thisiskeithb, 2020-05-23:
> Do you know of any STM32 controllers where the USB composite works?

This was working on an SKR Mini E3 1.2 not too long ago. I’ll get one flashed with the latest bugfix and test it later today.

swilkens, 2020-05-24:
I just flashed bugfix of today on my BOARD_BTT_SKR_MINI_E3_V1_2 with USB composite, everything works swimmingly.

Could it be the windows Maple driver you are using that is causing issues?

thisiskeithb, 2020-05-24:
> I just flashed bugfix of today on my BOARD_BTT_SKR_MINI_E3_V1_2 with USB composite, everything works swimmingly.

Hey, me too! No issues here on macOS Mojave or Windows 10.

Bob-the-Kuhn, 2020-05-24:
Sorry - the board I was actually using is a BTT_SKR_E3_DIP.

I'll open a new issue with the correct info.

closed 2020-05-24 by Bob-the-Kuhn