MarlinFirmware/Marlin #18075: Can't compile LPC1768

Kyosuke33, 2020-05-22:
Hello, i have a problem to compile LPC1768 with VSC.

Just take the new Marlin 2.0.5.3 (also try with bugfix-2.0.x), change default_envs for LPC1768, change the motherboard for BOARD_MKS_SGEN_L, compile, direct error LPC1768 failed.

I have no red line to read to find the error :/

Same error happen to me few month ago with the official release and i have to use the bugfix to make it work, but this time, both won't work :/

Thank you.

EvilGremlin, 2020-05-22:
show full log

Kyosuke33, 2020-05-22:
- toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe9 in position 9: ordinal not in range(128):
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 156:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:

    exec _file_ in call_stack[-1].globals
  File "C:\users\kyosuke\.platformio\platforms\nxplpc-arduino-lpc176x\builder\main.py", line 148:
    target_elf = env.BuildProgram()
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 118:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 327:
    env.ConvertInoToCpp()
  File "C:\Users\Kyosuke\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 240:
    out_file = c.convert(ino_nodes)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 91:
    contents = self.merge(nodes)
  File "C:\users\kyosuke\.platformio\penv\lib\site-packages\platformio\builder\tools\piomisc.py", line 111:
    return "\n".join(["#include <Arduino.h>"] + lines) if lines else None
 [FAILED] Took 0.73 seconds

EvilGremlin, 2020-05-22:
looks like it's hopping to python3, add python2.7 location to PATH system variable

Kyosuke33, 2020-05-22:
Euh, it's like chineese for me xD

Kyosuke33, 2020-05-22:
I search for something call Python on my PC and i find python2.7. Uninstall it and now it work. Don't know what is it and how it was on my PC, but thank you very much for the help !

EvilGremlin, 2020-05-22:
Unusual solve, but oh well... I don't use windows, so no idea how exactly recent platformio behave there. Probably couldn't decide between two python installs.

Kyosuke33, 2020-05-22:
I don't know but thank you for the great help.

sjasonsmith, 2020-05-29:
@Kyosuke33 if this is resolved could you go ahead and close this issue?

Kyosuke33, 2020-05-29:
Yes, sorry, thank you.

closed 2020-05-29 by Kyosuke33