MarlinFirmware/Marlin #18077: Multiple homing mesurement for Z axis

jrf91, 2020-05-23:
Hi everyone!
I'm new on this forum so please explain before shooting!!! I am system engineer and I like complex systems with a lot of physics.

My question is quite "technical", so I apologize in advance for being maybe too long for this new issue.

 I bought a 3d printer for my children 4 months ago (Christmas... ) and work to make it fully plug and play, as they are not able to use it alone.
I recently bought an Ender 3 pro which has been deeply modified to improve its capabilities. After testing 3D-touch which has been successfully mounted inside the nozzle cover, I came to the conclusion that this feature was not accurate enough to get the targeted printing accuracy for the first layer. The M48 test shows a standard deviation between 5 to 12µm highly depending on the temperature of the bed. After watching the so good sensor accuracy video [https://www.youtube.com/watch?v=BLzCeH1hS9g](url), I decided to use a basic endstop one can find on Creality limit switch. I mounted the switch on the X carriage slightly upon the nozzle output so that it doesn't disturb the print, and a retractable floating brass tip is controlled with a SG90 mini servo, and inserts vertically between the switch and the bed. When the nozzle goes down, the tip hits the bed and controls the switch directly without any strength. The servo is controlled via the 27 pin of the main board, and the switch is connected on the z_limit connector. The z_probe works like a switch mounted on an arm, with the advantage of being incentive to any working game. Very accurate! The M48 gives a standard deviation between 1 to 2.6µm, with very low temperature effect. Considering that the range of measurement is roughly 3 to 3.5 times the standard deviation, this results in up to 8µm range random measurement for a simple z homing. My target is to reach less than 5µm, ideally 2µm, that is to say 1% of 200µm (one standard layer) for a perfect first layer control. An easy way to improve the measurement is to mean a set of N measurements, as the mean accuracy will be the standard deviation divided by SquareRoot(N). 10 to 16 measurement would be great! Not too long and very effective... Here comes my issue : 

 **A great Marlin improvement would be to allow multiple homing measurement on the Z axis with a control parameter (just as MULTIPLE_PROBING) , returning the mean of the measurements as the Z homing position.**

I tried to use the linear bed leveling with multiple probing with only one point grid (GRID_MAX_POINTS_X == 1), but it is not allowed. I do not want to use the bed leveling option, because it would be too long at the beginning of the print with so many points to measure. More over, my Z axis has a dead working zone of about 30µm : this means that it is useless to try to drive movements of different direction in this zone, because nothing will happen. This is due to the resistive torque of the Z carriage vs the stepper torque curve : very difficult and expensive to upgrade! As my bed is quite flat (less than 20µm difference between the corners and the center), it would be nonsense to use the bedleveling which would probably result in side effects. To improve the precision of the first layer, the nozzle is first driven to Z=0.1 in the g-code file at the beginning of the print, and then the print is achieved with upward steps (g-code generated with Cura) in order to have constant layer hight. This works very well and is reliable.

This issue "only needs" to implement a loop in the marlin_main.ccp "void homeaxis" line 3278 Marlin 1.1.9. I am not able to do it alone, it is now a long time I didn't code anything, and C++ is hard to me!

I apologize again for being so long. I hope this will catch your attention and I am looking forward to reading your thoughts soon! Have nice 3d prints!

I

GMagician, 2020-05-23:
I read all your document, just one note I want to enlight. Considering a prusa like printer with 200 steps, 8mm on screw, 16 microstepping, theoretical step position is 0,0025, then in a really really really better situation home precision can't be lower than this, but, I really doubt motor will move exactly such step. Microstepping is done splitting current on coils and that position is "unstable", maybe not always reachable (due to mechanical frictions) then, even if you can detect a really precise homing of 2,5µm then no one will assure you motor will reach such position when printing first layer.
If you add to this backlash, bed unplanarity, thermal deformations, then I think that there are other component in this equation, not only home position detection. By the way, a "multiple probing" homing using mean value may help with "unprecise" bltouch (clones)

jrf91, 2020-05-23:
GMagician,
many thanks for your answer and the many areas of improvement! I wake up this morning just as excited as Christmas day!!!
You are right for the microstepping, resistive torque, friction, etc. But here is my way of thinking : 
1 - I recorded the precision of Z drive. It is better than 5µm provided that the direction of variation is respected. When i order let's say Z=100mm (upwards), then back to 90mm, and 100 and 90 - 100, etc, the highest position is reached within a range of 5µm measured by a dial gauge. This is really reliable!
2 - The M48 test with the very precise micro switch supports this results, as the SD is 2.5µm worst case, which corresponds to a range of 7.5µm. This is comparable.
3 - In order to have the accuracy only limited by the mechanical frictions of the machine, I need the homing procedure to be theoretically more precise. That is why, regardless of entering the devilish details of all that may happen behind the doors, I decided to target the theoretical 2µm position accuracy using multiple homing. Anyway, a single touch is not enough. From the coding point of view, there would be no great difference of work between 2 or 50 touches...
4 - You are also right in noting that multiple homing would be an easy way to improve poor bed sensors...
5 - If all above is true, then I assume that the final accuracy will be near 5µm in position (2µm in standard deviation) reached from lower Z, so that the accuracy on my first layer will be about 2.5%. that would be perfect for me!

So, my hope is at least one of the Marlin developers  will be interested in this improvement! May be you can help? Shall post anywhere else? How can I have some more people in the discussion? Thanks again.

swilkens, 2020-05-23:
There are some things that may be relevant;

You could change the Z-axis lead screw to one with lower speed. Standard is 8*8, you could switch to a 8*2 to obtain 4x higher z-resolution per step.

If you have trinamic drivers, Marlin recently implemented phase homing. Where the phase of the stepper motor when the Z-endstop is triggered is recorded. Subsequently, the stepper is then brought to one of the 4 main steps where the coils are in direct alignment (max. torque). This last one especially will mean the variation on the endstop is no longer relevant (within some accuracy bandwidth), the stepper will always return to the same max-torque phase - which should represent the same position in cartesian space - every tme.

GMagician, 2020-05-23:
@jrf91 please note that in your point 1, moving up, down, up your 5µm are 2 microstep (this is due to what we said before), you forgot backlash.. You have to move down, then move up an amount and measure it. Difference is backlash. Backlash is not usually considered because of method of Z calibration.
You move down until a sheet of paper is free to move...
@swilkens changing screw from 8mm round to 2 means also 4xe "theoretical" precision, 1/4 max speed (you can't raise frequency to infinite and 8bit board may have problem). And home on full torque phase is important only for home repeatability, not precision, and for sure when you move on first layer...if not on full phase position you may not reach it....

jrf91, 2020-05-23:
Tempting solution... the interest I see in this solution (T8 lead screw 2mm/turn) is the resistive driving torque of the T8 nut being divided roughly by 4... so that the overall resistive torque of the Z carriage should be lessen, and the dead zone reduced as a consequence (but not by 4!!!). Note that the Z homing is achieved going downwards, so that the resistive torque is quite probably low thanks to gravity. Actually, I am not able to determine precisely the contributors to the Z resistive torque and inaccuracy. Let alone give a descending order! Anyway, my prints have a small inaccuracy on the Z layers (the vertical wall slightly waves with some 2mm period ), and your idea makes me thinks the T8/8mm lead screw could be hold for responsible. Thank you for this idea!!!
Concerning the phase homing, i didn't know this feature; I will take a look at this very soon. I run Marlin 1.1.9. I could also try to lock the Z stepper on full phase, and tune the bed hight to meet Z=0. More difficult... For me, the easiest way is multiple homing, but I think T8/2mm screws would help (and it is not an expensive upgrade).

jrf91, 2020-05-23:
@GMagician, I assume the Z backlash being less than the 2 microsteps, because I use inverted anti backlash T8 nuts. they are mounted upside down on the Z side carriages so that the springs push the nut mounting part downwards, in the same direction of the gravity. Said in another way, the gravity and the spring of the spring nut work in the same direction, downwards. This is why my Z position is so reliable. Today, I would rather think that the position inaccuracy rather comes from random frictions on the Z carriage (Nut/Lead, wheels, V-slot, bearings, belt...) which are very difficult to master.
Multiple homing is an easy way to diagnose if this is true or not. This conversation may decide me to modify the code in  the 

> marlin_main.ccp "void homeaxis" line 3278 Marlin 1.1.9

A simple loop may do the job. I will post if done!

jrf91, 2020-05-23:
@GMagician, more over, both Z screws are constrained vertically on their bearings, so that they have no backlash (but a higher resistive torque...)

swilkens, 2020-05-23:
> ...And home on full torque phase is important only for home repeatability, not precision, and for sure when you move on first layer...if not on full phase position you may not reach it....

Isn't that the goal though? In the end, the first layer distance from the bed should be two things:

1. Accurate
2. Repeatable

It doesn't matter if the endstop triggers with a certain bandwidth if the stepper motor then always jumps to the closest coil phase (1/4th rotation). This is safe, as the endstop trigger bandwidth will never be a full rotation off. 

This will give you an accurate and repeatable Z position. The bed can then be leveled at this position. Note this only works for situation where the bed itself can be leveled with screws, like the ender 3.

GMagician, 2020-05-23:
backlash has nothing to to with constrain a screw on both sides (even if this may destroy bearing in a long term if not proprly aligned) backlash is some "play" on POM nut. it can be avoid. No backlash is "fixed joins" so you have it. Pn industrial machines is measured as I told you. In vertical axes gravity remove it since load remove it on one direction. But when you go down and you set this as home position (or move down and set z probe offset). When you move up (first layer) you see all backlash you have...

bellerofonte, 2020-06-04:
> After testing 3D-touch which has been successfully mounted inside the nozzle cover, I came to the conclusion that this feature was not accurate enough to get the targeted printing accuracy for the [...]

Hi everyone!
I have the same setup and I've came to the same conclusion that would be great to make Z-homing process more sophisticated. 
Here is an example approach how to achieve that:

F.e. user sends `G28 P4 S0.06`
There
- P{N} - take at least N probes
- S{StDev} - desired StDev

Printer does following
1. Take 1 fast probe, N-1 slow probes
2. We have array z_offset[N]. Compute StDev.
3. If StDev is less (not greater) than given - take average as result and return. 
4. Otherwise - take one more probe.
5. Now we have array z_offset[N + 1]. Compute average z_offset_avg.
6. Compute z_offset_err[i] = abs(z_offset[i] - z_offset_avg)
7. Find index of max{z_offset_err} and remove this element from z_offset.
8. Go to 2.
9. Report failure if StDev is still greater than desired after f.e. 2*N probes

jrf91, 2020-06-04:
@bellerofonte : what a joy to feel less alone in this immense universe!
Sorry i didn't follow my own issue; your comment reminded me to erase this oversight. I did recently implement a multiple homing in the Marlin 1.1.9 software and made it work. The result is more than convincing! With a standard deviation between 1 and 1.6µm measured with the M48 Probe Accuracy Test (simple end-stop switch near the hot end as probe), I decided to use a 10 homing mean for zero Z. Here is how I proceeded: (Please excuse my very rough C++ approach, I didn't understood everything in the Marlin software...).
I defined z_multiple homing in configuration.h
#define Z_MULTIPLE_HOMING 10;

I inserted the following lines in Marlin_main.cpp, near the end of the gcode G28 procedure : 
        //JR***
        #if ENABLED(Z_MULTIPLE_HOMING)
          do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_MULTI_PROBE, MMM_TO_MMS(Z_PROBE_SPEED_FAST));
          DEPLOY_PROBE();
          current_position[Z_AXIS] = run_z_probe();
          do_blocking_move_to_z(current_position[Z_AXIS], MMM_TO_MMS(Z_PROBE_SPEED_SLOW));
          set_axis_is_at_home(Z_AXIS);
          sync_plan_position();
          destination[Z_AXIS] = current_position[Z_AXIS];
          do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_MULTI_PROBE, MMM_TO_MMS(Z_PROBE_SPEED_FAST));
          STOW_PROBE();          
        #endif
        //JR***

of course this is working only for my Cartesian printer as I am not able to verify this for different types of printers. But  it seems to work for me. I just need to define the number of probing for the homing mean, and it works.
I tested this add-on, with a 1 layer ring printed at the center of the bed where the head probes for multiple homing with the G28 control. The measured thickness of the rings ranges between 198µm and 205µm (with an outside micrometer), printed at different time and days (6 independent prints). So, I improved the first layer accuracy from 35µm range to 7µm range!!! The first layer accuracy is then 3.5% as my target was 2.5%. Still need some upgrade!
Of course, I will need to copy this every time I change my Marlin version, unless the improvement becomes native.
Have nice prints!

bellerofonte, 2020-06-05:
Hi @jrf91,
First, thank you for your response in that question.
I took a look at your code and got confused.
Where is the value of `Z_MULTIPLE_HOMING` used? It's defined as `10`, but your code only checks the value exists or not.
I suppose you also use this value somewhere else. Probably, you use it either inside the `run_z_probe` method or in more elegant way in `Configuration.h`, smth like following:
```
// The number of probes to perform at each point.
//   Set to 2 for a fast/slow probe, using the second probe result.
//   Set to 3 or more for slow probes, averaging the results.
#if ENABLED(Z_MULTIPLE_HOMING)
  #define MULTIPLE_PROBING Z_MULTIPLE_HOMING
#else
  // Use Marlin's default value
  #define MULTIPLE_PROBING 3
#endif
```
Am I right? If not please describe your approach a bit deeper.

jrf91, 2020-06-05:
@bellerofonte , I wrote this a bit too fast without proofreading. Actually, I only define Z_MULTIPLE_HOMING, but I specify the number of home probings through MULTIPLE_PROBING, which I valued at 10. run_z_probe does all the necessary stuff alone, so that I had not a great and complicated work to achieve multiple z homing.
The //JR*** code is inserted at the end of the Z_HOME test 👍 , ~line4587, marlin_main.cpp
    // Home Z last if homing towards the bed
    #if Z_HOME_DIR < 0
      if (home_all || homeZ) {
        #if ENABLED(Z_SAFE_HOMING)
          home_z_safely();
        #else
          homeaxis(Z_AXIS);
        #endif

        //JR***
        #if ENABLED(Z_MULTIPLE_HOMING)
          do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_MULTI_PROBE, MMM_TO_MMS(Z_PROBE_SPEED_FAST));
          DEPLOY_PROBE();
          current_position[Z_AXIS] = run_z_probe();
          do_blocking_move_to_z(current_position[Z_AXIS], MMM_TO_MMS(Z_PROBE_SPEED_SLOW));
          set_axis_is_at_home(Z_AXIS);
          sync_plan_position();
          destination[Z_AXIS] = current_position[Z_AXIS];
          do_blocking_move_to_z(current_position[Z_AXIS] + Z_CLEARANCE_MULTI_PROBE, MMM_TO_MMS(Z_PROBE_SPEED_FAST));
          STOW_PROBE();          
        #endif
        //JR***

        #if HOMING_Z_WITH_PROBE && defined(Z_AFTER_PROBING)
          move_z_after_probing();
        #endif

      } // home_all || homeZ
    #endif // Z_HOME_DIR < 0

If i want do deactivate multiple z homing, i just comment //#define Z_MULTIPLE_HOMING; and everything is if it had never been. As i didn't understand deeply the following line :  
 Line 1461 SBI(axis_known_position, axis);
  SBI(axis_homed, axis);

nor the planner / buffer move control (do_probe_move or do_blocking_move_to_z), so that i am not 100% sure this is perfectly well implemented without any bug...  Keep your printer safe! Hope you get it, so you can try it on your side if you wish. Regards.

bellerofonte, 2020-06-05:
@jrf91, it seems that I've understood everything right.
I will try this code on my Ender 3
Thanks!

jrf91, 2020-06-05:
@bellerofonte , may I allow me to clarify that my printer is no more a stock ender 3 pro. It has quite a lot of mechanical add-ons necessary to meet the above accuracy. I upgraded it with double Z axis (mono stepper + belt, and precise alignment of the lead screws), inverted anti backlash nuts for lead screws, constraint on all bearings, fine tuning of the perpendicularity of the frame, upgraded V-slot wheels + tuning of the bearings constraint, direct drive extruder, independent unspooler, no spring for bed leveling,  X&Y belt tension management system, foam golf ball feet, external MOSFET for bed heater, silent 1.1.5 board with TMC drivers, always-on fan for the main board, X-carriage endstop switch for z-probe with a retractable tip, upgraded heatsink for TMC extruder driver, upgraded hot-end mount, auto-stop switch. Quite a lot of work, but it is now really plug and play (and forget)! It would be very interesting to compare the soft upgrade on a stock machine, and see if the work i did was useful! Would you mind telling me the result, should you have the mean to measure precisely? I am quite interested. Regards.

bellerofonte, 2020-06-05:
@jrf91, sure I will. It will take some time, but I will write back asap.
Unfortunately, the only tools I have to measure results - are 1) {G29, M48, etc} g-codes output in terminal and 2) print results. Both of them are kinda stochastic, but I think I would see if this improvement helps or not.

boelle, 2020-06-21:
@jrf91 still an issue?

closed 2020-06-21 by jrf91