MarlinFirmware/Marlin #18081: [BUG] Adafruit_MAX31865 @ 1.2.0 on stm32 is not compatible with marlin

ellensp, 2020-05-23:
### Bug Description

If your a new user and attempt to compile for a  stm32 based board. platformio will grab the latest Adafruit_MAX31865 from https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
This is currently v1.2.0 

This results in 
|-- <Adafruit MAX31865 library> 1.2.0
|   |-- <Adafruit BusIO> 1.2.6      
|   |   |-- <Wire> 1.0

and  Adafruit BusIO fails to compile with
Adafruit_SPIDevice.h:37:43: error: 'SPI_BITORDER_MSBFIRST' was not declared in this scope

### My Configurations

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/4671356/configs.zip)

### Steps to Reproduce

1. Get latest marlin bugfix
2. Apply provided Configs
3. Set default_envs = STM32F103RC_btt_512K
4. Compile

**Expected behavior:**
It should compile

**Actual behavior:**
It does not compile

#### Additional Information
Full error log https://pastebin.com/366r7YWb

The quick fix is to change 
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
to
Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip
this forces it to use 1.1.0. It then compiles.

Was discovered by a user celex on discord. 

This may effect other platforms also...  haven't had a chance to check.

q32103940, 2020-05-23:
new user here. i wasted half a day trying to figure out what went wrong. relieved to hear it is a bug. I eventually got around it by commenting out this library as my setup did not actually require it.

sjasonsmith, 2020-05-24:
I believe this is just an issue with the STM32F1 HAL, but I didn't check _all_ of them. The newer version uses a 3-argument flavor of `TwoWire::requestFrom` that doesn't exist in the maple framework. I will post a change shortly to revert the version, but this will do it for all environments. A different solution may be needed long-term.

crysxd, 2020-05-25:
I can confirm this behaviour :( Maybe it's a good idea to reference a specific version instead of master. Also old builds which worked no longer compile...

kramer04, 2020-05-25:
I confirm this bug with Marlin bugfix downloaded today
fix if i change for Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip 
Thanks to @ellensp

qwewer0, 2020-05-25:
I don't have this issue with SKR Mini E3 v1.2

sjasonsmith, 2020-05-25:
> I don't have this issue with SKR Mini E3 v1.2

@qwewer0, this probably means your builds are using an older version from the library cache.
If you are on Windows you can delete your `Marlin/.pio` and `%USERPROFILE%\.platformio\.cache` and then the problem will appear for you as well.

pa3axa, 2020-05-26:
Same problem when building for BOARD_MKS_ROBIN_NANO, problem exist on OSX, Linux and Windows 7. Changing MAX31865 from master to 1.1.0 in platformio.ini fixes this.

ellensp, 2020-05-26:
A patch was added to Marlin bugfix for this 14 hours ago, you need the latest bugfix

closed 2020-05-26 by ellensp

AudioKev, 2020-05-28:
> I confirm this bug with Marlin bugfix downloaded today
> fix if i change for Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip
> Thanks to @ellensp

Where, in which portion of the Marlin code is the change made? I found a url reference to "https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip" in the .piopkgmanger.json file under libdeps > STM32F103_btt_512K > Adafruit_MAX31865 > .pio. Changing that reference to "https://github.com/adafruit/Adafruit_MAX31865/archive/1.1.0.zip" does not correct the probelm; the reference to "master.zip" is restored when compilation is attempted. So, where is the correct location to make the change to "1.1.0.zip?"

Thanks.

ellensp, 2020-05-28:
@AudioKev it is only needed in older code. It has already been patched in bugfix a different way See https://github.com/MarlinFirmware/Marlin/pull/18089/files for details.  
It was in platformio.ini

AudioKev, 2020-05-29:
Thanks @ellensp, switching to the bugfix 2.0.x build solved the problem.

sam01508, 2020-07-08:
I had a marlin config for skr mini e3 v1.2 it compiled fine until yesterday evening when I updated st stm32 version 7.7.1. Is there anyway to change to the older version? fresh copy of marlin (bugfix) does't compile either.

thinkyhead, 2020-07-09:
> I had a marlin config for skr mini e3 v1.2 …

In an Ender 3?

thinkyhead, 2020-07-09:
The latest `platformio.ini` has `platform = ststm32@<6.2.0` to stay within the supported realm. This patch will be in 2.0.6 but it may also be made in a quickie 2.0.5.4 release.