MarlinFirmware/Marlin #18083: [FR] add support for 2004 character OLED

knickohr, 2020-05-23:
Hi !

Is it possible to add support for the Winstar or Raystar OLED ?

https://www.winstar.com.tw/products/oled-module/oled-character-display/20x4-oled.html
https://www.raystar-optronics.com/oled-character-display-module/20x4-oled-display.html

They use the WS0010/RS0010 controller, is nearly compatible with the HD44780 LCD controller, but not 100%. The initialization routine is a bit different. Some guys from Prusa community managed it working.

Datasheet for the controller :
https://datasheetspdf.com/pdf-file/1110077/ETC/WS0010/1

thisiskeithb, 2020-11-18:
To get it working in the interim, change this line in `platformio.ini` from:
https://github.com/MarlinFirmware/Marlin/blob/41529b65988a58ba512977e0fe1692aaeeb6c811/platformio.ini#L227

To:
`USES_LIQUIDCRYSTAL      = LiquidCrystal=https://github.com/gerdj/LiquidCrystal/archive/master.zip`

It's a custom library that works with both traditional HD44780 LCDs and the WINSTAR OLED.

jbyerline, 2022-08-20:
Bump... I would also love to see this feature. I cant imagine it would be hard because the display is so similar to the standard 2004 LCD. @thisiskeithb I tried the library above but it doesnt seem to work in newer versions of marlin.

thisiskeithb, 2022-08-20:
> I tried the library above but it doesnt seem to work in newer versions of marlin.

I use it in my Prusa AIO fork, but it’s only compatible with LPC-based builds.

https://github.com/thisiskeithb/PrusaAIO/blob/79b53616406ea09230a93c031aac6ef9bb3449ac/ini/lpc176x.ini#L26-L27

`custom_marlin.USES_LIQUIDCRYSTAL = LiquidCrystal=https://github.com/thisiskeithb/LiquidCrystalOLED/archive/1.0.8.zip`

jbyerline, 2022-08-20:
> 

@thisiskeithb What does LPC stand for?

thisiskeithb, 2022-08-20:
> What does LPC stand for?

It’s the processor type on your motherboard. LPC1768/LPC1769.

jbyerline, 2022-08-20:
Ah, okay I am using an Einsy Rambo. So in that case, back to my original point. Is this a feature that is relatively simple to integrate?

jbyerline, 2022-10-07:
Any new thoughts on adding this functionality to Marlin 2.0?

closed 2022-12-17 by knickohr

jbyerline, 2022-12-18:
What version was this feature added in?

thisiskeithb, 2022-12-18:
It was never added.

jbyerline, 2022-12-18:
So why was it marked as closed?

ellensp, 2022-12-18:
original requester closed it, probably because they no longer want it, or just got sick of getting e-mails

thisiskeithb, 2022-12-18:
@jbyerline Feel free to [open a new Feature Request](https://github.com/MarlinFirmware/Marlin/issues/new?assignees=&labels=T%3A+Feature+Request&template=feature_request.yml&title=%5BFR%5D+%28feature+summary%29) if you would still like to see OLED 2004/Character LCD support in Marlin.

knickohr, 2022-12-18:
Found a solution, replaced a modified library. But the code in not from my brain, so it’s not public.

[image]

Works with a 1:1 replacement OLED, like the Winstar series.

jbyerline, 2022-12-18:
Can you post a link to it please.