MarlinFirmware/Marlin #18084: Is Marlin sending the wrong info?

ruggb, 2020-05-23:
bugfix-2.0.x.dde8fb9
also occurs in at least the previous 2 builds
Repetier v2.1.6

### Bug Description
Reperier log reports
20:24:21.634 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000
when nozzle is at 0.10mm

### My Configurations
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4672135/Marlin.zip)

### Steps to Reproduce
I home all with G28  
Z_AFTER_HOMING is set to 30mm
I then run this script to measure my offset

g0 x50 y50 ; place XY on bed
g28 z ; home Z and raise to 30mm (Z_AFTER_HOMING)
g0 z0 ; Z=0 = measure offset

It goes to x50, y50
then homes Z which ends up at Z=30mm
then it move Z to 0 to test my offset which is set at 0.20mm

Repetier Log
20:23:54.770 : N79 G28*45
20:24:03.945 : X:0.00 Y:0.00 Z:30.00 E:0.00 Count A:0 B:0 Z:24000
20:24:11.405 : N80 G0 X50 Y50*16
20:24:11.405 : N81 G28 Z0*96
20:24:11.405 : N82 G0 Z0*89
20:24:21.634 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000

So why is Repetier reporting Z=30? The probe actual position is about 0.10mm from the bed. (paper thickness)
If I click Z 1 + step, Z rises to 31mm.

21:01:42.183 : N100 G1 Z31 F1000*54

**Expected behavior:** 
I expected Repetier to report Z=0
Since Repetier appears to think the nozzle is at 30, I can understand why it moved to 31 when I told it to move 1mm. Although, if I clicked step 1mm it should do a relative 1 step not absolute = Repetier issue.

**Actual behavior:**
Wrong position reported.

#### Additional Information

If I send the commands separately, not via the script, the position is correctly reported.
If I do not have G0 Z0 in the script and send it after the script runs it also is correctly reported.

So, is it Marlin or Repetier stumbling over the script??

Lord-Quake, 2020-05-23:
For comparison what does pronterface say?

ruggb, 2020-05-23:
I have not used PF.

GMagician, 2020-05-24:
What does repetier log says if you enable ack? Marlin should report its positions after home.
If in such report Marlin return correct Z you have to move this question to repetier developper.

P.S. Old repetier, after home, set its positions to the one configured in it. Newer should get them from Marlin reported positions

ruggb, 2020-05-24:
Not sure what I am looking for but this is what I get with ACK
08:03:28.293 : N17 G28*37
08:03:28.293 : N18 G0 X50 Y50*17
08:03:28.293 : N19 G28 Z0*97
08:03:28.293 : N20 G0 Z0*81
08:03:28.608 : T:25.00 /0.00 B:24.84 /0.00 @:0 B@:0
08:03:29.615 : T:25.09 /0.00 B:24.88 /0.00 @:0 B@:0
08:03:30.311 : echo:busy: processing
08:03:30.623 : T:25.23 /0.00 B:24.77 /0.00 @:0 B@:0
08:03:31.631 : T:24.96 /0.00 B:24.92 /0.00 @:0 B@:0
08:03:32.325 : echo:busy: processing
08:03:32.636 : T:25.11 /0.00 B:24.92 /0.00 @:0 B@:0
08:03:33.645 : T:25.28 /0.00 B:25.11 /0.00 @:0 B@:0
08:03:34.340 : echo:busy: processing
08:03:34.652 : T:25.26 /0.00 B:25.09 /0.00 @:0 B@:0
08:03:35.661 : T:25.17 /0.00 B:25.03 /0.00 @:0 B@:0
08:03:36.355 : echo:busy: processing
08:03:36.666 : T:25.23 /0.00 B:25.06 /0.00 @:0 B@:0
08:03:36.976 : X:0.00 Y:0.00 Z:30.00 E:0.00 Count A:0 B:0 Z:24000
08:03:36.976 : ok
08:03:36.978 : ok
08:03:37.674 : T:25.03 /0.00 B:24.53 /0.00 @:0 B@:0
08:03:38.682 : T:24.65 /0.00 B:24.73 /0.00 @:0 B@:0
08:03:38.992 : echo:busy: processing
08:03:39.689 : T:25.14 /0.00 B:24.92 /0.00 @:0 B@:0
08:03:40.724 : T:25.14 /0.00 B:24.65 /0.00 @:0 B@:0
08:03:41.006 : echo:busy: processing
08:03:41.730 : T:25.20 /0.00 B:24.84 /0.00 @:0 B@:0
08:03:42.737 : T:25.23 /0.00 B:25.00 /0.00 @:0 B@:0
08:03:43.021 : echo:busy: processing
08:03:43.745 : T:25.09 /0.00 B:24.69 /0.00 @:0 B@:0
08:03:44.752 : T:25.40 /0.00 B:24.88 /0.00 @:0 B@:0
08:03:45.036 : echo:busy: processing
08:03:45.760 : T:25.20 /0.00 B:25.03 /0.00 @:0 B@:0
08:03:46.766 : T:25.17 /0.00 B:25.00 /0.00 @:0 B@:0
08:03:47.052 : echo:busy: processing
08:03:47.196 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000
08:03:47.196 : ok
08:03:47.198 : ok
08:03:47.775 : T:24.84 /0.00 B:24.77 /0.00 @:0 B@:0
08:03:48.782 : T:25.11 /0.00 B:24.92 /0.00 @:0 B@:0

GMagician, 2020-05-24:
> 08:03:47.196 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000

This is position feedback from Marlin, it reports correctly 30 after home. Since you have a script that moves to 0 after home. Marlin does what it's required and terminate to such position.
Now let's see repetier side: It read 30 after home, it doesn't move the Z and it doesn't read anymore postion from Marlin (this is what I know) then it uses last position it was aware of. And if you move by 1 mm it moves to 31.
I had this same problem some time ago, even without movement. After homing repetier used only it's configured home position, even if Marlin comunicated correctly its new postion. This has been changed in Repetier.
1) Repetier could poll via M114 new position from time to time, but it doesn't.
2) Marlin could send current position after a move (but as written before, I suspect repetier doesn't care of this)

ruggb, 2020-05-24:
So, what you are saying is that it is a Marlin issue because this was the last command that was sent and processed
08:03:28.293 : N20 G0 Z081
And this was Marlin's answer
08:03:47.196 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000
when actual Z=0.1 (which is Z=0 + offset)

GMagician, 2020-05-24:
Depends on where you are looking at.
Is Marlin because it doesn't sent its position everytime or is repetier because it doesn't poll it (or worst it doesn't use it)?
Now you may try do execute a M114 and see if repetier sync its position. You can put it as last command of your script, but as I wrote more than once, I suspect repetier will not use such info

ruggb, 2020-05-24:
with M114 at end of script RH reports Z=0. So, it appears that RH is responding to all info it gets from Marlin.

09:53:07.885 : T:25.77 /0.00 B:25.60 /0.00 @:0 B@:0
09:53:08.709 : echo:busy: processing
09:53:08.874 : X:50.00 Y:50.00 Z:30.00 E:0.00 Count A:20000 B:0 Z:24000
09:53:08.874 : ok
09:53:08.876 : ok
09:53:08.878 : X:50.00 Y:50.00 Z:0.00 E:0.00 Count A:10000 B:10000 Z:0
09:53:08.878 : ok
09:53:08.891 : T:25.80 /0.00 B:25.91 /0.00 @:0 B@:0

So, it is a Marlin issue. After the move to Z=0 Marlin reports the wrong Z position.
It doesn't NOT report, it reports it wrong.
If M114 then Marlin reports the correct position which RH then displays.

And why didn't the M114 command show up?
It does when I send it separately. Does Marlin echo the command or is that strictly an RH thing?

10:00:59.263 : N22 M114*23
10:00:59.268 : X:50.00 Y:50.00 Z:0.00 E:0.00 Count A:10000 B:10000 Z:0
10:00:59.268 : ok

RH also issues step commands as absolute. It works if it know the correct position.

GMagician, 2020-05-24:
> with M114 at end of script RH reports Z=0

Then maybe something is changed in meantime.

> It doesn't NOT report, it reports it wrong.

It doesn't report it seems, looking at your log. Only returned position is after homing.

I think only viable fix is to return position on end of a script (implicit M114) because it almost impossible in other ways (movements go into a queue and this queue is purged inside an ISR)

> And why didn't the M114 command show up?

because TH shows command it sends...but where is written your script? in TH or in Marlin?

If it's in TH then is th issue

ruggb, 2020-05-24:
If I am following you;
When Marlin returns the position info it is ONLY after G28 and G28 Z.
It does not return any info for any other moves. Is that design intent???
If so, then RH MUST issue an M114.
Script is in RH. So, RH should show M114.

I think I have found it. RH problem with scripts
G28 Z
G0 Z0
RH does not catch catch the g0 Z0 and sets Z=30
inputting G28 Z then running script with only G0 Z0 RH indicates Z=0
or sending commands separately works also.
Since Z=0 the commands are sent together. Marlin then responds with position info and RH displays that. Totally forgetting that it also send G0 Z0.

ruggb, 2020-05-24:
the work around is to end the script with an M114 until/unless/whenever RH is fixed.

closed 2020-05-24 by ruggb