MarlinFirmware/Marlin #18085: [FR] enable remainting print time for character type displays

knickohr, 2020-05-23:
Hi !

In Marlin 2 I found the remainting print time and status bar calculation. Ist's unfortunately only activated with graphical displays. So why, what*s the difference ?  Is it possible to activate it also for character type displays, like the RipRap 2004 LCD controller ?

valeriu-ene, 2021-03-06:
This feature is already available in Marlin 2.0.7.2, I'm using it since February, with 20x4 LCD. The gcode must has have M73 commands in it.
Check my pictures below, the feature is working great but it has a small defect related to LCD display.
https://github.com/MarlinFirmware/Marlin/issues/21271
(also containing a link to my uploaded config files)

In config file I have enabled: 
SDSUPPORT
REPRAP_DISCOUNT_SMART_CONTROLLER

In config_adv I enabled: 
LCD_SET_PROGRESS_MANUALLY
SHOW_REMAINING_TIME
USE_M73_REMAINING_TIME
ROTATE_PROGRESS_DISPLAY

I hope this helps.

thisiskeithb, 2021-03-06:
This was implemented in https://github.com/MarlinFirmware/Marlin/pull/19416. Closing.

closed 2021-03-06 by thisiskeithb