MarlinFirmware/Marlin #18086: [FR] Support for Geeetech A30 OEM LCD

SonOfSardaar, 2020-05-23:
I can see there is a configuration available for Geeetech A30 printer, it does not sound like the OEM LCD is supported yet. It would be really great to have the LCD working without having to replace it. Also i could not find any instructions on making Marlin work for Geeetech A30 (even with a new lcd)

ellensp, 2020-05-24:
Very little to go on.
Product page. https://www.geeetech.com/32-inch-touch-screen-for-e180-a30-a30m-p-1084.html
Firmware binary. https://www.geeetech.com/download.html?version_id=62 
No Schematics or source code for display that I can find.

SonOfSardaar, 2020-05-25:
How about Geeetech github repositories? there seems like some sort of source code there . I am not to sure what it is for though
https://github.com/Geeetech3D/Smartto-Eclipse

SonOfSardaar, 2020-05-25:
[Here ](https://www.geeetech.com/download.html?download_id=43)are some more documents / schematics

Phr3d13, 2020-08-02:
I don't think they ever released the LCD firmware source, only the printer firmware source.

Vertabreak, 2021-04-02:
closed source little we can do.

TheThomasD, 2021-04-20:
Hi,

I was in contact with @Vertabreak about the A30T that I have some time ago. From the picture from the Geeetech store link it seems that my A30T has a similar touch screen. I collected some information about the display [here](https://github.com/TheThomasD/GeeetechA30T/tree/master/touchscreen). Hopefully I will have the time to collect some more information. If you have some specific information that you  need, just give me a ping and I'll try to retrieve some information. Hopefully the information that I collected helps to get started with support for the display. My C skills are a little rusty at best (I used to do some robot programming in the old days, but now I'm mainly focused on Java), but as soon as I can find the time I'd be happy to help with some coding. I'd need some guidance into the Marlin stuff, though. BTW, my Geeetech printer has a control board that seems not to be supported as of now. I'd also like to look into that, but I wanted to be sure that the display can be supported before I switch to Marlin :)

asos-gurpreetsingh, 2021-07-06:
@TheThomasD 
I believe in order to make a start we need source examples of how to tell LCD screen which image to put where in order to create a UI for sending signals to marlin (GTM32 Board).

TheThomasD, 2021-07-07:
Hi @asos-gurpreetsingh,

I was aiming for a different approach. My aim was to use the original firmware on the touch screen and just implement the exchange with the board over the serial connection as an addition to the existing Marlin code. There, the display-specific codes can be translated to “real” g-code or to direct method calls. That restricts what can be done via the UI to what is implemented by geeetech in the LCD software, but most of the important parts would be accessible. The protocol for communication seems rather easy.
As @Vertabreak pointed out, the source of the LCD software is closed, so I think creating a firmware for that will be much harder.
What do you think?

SonOfSardaar, 2021-07-25:
I have been able to obtain a document from them which explains protocol of lcd commands. It is in Chinese which can either be translated via google or some one can help translate it. I tried reading some using google translate and it could be useful if we use existing UI on the LCD and parse the commands sent by it. This wont be ideal but it will help use current LCD with limited functionality.
[SmartoLCD.docx](https://github.com/MarlinFirmware/Marlin/files/6874693/SmartoLCD.docx)

ellensp, 2021-07-26:
Docx is Microsoft only format, A lot of the developers use linux and mac's. So here is the docx translated into a pdf
[SmartoLCD.pdf](https://github.com/MarlinFirmware/Marlin/files/6875190/SmartoLCD.pdf)

catalinalb, 2021-10-01:
Looking forward for that LCD firmware :)

SonOfSardaar, 2021-10-01:
They won't fcuking give it man I tried.

Regards
Gurpreet

SonOfSardaar, 2021-10-01:
They won't give it, I tried. I will be happy to use  another LCD if I can plug it into this board but even that's a challenge.

Regards
Gurpreet

SonOfSardaar, 2021-10-01:
This is better than nothing , considering if you use octoroprint , you would hardly touch LCD. But i understand there are still issues controlling y axis as I tried Marlin and its not just LCD.

Regards
Gurpreet

catalinalb, 2021-10-01:
Have you managed to get the marlin firmware working? I have compiled the right configuration and managed to upload the bin file with the geeetech flashing tool but I have to change the name of the bin file in order to be recognized by the flashing tool.
So it flashed the baseboard successfully.
It dose not respond to the Touchscreen and also dose not work over octoprint. Octoprint tries to connect but it gives two messages something with App1 App2 that's it.

So I couldn't manage to get it to work.

If you still have your binary file I appreciate if you could share it or tell me if I do something wrong here.

Thank you,
Catalin

SonOfSardaar, 2021-10-01:
You need to use jumper on the board and flash Marlin. There is a jumper on board that says boot 0 . Then Marlin will be flashed bilut y axis still does not work

Regards
Gurpreet

catalinalb, 2021-10-01:
Thanks a lot,👍
I will try tomorrow and let you know. 🙂

SonOfSardaar, 2021-10-01:
Touchscreen will no and can not respond when you flash Marlin, cuz Marlin does not know what touch screen is talking about

Regards
Gurpreet

SonOfSardaar, 2021-10-01:
I used some stm32 tool to flash Marlin. You can find it if you search for flash Marlin on here tech a30. I will try to find it too.

Regards
Gurpreet

Vertabreak, 2021-10-01:
having the lcd attached to the board at all locks up the coms port preventing anything else from connecting to the printer.
are for y axis issues i had no issues with moving/homing any axis on my A30 with marlin.
as for attaching another display i plugged in a reprap discount display from my A10 to the A30 and works perfectly.

SonOfSardaar, 2021-10-01:
Where can I get that display from, if I don't have a10. One with flex ribbon

Regards
Gurpreet

catalinalb, 2021-10-02:
Hi,
I did managed to put the board in bootloader mode by inserting the jumper. But now the original programmer complains that it is not seeing the board over serial. I did try with the display removed as well. 

I did use STM32Cube Programmer but did not find any board in DFU mode. 

Shortening the boot jumper means it is in DFU mode? Can I use something like dfu-utils or with what else can I flash the board?

Regards,
Catalin

SonOfSardaar, 2021-10-02:
You will need to use a different starting address with stm32 cube. There is a post I saw somewhere . Try using 8300000##

Regards
Gurpreet

catalinalb, 2021-10-02:
Thank you,

I did managed to flash it with FLASHER-STM32. 

I can talk with it over octopi the only issue is confirmed that the Y axies tries to move but it dose not have enough power. Heating, X and Y axies are working great! 

From now on I just have to tweak the code maybe change the stepper motors etc.

Vertabreak, 2021-10-02:
when i flash my A30 its using the usb cable and uploading direct from vscode.

catalinalb, 2021-10-02:
Hi,

Can you Please send me the binary file?

I did managed to put Marlin but the Y axis and the Extruder dose not work.

Thank you

krabbs, 2021-10-03:
Hello,

Hello,

i am running an A30 with the "GTM32 Pro VD" Board with bugfix tree . i got almost all things to working but i need more testing:

-BLtouch is working
-Beeper is working
-x,y,z and extruder axis is working
-SD Card is not working
-Serial Connection is limited because of following hotfix for y-axis (however octorpint works)
-fans are working 

According the y-Axis i run in the same problem but i could resolve it:

MarlinCore.cpp needs one modification at the end of the setup(). Add a line with:
Serial.end();

This is a terrible hotfix but the y-axis is working now

The pin mapping is wrong for my board, so here is my working one:
```

#pragma once

#include "env_validate.h"

#ifndef BOARD_INFO_NAME
  #define BOARD_INFO_NAME    "GTM32 Pro VB"
#endif
#define DEFAULT_MACHINE_NAME "STM32F103VET6"

#define BOARD_NO_NATIVE_USB

#define DISABLE_DEBUG

//
// It is required to disable JTAG function because its pins are
// used as GPIO to drive the Y axis stepper.
// DO NOT ENABLE!
//
#define DISABLE_JTAG

//
// If you don't need the SWDIO functionality (any more), you may
// disable SWD here to regain PA13/PA14 pins for other use.
//
#define DISABLE_JTAGSWD

// Ignore temp readings during development.
#define BOGUS_TEMPERATURE_GRACE_PERIOD    2000

// Enable EEPROM Emulation for this board as it doesn't have EEPROM
#if EITHER(NO_EEPROM_SELECTED, FLASH_EEPROM_EMULATION)
  #define FLASH_EEPROM_EMULATION
  #define MARLIN_EEPROM_SIZE              0x1000  // 4KB
#endif

//
// Limit Switches
//
#define X_MIN_PIN                           PE5   // ENDSTOPS 15,17
#define X_MAX_PIN                           PE4   // ENDSTOPS 16,18
#define Y_MIN_PIN                           PE3   // ENDSTOPS 9,11
#define Y_MAX_PIN                           PE2   // ENDSTOPS 10,12
#define Z_MIN_PIN                           PE0   // ENDSTOPS 3,5
#define Z_MAX_PIN                           PE1   // ENDSTOPS 4,6                          

//
// Steppers
[... 147 lines cut ...]
// #define SS_PIN PB12 // PC<<<11
// #define SCK_PIN PB13 // PC12 // PC1
// #define MOSI_PIN PB15 // PD2 // PD2
// #define MISO_PIN PB14 // PC8
// #define SD_DETECT_PIN PC7

  #define SD_SS_PIN                         PC11
  #define SD_SCK_PIN                        PC12
  #define SD_MOSI_PIN                       PD2
  #define SD_MISO_PIN                       PC8
  #define SD_DETECT_PIN                     PC7
#else
  //
  // Use the on-board card socket labeled TF_CARD_SOCKET
  //
  #define SD_SS_PIN                         PA4
  #define SD_SCK_PIN                        PA5
  #define SD_MOSI_PIN                       PA7
  #define SD_MISO_PIN                       PA6
  #define SD_DETECT_PIN                     -1    // Card detect is not connected
#endif

#define SDSS                           SD_SS_PIN

```

catalinalb, 2021-10-03:
Wooo, thanks a lot! 🙂

I will give it a try for sure!

I am planning of using the BL touch and the Biqu H2 direct extruder mounted on linear rails, so once I get the marlin working I will create a metal holder for the extruder.
I would appreciate if you could share your bin file since I think I will get stuck at compiling the code, but I will definitely give it a try!

Thanks once again,
Catalin

krabbs, 2021-10-04:
> Wooo, thanks a lot! 🙂 I will give it a try for sure! I am planning of using the BL touch and the Biqu H2 direct extruder mounted on linear rails, so once I get the marlin working I will create a [...]

Here you can find my stuff and i made a release for you
https://github.com/krabbs/Marlin/releases/tag/0.1

Be careful. my board is different to yours! i think extruder and endstops wont match. but i am unsure. there are so much different informations about these boards

krabbs, 2021-10-04:
> having the lcd attached to the board at all locks up the coms port preventing anything else from connecting to the printer. are for y axis issues i had no issues with moving/homing any axis on my [...]

Hey, how did you manage to flash without FLASHER-STM32? 
Can you share your settings for the A30. I have everything working but i think my workaround is a bad one (adding Serial.end() to setup()).

I think i missed something according the serial communication

Vertabreak, 2021-10-04:
all of my work with the A30 was merged officially in to marlin more then a year ago nothing to share that is not already published.

asos-gurpreetsingh, 2021-10-04:
Hey @Vert what "flat FFC" cable will connect from discount lcd display to a30 board ? the connectors seem to be of different sizes

Thanks

Vertabreak, 2021-10-04:
the A30 came with more then a dozen different boards over its life span and there still sneaking in changes to this day so its hard to say as some of the board don't even have the lcd port at all.

catalinalb, 2021-10-04:
> > Wooo, thanks a lot! 🙂 I will give it a try for sure! I am planning of using the BL touch and the Biqu H2 direct extruder mounted on linear rails, so once I get the marlin working I will create a metal holder for the extruder. I would appreciate if you could share your bin file since I think I will get stuck at compiling the code, but I will definitely give it a try! Thanks once again, Catalin Get Outlook for Androidhttps://aka.ms/ghei36
> > […](#)
> 
> Here you can find my stuff and i made a release for you https://github.com/krabbs/Marlin/releases/tag/0.1
> 
> Be careful. my board is different to yours! i think extruder and endstops wont match. but i am unsure. there are so much different informations about these boards

Hi, I just upladed the bin file and it works! 👍 💯 
Do you know where to connect the BL Touch sensor on the board? 

Regards,
Catalin

krabbs, 2021-10-05:
> Hi, I just upladed the bin file and it works! 👍 💯 Do you know where to connect the BL Touch sensor on the board?
>
> Regards, Catalin

In my config there is
Z_MIN_PROBE_PIN PA7

PA7 is the JST connector next to the fillament runout connector in my board.
As @Vertabreak said, there are so many differences between the boards. 
I had to puzzle together all the interfaces myself with a connection tester starting from the chip because most of the information on the Internet didn't fit to my board. I would be surprised if everything works for you if you take over my bin

TheThomasD, 2021-11-18:
Hi all, I managed to make my Geeetech A30T with GTM32_103_V1 board work with Marlin 2.x. Everything to be working so far (including BLTouch). Now I'm back at the display problem. I was able to switch the display on (i.e. supplying 5v power) by pulling up the PD12 pin (59 on my processor layout sheet, but somehow numbered 60 when using Marlin). I was even able to create a simple LCD extui that pulls this pin up on startup. Now, the problem I'm facing is communicating with the display. I found out that the RX/TX lines to the display are on pins PD9/PD8. In the pinout description of the chip these seem to be part of the FSMC connection. However, a remapping of USART3 would exactly do what I need: USART3 connected to RX/TX of the display. Unfortunately, I cannot find out how to do the remapping of USART3 to those pins. I read something about afio_remap(), but I couldn't really manage to configure the remapping in Marlin.

Can someone provide some insight on how this can be done? On this basis I'd like to implement the Marlin part to communicate with the OEM touch screen that came with my printer.

SonOfSardaar, 2021-11-18:
@TheThomasD  how do i find out the version of my GTM32 board ?
and regarding the USART3 remapping i did a quick search and i [found this ](https://os.mbed.com/questions/79855/Is-there-a-function-to-remap-UART1-STM32/) . Not sure if this will help

TheThomasD, 2021-11-18:
@SonOfSardaar The version of my board was printed on it. You can find photos of my board [here](https://github.com/TheThomasD/GeeetechA30T). Thank you for the links, however, from what I saw in the code Marlin seems to already provide the config for the UARTs, I just haven't figured out how this config works... or if it works at all. At least the USB-to-serial converter that I connected to the UART does not show any output nor could I send commands. Maybe it IS actually correctly configured to using the desired ports, but I just didn't make it work correctly?!

BTW. I have the working config for my Geeetech A30T [here](https://github.com/TheThomasD/Marlin/tree/geeetech-A30T). As soon as the display is working I'm planning on merging that into the Marlin base... but the display is a must for me.

TheThomasD, 2021-11-19:
FYI, I was able to solve my problem and I'm currently on my way implementing an extui for the OEM touchscreen. Maybe this also helps for the A30 OEM LCD? You can see my current progress (which is not too much right now, but a start) [here](https://github.com/TheThomasD/Marlin/tree/geeetech-A30T). I also collected some more info about the Geeetech A30T [here](https://github.com/TheThomasD/GeeetechA30T/) so you can compare your boards to my setup.

SonOfSardaar, 2021-11-26:
Hi @TheThomasD  good work man. Were you able to upload the firmware from VS Code ? or you used something else to upload it. I remember someone mentioning that they were able to upload code from VS Code. My board is GTM32 Mini S and i wanted to give that a try as the chips are same.

SonOfSardaar, 2021-11-27:
Thanks. @TheThomasD . I read your wiki page and that was good documentation. As I have GTM32 Mini S , i was wondering if the SD card method will work for flashing marlin on that board.

TheThomasD, 2021-11-27:
@SonOfSardaar I always used the STM Cube Programmer tool as described in the USB flashing part of the tronxy. I‘ve never tried the SD method as I need to flash very often for testing my touch screen implementation progress. It also did not work on my Tronxy BTW so I didn‘t even try on the Geeetech. But I guess if you find a guide from Geeetech for upgrading and rename your firmware upgrade file to what Geeetech expects in that process it might work?! Have a backup ready, though ;)

moccor, 2022-02-22:
> Hi all, I managed to make my Geeetech A30T with GTM32_103_V1 board work with Marlin 2.x. Everything seems to be working so far (including BLTouch). Now I'm back at the display problem. I was able to switch the display on (i.e. supplying 5v power) by pulling up the PD12 pin (59 on my processor layout sheet, but somehow numbered 60 when using Marlin). I was even able to create a simple LCD extui that pulls this pin up on startup. Now, the problem I'm facing is communicating with the display. I found out that the RX/TX lines to the display are on pins PD9/PD8. In the pinout description of the chip these seem to be part of the FSMC connection. However, a remapping of USART3 would exactly do what I need: USART3 connected to RX/TX of the display. Unfortunately, I cannot find out how to do the remapping of USART3 to those pins. I read something about afio_remap(), but I couldn't really manage to configure the remapping in Marlin.
> 
> Can someone provide some insight on how this can be done? On this basis I'd like to implement the Marlin part to communicate with the OEM touch screen that came with my printer.

Hi. So are you saying this is working very well if using something like OctoPrint?

TheThomasD, 2022-02-22:
Hi @moccor,

I haven’t done extensive testing, yet. My main focus was on the implementation of the display part. I recognized that the printer stops occasionally when I enable my display implementation. I didn’t have the time, yet, to improve the code there.

if you like you can try out the printing process without the display code and report back. I would be interested to get to know the results.

cheers
Thomas

moccor, 2022-04-14:
@TheThomasD what would you say could be the worst case scenario for trying this and something bad happening? I only ask cuz, I have no experience with flashing firmware on printers, and I don't mind having to spend 30$ or so on a new motherboard if need be. Buttt I am kinda hoping that like, there is no greater potential of fires than normal or anything like that lol, since my printer is in a wood cabinet.

I went to go calibrate my A30T esteps, and ofc the Smartto software is trash and not only doesn't fully work, but it explains nothing and the "Send command" seems to not work, so I have no idea how I can even calibrate it if I can't send commands. So now I kinda wanna try using a stock build of Marlin

TheThomasD, 2022-04-15:
@moccor Well, if you have the same board as I have: there are at least two other people that tried my Marlin adoption successfully. You should do a PID calibration after installation and might need to reduce the bed size a little bit. Apart from that, I cannot think of any risks. Be sure to take a backup of the original firmware so you can go back. However, I cannot guarantee nothing will go wrong, so you have to decide whether you want to try it.

As for the calibration and sending commands: I used repetier server to connect to the original firmware and that worked well. However, you can only change the e steps for the first extruder with the original firmware. However, with Marlin and a mixing extruder config you cannot tune the e steps separately as well. So if that is your aim: cannot be done with this Marlin config.

Cheers
Thomas

TheThomasD, 2022-04-15:
Oh, and one general comment: with the Marlin software it should be possible to connect another display easily. The original display for the A30T uses a serial connection. The same is true, for example, for the Artillery Sidewinder X1 display. I haven’t tried, but with a slight change of the Marlin config you should be able to use the Artillery display. The software for that display is customizable.

moccor, 2022-04-15:
> 

I have a couple dumb questions. I downloaded Repetier Server and am trying to use that, but it still seems to not work. Is it something I am doing wrong? Here is a screenshot of what I was doing. I was both sending G1 E100 F50 and using the little extrude tab in the button right, but neither of them did anything.

This printer annoys me so much that simple well known commands don't work. I emailed them yesterday telling them they should at the least have a firmware file that allows using these settings even if it means revoking warranty/support. It isn't many commands, but it sure is a few of the very important commands. For that reason alone I keep considering your custom firmware.

I believe it is the same motherboard, if not the same exact one it might be  a slightly different version but it looks almost 1:1, but I would def make sure its the same before I flash anything. Thank you for this info and any help

TheThomasD, 2022-04-15:
@moccor I think your questions are a little bit off-topic here. Maybe you want to the telegram chat https://t.me/+qlIpbz8vWBljNTNh

Apart from that: did you preheat the hotend? Most firmwares require a minimum hotend temperature before extrusion is possible. The repetier console should ideally report that in a message („cold extrusion prevented“ or similar).

moccor, 2022-04-15:
You are very right, sorry I didn't notice. I will def join the telegram, thank you.

SonOfSardaar, 2024-01-27:
> Hi @asos-gurpreetsingh,
>
> I was aiming for a different approach. My aim was to use the original firmware on the touch screen and just implement the exchange with the board over the serial connection [...]

I guess it is too late to say it is a good idea :) Did not you make any progress ?

TheThomasD, 2024-01-28:
Hi, I implemented a (not yet very good) [ExtUI](https://github.com/TheThomasD/Marlin/tree/2.1.x-geeetech-a30t/Marlin/src/lcd/extui/geeetech_a30t) for Marlin. Nevertheless, I documented some info about the interface with the display [here](https://github.com/TheThomasD/GeeetechA30T). Maybe this is similar to the display you have?

SonOfSardaar, 2024-01-28:
Wow . Looking at that two and other documentation you have put together , great job man. I will try this out on my gtm32 mini s board . I was wondering if you came across something that will allow flashing without boot mode jumper . And would the flashing be done on 800000 or 8002000

Regards
Gurpreet

TheThomasD, 2024-01-28:
I think all the info about flashing and so on should be in my documentation. As far as I remember I needed the boot jumper, but I just connected a small switch to that to not be required to even the printer every time. Apart from that, I think I used the standard settings of the STM Cube Programmer. That is 800000, I think. Be sure to make a backup of the whole chip with the tool to have something to revert to :)

SonOfSardaar, 2024-01-28:
Ya I have a backup. I always thought that 800000 had bootloader and it will load the user program from 8002000 other another address. I tried klipper too but it did not work. Klipperscreen would be good option to use as printer LCD. By the way I flashed Marlin on my board but I am yet to try your ui. I will let you know.

Regards
Gurpreet