MarlinFirmware/Marlin #18091: [BUG] RESTORE_LEVELING_AFTER_G28 ignored in Marlin 2.0.x

lockheed, 2020-05-24:
I have tried several versions of Marlin 2.0.x, including couple of -bugfix version released over the last months: mid-April and 24th of May 2020), as well as 2.0.5.3.

In all cases, even though they are compiled with Mesh Bed Leveling enabled and RESTORE_LEVELING_AFTER_G28 uncommented, the bed leveling settings are not being restored before every print. Even running "Load settings" from Bed Leveling menu doesn't work. The only thing that works is running gcode **M420 S**

Here's printer log: https://pastebin.com/93sb0Mfe

xaviervilla, 2020-05-28:
Same experience here. This topic has been opened and closed several times. The documentation in Merlin should be more clear about what RESTORE_LEVELING_AFTER_G28 is actually doing. It isn't re-enabling the bed leveling, it is simply restoring the state, so if it is off, then it stays off. AKA doing literally nothing. (See here: https://github.com/MarlinFirmware/Marlin/issues/14609) 

It shouldn't be difficult to implement the proper solution. Thousands of users want to use mesh bed leveling on every print. We shouldn't have to remember to add M420 command to every gcode file for each slicer we use.

I guess this should really be a 'feature request'

xaviervilla, 2020-05-28:
here is another description of the misunderstanding we both shared: https://github.com/MarlinFirmware/Marlin/issues/18015

lockheed, 2020-05-28:
Well, it's not just us. Every how-to, youtube or otherwise, on the subject of mesh bed leveling is saying the same thing. Even comments in Marlin code say so.
Plus, I am pretty sure it worked like this in Marlin 1.1.9.

xaviervilla, 2020-05-28:
I agree, the misinformation is everywhere about this. This is my first time using mesh bed leveling, so I can't speak on previous versions of Marlin. 

I've been studying the code to see if there is an easy approach to auto-enabling MBL after the G28 command is called. Perhaps we need a #define ENABLE_LEVELING_AFTER_G28, that simply checks if there is a mesh present and enables the mesh if so. It seems a bit absurd that this isn't already the case, but I understand a lot more efforts go to things like ABL and the repo mantainers don't always have time to fix issues like this when users can simply add M420 S1 to the gcode.

The other (probably better) option would be to simply **save the bed leveling state ( planner.leveling_active) to EEPROM when the M500 command is run.** so that on boot it is always on. That way if RESTORE_LEVELING_AFTER_G28 is enabled, the bed leveling will always be on unless a user disables it and updates eeprom again. I'm hoping a maintainer can pitch in and provide insight. This may be something I'm willing to implement in the "bugfix" branch for community testing.

BTW do you work for LM? I start work there in July!

xaviervilla, 2020-05-29:
TLDR;

RESTORE_LEVELING_AFTER_G28 is working as **intended**, but _not_ as we **expected** it to- due to the wording of the documentation.

It is **intended** to keep the state (on or off) after G28 command.
If MBL was off, it remains off after G28. If MBL was on, it remains on after G28.

Here is my proof it is working as intended. After calling M420 S1 and then G28, MBL reains on.
https://pastebin.com/VbCt88fX

I'd like to know your opinion. My opinion is that the **REAL issue** is that every time the printer reboots, MBL defaults to off, even when storing EEPROM.

xaviervilla, 2020-05-29:
I reccomend if you want to get persistant, always on MBL, read the pull request I just made. You can also install the branch I just created and test it if you'd like. I tested it on my spare printer motherboard, and it seems to work as expected.

lockheed, 2020-05-31:
Thanks for your exhaustive explanation. I do agree that the problem is MBL is not automatically on once it is set and saved to EEPROM. It used to worked like that in Marlin 1.1.9. As it is now, there is no way for user to enable it permanently, short of adding the gcode to every print. Which is neither intuitive nor easy to stumble upon as web is littered with advices that do not apply to Marlin 2.0.x.

However, It seems Marlin team is happy with how the things are now...

No, I'm not working for LM. I'm not even in America. The nickname is of sentimental nature.

xaviervilla, 2020-06-02:
I received feedback from the community. MBL was designed to be manually turned on each time. What you and I want to do is actually UBL. There is no Z probe or extra hardware required for UBL. Gosh why was that so hard to discover😕

see here for details https://github.com/MarlinFirmware/Marlin/issues/18151

boelle, 2020-06-21:
@lockheed still an issue?

shitcreek, 2020-06-22:
@lockheed You can always use `#define STARTUP_COMMANDS M420 S` to have it activate bed leveling at start up.

Closing because `RESTORE_LEVELING_AFTER_G28` is working as intended ie 'restores' bed leveling state after homing.

closed 2020-06-22 by shitcreek

AnHardt, 2020-06-22:
> Closing because RESTORE_LEVELING_AFTER_G28 is working as intended ie 'restores' bed leveling state after homing.

To be exact: Restoring to the state directly before starting G28 - Not to a state at a random time before.