MarlinFirmware/Marlin #18092: Compile Error LPC1768 Re-ARM

Hubertusfan, 2020-05-24:
### Bug Description

Compiling failed with Error:  1501 |     #error "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

### My Configurations

BQ PRUSA I3 with Panucott Re-ARM, RAMPS Board, marlin 2.0.x 

[Configuration.zip]
(https://github.com/MarlinFirmware/Marlin/files/4674148/Configuration.zip)
(https://github.com/MarlinFirmware/Marlin/files/4674149/Configuration_adv.zip)
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/4674201/pins_RAMPS.zip)

### Steps to Reproduce

1. Compile with Visual Studio Code and PIO

**Expected behavior:** Should compile without error

**Actual behavior:** "USE_CONTROLLER_FAN requires a CONTROLLER_FAN_PIN. Define in Configuration_adv.h."

#### Additional Information
Same source compiles for Atmega 2560 error free. 
I've copied source in new project folder and changed only as requested  at https://marlinfw.org/docs/basics/install_rearm.html
 the Board to #define MOTHERBOARD BOARD_RAMPS_14_RE_ARM_EFB and #define SERIAL_PORT -1 
I'am not the big programmer, but is it possible that line 217 at pins_RAMPS.h is the problem?
Original line: #if ENABLED(IS_RAMPS_EFB)
I've changed to:  #if ENABLED(IS_RAMPS_RE_ARM_EFB), same error in compile. 

With disabled #define USE_CONTROLLER_FAN compilation is error free.

borland1, 2020-05-25:
You shouldn't need to modify the pins file ( actual file for your board define in configuration.h file is named: pins_RAMPS_RE_ARM.h ).  FAN_PIN is defined in pin file as: RAMPS_D9_PIN.

I don't see the problem with your config files.  Perhaps your platformio.ini file is in error.

ellensp, 2020-05-25:
The re-arm and RAMPS has 3 MOSFET's
HEADED BED, EXTRUDER and Cooling Fan. All of these are in use. So you have no standard places to put a Controller fan.
By default it uses the following in Configuration_adv.h
#define CONTROLLER_FAN_PIN    FAN1_PIN // Set a custom pin for the controller fan 
But the re-arm doesn't have FAN1_PIN defined.

You need to add a #define FAN1_PIN P?_?? {the pin you have a external mosfet on)

Hubertusfan, 2020-05-25:
Oh shit, i forgot that i've used a external MOSFET Print on my printer with Arduino. 

Thank you very much for your input. I think i can solve now my mistake.

ellensp, 2020-05-26:
Please close this issue.

closed 2020-05-26 by Hubertusfan