MarlinFirmware/Marlin #18096: [FR] CUSTOM_USER_MENU As Submenu To Configuration Menu

borland1, 2020-05-24:
CUSTOM_USER_MENU's option in Configuration_adv.h, are only available to be placed as menu item in Main Menu.  Request CUSTOM_USER_MENU's also be available as sub-menu items for "Configuration" menu item.

I am implementing peripheral hardware equipment (WiFi radio) to be controlled using g-code M118 to send commands over wired SERIAL_PORT_2.  This would allow turning ON/OFF the WiFi/Bluetooth radio to conserve power should user wish to use printer with USB or as standalone.  For this implementation, CUSTOM_USER_MENU would be better placed as a sub-menu item under the "Configuration" setting main menu item.

Tannoo, 2020-05-26:
You could copy these lines from `Marlin/src/lcd/menu/menu_main.cpp`:
```
#if ENABLED(CUSTOM_USER_MENUS)
  void menu_user();
#endif
```
and paste them into `Marlin/src/lcd/menu/menu_configuration.cpp` at LN# 180.

I give you this for your test purposes.

borland1, 2020-05-27:
Thanks for the hints Tannoo.  I followed your guidance, but results were unsuccessful.

Here's what I did..  the Custom_menu setup in Configuration_adv.h file still looks like this:
```
/**
 * User-defined menu items that execute custom GCode
 */
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  #define CUSTOM_USER_MENU_TITLE "Custom Commands"
  //#define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN

  #define USER_DESC_1 "Home & UBL Info"
  #define USER_GCODE_1 "G28\nG29 W"

  #define USER_DESC_2 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  //#define USER_DESC_3 "Preheat for " PREHEAT_2_LABEL
  //#define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  //#define USER_DESC_4 "Heat Bed/Home/Level"
  //#define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  //#define USER_DESC_5 "Home & Info"
  //#define USER_GCODE_5 "G28\nM503"
#endif
```
Note the menu defines in USER_ DESC and USER_GCODE above are labeled 1 through 5.  I added this new code block with similar labels, but for 6 through 10.
```
#define CUSTOM_CONFIGURATION_MENUS
#if ENABLED(CUSTOM_CONFIGURATION_MENUS)
  #define CUSTOM_USER_MENU_TITLE "Custom Commands"
  //#define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN

  #define USER_DESC_6 "Home & UBL Info"
  #define USER_GCODE_6 "G28\nG29 W"

  #define USER_DESC_7 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_7 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  //#define USER_DESC_8 "Preheat for " PREHEAT_2_LABEL
  //#define USER_GCODE_8 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  //#define USER_DESC_9 "Heat Bed/Home/Level"
  //#define USER_GCODE_9 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  //#define USER_DESC_10 "Home & Info"
  //#define USER_GCODE_10 "G28\nM503"
#endif
```
Then, In file menu_configuration.cpp, I added this code block, per your direction, right after #ifENABLED(DUAL_X_CARRIAGE)... #endif:
 ```
#if ENABLED(CUSTOM_CONFIGURATION_MENUS)
  void menu_user();
#endif
```
Problem is, the Menus 1, 2, 6, and 7 are all placed in the Main menu under "Custom Commands".  No "Custom Commands" are shown in the Configuration menu.  I still haven't figured out how the menu macros work.  I'll try looking at it later.

Tannoo, 2020-05-27:
Okay. I will have to test it myself. I thought it would work. I will see what I can do in the am.

Tannoo, 2020-05-27:
Alright... I have done and tested this:

To add items to the custom user menu, edit `configuration_adv.h` here (This section starts @ LN #3059):
 
```
/**
 * User-defined menu items that execute custom GCode
 */
#define CUSTOM_USER_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
  #define USER_SCRIPT_DONE "M117 User Script Done"
  #define USER_SCRIPT_AUDIBLE_FEEDBACK
  //#define USER_SCRIPT_RETURN  // Return to status screen after a script

  #define USER_DESC_1 "Home & UBL Info"
  #define USER_GCODE_1 "G28\nG29 W"

  #define USER_DESC_2 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  #define USER_DESC_3 "Preheat for " PREHEAT_2_LABEL
  #define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  #define USER_DESC_4 "Heat Bed/Home/Level"
  #define USER_GCODE_4 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nG28\nG29"

  #define USER_DESC_5 "Home & Info"
  #define USER_GCODE_5 "G28\nM503"

  #define USER_DESC_6 "Endstop Status"
  #define USER_GCODE_6 "M119"

  #define USER_DESC_7 "TEST 7"
  #define USER_GCODE_7 "M503"

  #define USER_DESC_8 "TEST 8"
  #define USER_GCODE_8 "M503"

  #define USER_DESC_9 "TEST 9"
  #define USER_GCODE_9 "M503"

  #define USER_DESC_10 "TEST 10"
  #define USER_GCODE_10 "M503"

  #define USER_DESC_11 "TEST 11"
  #define USER_GCODE_11 "M503"

  #define USER_DESC_12 "TEST 12"
  #define USER_GCODE_12 "M503"

  #define USER_DESC_13 "TEST 13"
  #define USER_GCODE_13 "M503"
#endif
```

It is coded in `menu_custom.cpp` for up to 25 items. If you need more, I'm sure that can be customized also.

Now, to get it in the `CUSTOM_USER_MENUS` in the configuration menu as a submenu is not too difficult either.
In `menu_configuration.cpp` (Starting @ LN #346):
```
void menu_configuration() {
  const bool busy = printer_busy();

  START_MENU();
  BACK_ITEM(MSG_MAIN);

  #if ENABLED(CUSTOM_USER_MENUS)           // <-- ADD THESE THREE LINES
    void menu_user();                                         //
  #endif                                                             //

  //
  // Debug Menu when certain options are enabled
  //
  #if HAS_DEBUG_MENU
    SUBMENU(MSG_DEBUG_MENU, menu_debug);
  #endif

  #if ENABLED(CUSTOM_USER_MENUS)             // <-- ADD THESE THREE LINES
    SUBMENU(MSG_USER_MENU, menu_user);   //
  #endif                                                               //

  SUBMENU(MSG_ADVANCED_SETTINGS, menu_advanced_settings);

  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    SUBMENU(MSG_ZPROBE_ZOFFSET, lcd_babystep_zoffset);
  #elif HAS_BED_PROBE
… ...
```

If you want a whole other dedicated submenu for your purpose, it is a little more complicated.

borland1, 2020-05-28:
Thanks Tannoo,  I tested it.   It does place the custom menus inside the Configuration Menu, however I found it duplicates the menus in the Main Menu.  I think, they need to be defined separately in `Configuration_adv.h`; also in the menu files.

Tannoo, 2020-05-28:
Yes, it does. I was giving that to you for the time being. Like I said, if you wanted it as it's own, it will take a little more doing.

You first stated that you wanted it in the configuration menu also. That is what I gave you this time.

Tannoo, 2020-05-28:
> Request CUSTOM_USER_MENU's also be available as sub-menu items for "Configuration" menu item.

Tannoo, 2020-05-28:
Have you nailed down the hardware to work?

Also remember, the user menu can have up to 25 items. Just add them in `configuration_adv.h`.

borland1, 2020-05-28:
Yes, I have the WiFi hardware, ESP8266 (D1 Mini Pro), custom PCB adapter for 3.3V/5V RX/TX bi-directional level translator, but waiting for materials to build cable between adapter and printer's AUX-1 connector.
Have burned ESP3D 2.1 WiFi host firmware, but waiting for ESP3D v3.0 to be released, which is supposed to support hardwired serial commands to turn ON/OFF radio transmitter.
Marlin can send ESP3D commands to radio over wired serial using G-code.
This G-code: `M118 [ESP110] OFF pwd=12345678`, will instruct the radio to turn off the transmitter.  ESP3D 3.0 is supposed to save setting in radio's EEPROM for next power cycle.
That's the plan.

Tannoo, 2020-05-28:
Did you want just a menu option to turn it on and off?
Or are there other commands to send to it also?

Tannoo, 2020-05-28:
Okay... here is a new submenu for you.

Insert this in `configuration_adv.h` just under LN # 3084:
```C++
/**
 * User-defined wireless menu items that execute custom GCode
 */
#define CUSTOM_WIRELESS_MENUS
#if ENABLED(CUSTOM_USER_MENUS)
  //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
  #define WIRELESS_SCRIPT_DONE "M117 User Script Done"
  #define WIRELESS_SCRIPT_AUDIBLE_FEEDBACK
  //#define WIFI_SCRIPT_RETURN  // Return to status screen after a script

  #define WIRELESS_DESC_1 "Wifi On"
  #define WIFI_GCODE_1 "M118 [ESP110] ON pwd=12345678"

  #define WIRELESS_DESC_2 "Wifi Off"
  #define WIRELESS_GCODE_2 "M118 [ESP110] OFF pwd=12345678"
#endif
```
Then add this line in `..\Marlin\src\lcd\language\language_en.h` after LN #119:
```C++
PROGMEM Language_Str MSG_WIRELESS_MENU                   = _UxGT("Wireless Commands");
```

Then these in `..\Marlin\src\lcd\menu\menu_configuration.cpp` LN #352:
```C++
  #if ENABLED(CUSTOM_USER_MENUS)
    void menu_wireless();
  #endif
```
and these at LN #363:
```C++
  #if ENABLED(CUSTOM_USER_MENUS)
    SUBMENU(MSG_WIRELESS_MENU, menu_wireless);
  #endif
```
Then copy this file to `..\Marlin\src\lcd\menu`.
[menu_wireless.zip](https://github.com/MarlinFirmware/Marlin/files/4695361/menu_wireless.zip)

That should give you a 2 option submenu in the configuration menu called "_Wireless Commands_".
Those options will be "_Wifi On_" and "_Wifi Off_".

Is this more of what you wanted?

Tannoo, 2020-05-28:
If you checked this by email. Please reload via GitHub as I have updated and edited the last post multiple times.

borland1, 2020-05-29:
I followed your guide and the new menus test good.  It is what I was wanting. Good job Tannoo!

It's unfortunate the `USER_SCRIPT_DONE`, `USER_SCRIPT_AUDIBLE_FEEDBACK`, and `USER_SCRIPT_RETURN` defines are not individually set for each custom menu item. Seems like they should be.

If you don't mind, I will try in the next few days, making some changes and submit a PR.  This to include:
- move `menu_wireless.h` code into file `menu_custom.h`. 
- make `CUSTOM_WIRELESS_MENUS` more generic.  Rename `CUSTOM_USER_MENUS` to `CUSTOM_MAIN_MENUS`.  Rename `CUSTOM_WIRELESS_MENUS` to `CUSTOM_CONFIGURATION_MENUS`.
- add individual defines for USER_SCRIPT_DONE, USER_SCRIPT_AUDIBLE_FEEDBACK, and `USER_SCRIPT_RETURN` on each menu item (1-5, 6-10).

Tannoo, 2020-05-29:
Re-read my edits. The `USER` parts of the tags were changed to the relative `WIRELESS` tags.

Tannoo, 2020-05-29:
I will let you submit any PR you want. I hate doing those anyway.

borland1, 2020-06-01:
> I will let you submit any PR you want. I hate doing those anyway.

Tannoo,  Can you look the PR I submitted from my depository to see why it fails checks.  It complies fine with FW 2.0.5,3, but checks says 'Bad Target' branch.

Tannoo, 2020-06-01:
> I will let you submit any PR you want. I hate doing those anyway.
> 
> Tannoo,  Can you look the PR I submitted from my depository (CustomMenus branch) to see why it fails checks.  It complies fine with FW 2.0.5,3, but checks says 'Bad Target' branch.

I looked around and I am not sure why there is a bad target. Maybe check path lengths? But, I would think that there would have been more errors if that were the case.

I am sure someone here on git knows what might be going on.

borland1, 2020-06-01:
Thanks for looking.  My first PR, so not surprised.  

Here's what I did to create the PR:
- With Marlin branch 2.0.x selected, clicked on 'fork' button to create forked copy in my git repository.
- From my forked Marlin repository, I selected Branch button and typed in 'CustomMenu' to create a new branch.
- With CustomMenu branch selected, I edited the appropriate files and saved changes.
- With CustomMenu branch selected, I clicked Pull Request button, edited background/description/reason text field and then submitted PR.

Wasn't sure which branch to fork, but my thinking was branch 2.0.x would be more stable, so I chose that.

Tannoo, 2020-06-01:
Your PR should be created against `bugfix-2.0.x`. Stablility is not the issue. It's going against the lastest work in progress.

ETE-Design, 2021-01-23:
@Tannoo Seems like your pretty good at the menu system... Is it posssible to make a menu ala the one Prusa use Normal/Stealth for steppers... On my board it is possible to change current on the stepper using M907 "PWM_MOTOR_CURRENT" So would like a menu item who said "Stepper Mode" on the Left side, and in same line on the right can change from Normal to Slient depending on what the PWM_MOTOR_CURRENT is set to... Can you help with that?

Tannoo, 2021-01-24:
Please start a new topic and someone might be able to help. This is old and I have actually been away from Marlin for quite some time. I am sure tons have changed.

I would need some time to look it over, but someone else may be much more up-to-date with the current menu coding for faster help with this.

slowbro, 2021-07-03:
#18177 was merged, this can be closed.

closed 2021-07-03 by ellensp