MarlinFirmware/Marlin #18097: [BUG] STM32F103RE (BTT_SKR_E3_DIP) - composite USB fails when SDSUPPORT enabled

Bob-the-Kuhn, 2020-05-24:
The replaces issue #18073.  I stated the wrong motherboard in that one.

---

Enabling USB and SDSUPPORT on a BTT_SKR_E3_DIP does not result in a useable virtual drive on the PC.

If SDSUPPORT is not enabled then the virtual drive appears on the PC.

PR #17400 enabled the composite USB option for the STM32F103RE.

**SETUP:**
BTT_SKR_E3_DIP motherboard  (STM32F103RE with 512K FLASH)
Today's Bugfix-2.0.x
PlatformIO in Atom
Configuration.h changes:
- set SERIAL_PORT to -1
- set MOTHERBOARD to BTT_SKR_E3_DIP
- enable SDSUPPORT

Auto Build PIO Upload used to build the firmware.bin file
- selected STM32F103RE
- selected USB support

Copied firmware.bin to SD card
Inserted SD card into motherboard
Attached USB cable to motherboard

**RESULT:**
Firmware is updated.
A window pops up on the PC saying the disk needs to be formatted.  
Host USB link is up while this window is up.  Host shows proper SD card contents
Clicked on FORMAT.  
PC said it was unable to format the disk.
Clicked OK and then cancelled all PC windows concerning formatting the drive.
Host USB link stays up.  Host shows proper SD card contents.
A mapped/virtual drive is not present.

**THINGS I'VE TRIED:**
Different size SD cards (128MB & 16GB) - format window shows correct disk size.
Formatted SD card on PC with exFAT & NTFS - now get different errors.

---

I'm not familiar with the Maple environment but it appears to me that the composite USB software wants to use the SD card in SDIO mode (MSC read & write) while the Marlin host interfacing software wants to use the SD card in SPI mode.//github.com/MarlinFirmware/Marlin/issues/new/choose

sl1pkn07, 2020-05-24:
same as #18073. works with RC chips? (DIP with RCT6)

Bob-the-Kuhn, 2020-05-25:
What is really strange is I get a useable mapped virtual drive if SDSUPPORT is not enabled.

When SDSUPPORT is enabled and the SD card is in the controller's socket and the controller is reset (or powered up) then Windows pops up the error message and says the drive needs to be formatted.

sjasonsmith, 2020-05-25:
> What is really strange is I get a useable mapped virtual drive if SDSUPPORT is not enabled.

With LPC you have to unmount the drive in Marlin before it is usable in the host. Are you perhaps encountering the same thing for STM32?

Bob-the-Kuhn, 2020-05-25:
Turns out the problem only occurs when **SD_CHECK_AND_RETRY** is enabled.  

With it disabled the virtual drive is always available and works as expected.  The host also has access to the SD card and can mount and unmount it.  The mount and unmount operations do not affect the availability of the virtual drive.

Bob-the-Kuhn, 2020-05-25:
I've created PR #18108 to keep this from happening to others.

closed 2020-05-25 by Bob-the-Kuhn