MarlinFirmware/Marlin #18102: LPC1769 Compile Error ('strcmp_P' was not declared)

Dist0rtion, 2020-05-25:
Hey,..
I dont know why, but i get an Error with the current Bugfix 2.0.x Branch if i want to compile it.

**Error Log**
```
queue.cpp (Marlin\src\gcode)*
'strcmp_P' was not declared in this scope; did you mean 'strcmp'? [526, 15]
'strcmp_P' was not declared in this scope; did you mean 'strcmp'? [530, 15]
'strcmp_P' was not declared in this scope; did you mean 'strcmp'? [531, 15]
```

**Output after Compile**

```
Marlin\src\gcode\queue.cpp: In static member function 'static void GCodeQueue::get_serial_commands()':
Marlin\src\gcode\queue.cpp:526:15: error: 'strcmp_P' was not declared in this scope; did you mean 'strcmp'?
  526 |           if (strcmp_P(command, PSTR("M108")) == 0) {
      |               ^~~~~~~~
      |               strcmp
Marlin\src\gcode\queue.cpp:530:15: error: 'strcmp_P' was not declared in this scope; did you mean 'strcmp'?
  530 |           if (strcmp_P(command, PSTR("M112")) == 0) kill(M112_KILL_STR, nullptr, true);
      |               ^~~~~~~~
      |               strcmp
Marlin\src\gcode\queue.cpp:531:15: error: 'strcmp_P' was not declared in this scope; did you mean 'strcmp'?
  531 |           if (strcmp_P(command, PSTR("M410")) == 0) quickstop_stepper();
      |               ^~~~~~~~
      |               strcmp
Compiling .pio\build\LPC1769\src\src\gcode\sd\M24_M25.cpp.o
*** [.pio\build\LPC1769\src\src\gcode\queue.cpp.o] Error 1
```

I tried with "Stock" Configs (for Ender 3), just changed the Board in the Configfile, no other Features, but im getting the same Error.

If I change 'strcmp_P' to 'strcmp', it compiles without problems. I guess it's not supposed to be like that?
I use Visual Studio code with PlatformIO, am I missing a new library or something?

Thanks in advance

ellensp, 2020-05-25:
add the following to end of src/HAL/LPC1768/HAL.h 
```
// Add strcmp_P if missing
#ifndef strcmp_P 
  #define strcmp_P(a, b) strcmp((a), (b))
#endif
```

ellensp, 2020-05-26:
has been added to bugfix, please close issue

Dist0rtion, 2020-05-26:
> 
> 
> has been added to bugfix, please close issue

**Closed**

closed 2020-05-26 by Dist0rtion