MarlinFirmware/Marlin #18104: G38 not compatible with Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

Korbinbrandon, 2020-05-25:
### Bug Description
If the probe is connected to the Z-Min pin with #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
G38.2 Z0 will not stop until Z0 is reached even if the probe and or the Z-min endstop gets hit.

### My Configurations

[Marlin config.zip](https://github.com/MarlinFirmware/Marlin/files/4676700/Marlin.config.zip)

### Steps to Reproduce

1. G28 X Y Z; Home printer
2. G0 Z100; Get away from Z0 by a comfortable distance
3. G38.2 Z0; Touch Probe or Z-min endstop with finger. Alternatively use a piece of soft foam as probe target.

**Expected behavior:** Movement stops as soon as the probe or the Z-min endstop gets hit.

**Actual behavior:** It keeps going until Z0 is reached

#### Additional Information

* I rule out a hardware problem as I can use my Probe just fine as Z-endstop with G28.
* Furthermore if I hook my probe up on Pin 30 instead with Z_MIN_PROBE_PIN 30 
G38 works as expected.

ruggb, 2020-05-26:
you defined the probe as the Z endstop.
It is a fix mounted probe
Z=0 is not offset from the probe.
Where is Z=0 and how does the printer know to stop there on a G38.2?
Does it display and error when failed?
Does G28 work as expected?

Korbinbrandon, 2020-05-27:
> you defined the probe as the Z endstop.

Right, as described in https://marlinfw.org/docs/configuration/probes.html
* If the probe is connected to the Z-Min pin, enable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.

> It is a fix mounted probe
> Z=0 is not offset from the probe.

Correct on both.

> Where is Z=0

Z=0 gets determined with G28. So its when either the probe or the Z endstop gets triggered. Unless the probe is placed outside of the bed, the probe will be triggered first. So Z=0 is when the probe touches the bed.

>and how does the printer know to stop there on a G38.2?

Z=0 gets specified as the target position of G38.2, as described in https://marlinfw.org/docs/gcode/G038.html
* G38.2 probes towards a target and stops on contact, signaling an error if it reaches the target position without triggering the Z endstop.

> Does it display and error when failed?

Of course. Since the probe doesn't register a hit on anything, G38.2 Z0 will reach Z0 and "Failed to reach target" will be sent. This error message is misleading by the way, it means that the target destination was reached without triggering the probe.

> Does G28 work as expected?

Yes

ruggb, 2020-05-27:
I have never tried G38.2. Maybe I should. I am still trying to figure out what the use is for G38.2. It appears that it is just testing your endstop position since "the Z endstop doubles as the probe for these commands". 
Since you have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled you are physically paralleling your endstop with the probe, correct?
And since the probe gets triggered first, the endstop is never used.
The error message is not misleading. It is telling you that Z0 was reached b4 the endstop (probe) was hit. Z0 in this case is not physical Z0. It is what Marlin has set in software as Z0 which might include an offset or hysteresis.
Moving the probe to pin 30 allows the endstop to function as the endstop and the probe to function as the probe. Since it works then, I suspect a hardware issue. It could be that G28 works because BOTH probe and endstop are triggered. Your testing indicated you are trying to trigger your probe OR your endstop but not both together. Try using a block of wood that will trigger both.
If that works, the circuit of the probe and endstop are in competition. Using Pin 30 is probably the easy fix.

Korbinbrandon, 2020-05-27:
>I am still trying to figure out what the use is for G38.2.

It allows direct usage of the probe. In my case I use it to scan the geomety of objects. For each point in a raster I do G38.2 Z0 followed by M114. I monitor my serial port and build a point cloud from that.

> Since you have Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN enabled you are physically paralleling your endstop with the probe, correct?

Yes, I am physically paralleling two normaly open endstops. Have a picture for clarity: 
[image]

> And since the probe gets triggered first, the endstop is never used.

Correct, when the probe is mounted you would have to ram it trough the bed by ~10mm in order to trigger the endstop.

> The error message is not misleading. It is telling you that Z0 was reached b4 the endstop (probe) was hit. Z0 in this case is not physical Z0. It is what Marlin has set in software as Z0 which might include an offset or hysteresis.

"Failed to reach target" is misleading imo since it actually did manage to reach its target position.
"Failed to find target object" would be more appropriate.

> Moving the probe to pin 30 allows the endstop to function as the endstop and the probe to function as the probe. Since it works then, I suspect a hardware issue. It could be that G28 works because [...]

There is no physical way G28 can trigger BOTH probe and endstop at the same time in my setup. Furthermore If I hit my probe OR my endstop with my finger during a G28 it stops as expected.
It works on pin 30 because #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN is commented out and #define Z_MIN_PROBE_PIN 30 is commented in instead.

> Using Pin 30 is probably the easy fix.

Using Pin 30 is a workaround not a solution. The next person comming around will ram their probe into the bed aswell.

ruggb, 2020-05-27:
"Failed to reach target" is misleading imo since it actually did manage to reach its target position.
"Failed to find target object" would be more appropriate.

Since the probe did not trigger, it did not reach the target b4 it reached Z0 (software).
More semantics than misleading.
Circuit wise, it should work. Try disconnecting the endstop.

Korbinbrandon, 2020-05-27:
> Circuit wise, it should work. Try disconnecting the endstop and test with finger

Done that. The problem presists.

Would be nice to know if it can be reproduced. You wouldn't even need a probe for testing as you can just touch your Z-endstop with your finger instead.

ruggb, 2020-05-27:
it is now a mystery. G28, G38 and G9 work if probe is connected to another pin and endstop connected to Z_MIN_PIN
Only G28 works with probe and endstop connected to same pin.
G38, nor G29 work with just probe connected to endstop pin, but G28 does either with one or both connected.
Any other clues? Or are any of above incorrect?

Korbinbrandon, 2020-05-28:
Well I never tried G29 but it makes sense that it wouldn't work either.
I don't realy know my way around this firmware but it seems that Z_MIN_PROBE_PIN is -1 unless it is set via #define Z_MIN_PROBE_PIN. So with #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN maybe it stays at -1?

ruggb, 2020-05-28:
-1 = unassigned
Z_MIN_ENDSTOP pin is default assigned based on the board so setting the probe to use it makes your probe the endstop also.

ruggb, 2020-05-28:
THIS IS A BUG folks.........................
I tested G38.2 on my machine. It is a CoreXY with a nozzle mounted piezo probe. It is the nozzle and not separate from it.
G28 > works fine
G29 > works fine
G38.2 Z0 > goes to Z0 even if endstop is tapped just like yours.
I believe it is a BUG because the endstops are not enabled. DUH!
Since it works when you connect your probe to another pin that would be a good indication.
G38 is not looking for physical Z0. It is looking for software Z0.
If the probe is the endstop, it doesn't work because the endstop is not enabled.
It also does not work if I set endstops always enabled == which may be another bug unless G38 is actually turning the Z endstop off.

boelle, 2020-06-21:
@Korbinbrandon still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle