MarlinFirmware/Marlin #18105: [FR] or [BUG] Auto-reset BLTouch

asko58, 2020-05-25:
Dear community,

since I've been having minor issues with my installed BLTouch (clone wired to SKR 1.3 board, TMC2209 drivers) for months I was wondering if a certain feature should not be implemented in Marlin.

Issues:
Sometimes auto bed levelling at the start of a print fails with BLTouch going in to error mode (blinking). (wiring triple checked, using shielded CAT5 cable, must be electrical noise?!) Mostly the levelling data is still ok, since it levels before each print, so even if it fails once the last set of data will be pretty OK. What is a problem, however, is that the probe's pin stays down, catching on edges of the printed plastic. A manual reset of the BLTouch via LCD menu makes it reset properly and pulls the pin up again.

Possible resolution/feature request:

A Marlin feature that auto-resets the probe on error. Or is that already implemented and I didn't find it? Obviously Marlin gets the information that probing has failed but does not reset the probe.

qwewer0, 2020-05-25:
You could place an `M280 P0 S90 ; Push-pin Up (Stow)` after your bed leveling in the start gcode.

asko58, 2020-05-25:
Thanks for your reply. As a temporary workaround that could do the trick, but it does not take care of a possible BLTouch failure during print (that could occur).

qwewer0, 2020-05-25:
Yes, it is just a temporary workaround.
For the the alarm release you could use this instead: `M280 P0 S160 ; Alarm Release & Push-pin UP`

psavva, 2020-05-28:
Have you installed a separate 5v line onto the BLTouch.  I had the issue with the pin dropping years ago, and installed a separate BEC (Battery Elimination Circuit) to fix the issue...

Grogyan, 2020-05-28:
The SKR1.3 has a known hardware bug, where the 5V linear regulator drops to ~4.8V

This just due to inherent excessive current drain on many parts of the circuit, The SKR1.4 fixes this by having a 2.5A 5V regulator.

MIchaeldm3, 2020-05-29:
Having a similar issue with SKR mini e3 v1.2 + bl touch v3.1

sjasonsmith, 2020-06-21:
@asko58, when probing fails Marlin doesn't really know why. In all likelihood a physical crash has occurred and any further action is risky. Resetting the error state could in many cases cause damage to the machine.

Is your printer actually proceeding with the print after probing fails? I thought that that would cause Marlin to die to prevent further damage.

asko58, 2020-07-19:
> Is your printer actually proceeding with the print after probing fails? I thought that that would cause Marlin to die to prevent further damage.

Indeed, the printer goes on as if nothing has happened. The Z height will be incorrect but the print proceeds.

asko58, 2020-07-19:
> @asko58, when probing fails Marlin doesn't really know why. In all likelihood a physical crash has occurred and any further action is risky. Resetting the error state could in many cases cause damage to the machine.

I understand your point, but I am not talking about resetting a Marlin error condition (there is none btw) but rather resetting the probe's own error state.