MarlinFirmware/Marlin #18106: Sensorless homing in Z with TMC2209 at Z2 no reaction

Bastler1971, 2020-05-25:
Tried around with 2.0.5.3 and also with Bugfix2.0.
Dual Z runs, only sensorless homing works on X / Y / Z, only with Z2 no reaction. 
If Z_MULTI_ENDSTOPS is activated, Z unfortunately continues to run until it triggers, but should react in exactly the same way as without Z_MULTI_ENDSTOPS on Z. So stop both at the same time!
Why do I need it? Because of the gallows whose leverage when X10 and X290 are touched, there are different Z values because of leverage.

ellensp, 2020-05-25:
No configs, hasn't followed template. Seems like a support request vs a bug. Cannot help.

Bastler1971, 2020-05-25:
here my configuration
[config.zip](https://github.com/MarlinFirmware/Marlin/files/4677594/config.zip)

ellensp, 2020-05-25:
You have #define EXTRUDERS 1 which is good, it will use E0 for the extruder and E1 as Z2
E1 must use one particular IO pin for DIAG in this case 
`#define E1_DIAG_PIN                        P1_26  // Y+ `

But in your Configuration_adv.h you have
`#define Z2_USE_ENDSTOP          _XMAX_

It needs to be

`#define Z2_USE_ENDSTOP          _YMAX_
`

Bastler1971, 2020-05-25:
Thanks for the answer
Unfortunately, it doesn't work whether I enter _YMAX_ or _ZMIN_

#define NUM_Z_STEPPER_DRIVERS 2   // (1-4) Z options change based on how many

#if NUM_Z_STEPPER_DRIVERS > 2 
  #define Z_MULTI_ENDSTOPS
  #if ENABLED(Z_MULTI_ENDSTOPS)
    #define Z2_USE_ENDSTOP          _ZMIN_ 
    #define Z2_ENDSTOP_ADJUSTMENT   0

ellensp, 2020-05-25:
are the _ chars around the YMAX or did the editor just eat them...

Bastler1971, 2020-05-25:
The _ has eaten browsers!
My jumpers are set for X- / Y- / Z- / X + / Y +, is that wrong?

ellensp, 2020-05-25:
that is fine. X+ isn't needed but doesn't matter  

another issue in Configuration.h
`#define Z2_DRIVER_TYPE TMC2209 //HTG TMC Dual Z2`
You don't use this when using E1 as Z2
Use this instead
`#define E1_DRIVER_TYPE TMC2209`

Bastler1971, 2020-05-25:
I changed and tried with _YMAX_ and _ZMIN_, does not work. Now E0 is missing in the TMC Hyprid threshold settings, before that it was there

Bastler1971, 2020-05-25:
have reset it, then extruder is back, only not as E0 but as E1 and Z2 is also in the settings again

#define Z2_DRIVER_TYPE TMC2209

#define E0_DRIVER_TYPE TMC2209

Bastler1971, 2020-05-26:
> that is fine. X+ isn't needed but doesn't matter
> 
> another issue in Configuration.h
> `#define Z2_DRIVER_TYPE TMC2209 //HTG TMC Dual Z2`
> You don't use this when using E1 as Z2
> Use this instead
> `#define E1_DRIVER_TYPE TMC2209`

If
#define Z2_DRIVER_TYPE TMC2209
// # define E1_DRIVER_TYPE TMC2209

then everything displays correctly under M122

Bastler1971, 2020-05-26:
Ich habe irtümlich 
#if NUM_Z_STEPPER_DRIVERS > 1 auf 2 geändert, das kann ja nicht funktionieren.

Danach ich 
#if NUM_Z_STEPPER_DRIVERS > 1
richtigen Wert eingetragen habe, kommt beim kompellieren immer Fehler

Egal ob ich _YMAX_ oder anderes eintrage und 
#define Z2_DRIVER_TYPE TMC2209
oder
#define E0_DRIVER_TYPE TMC2209 
nehme

Bastler1971, 2020-05-26:
I have irtually
#if NUM_Z_STEPPER_DRIVERS> 1 changed to 2, that can't work.

After that i
#if NUM_Z_STEPPER_DRIVERS> 1
entered the correct value, there is always an error when compiling

No matter whether I enter _YMAX_ or something else
#define Z2_DRIVER_TYPE TMC2209
or
#define E0_DRIVER_TYPE TMC2209
take

boelle, 2020-06-29:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-29 by boelle