MarlinFirmware/Marlin #18107: [FR] (SKR PRO V1.1) X/Y axis minimum speed is limited

frank-pian, 2020-05-25:
When I used the G0 or G1 command to control the X/Y axis movement, there was no significant speed change for the F parameter below 500 or so, but the Z axis was ok

If you have the same board, can you test whether 'G0 F50' and 'G0 F500' have different speeds

ellensp, 2020-05-26:
Please read https://github.com/MarlinFirmware/Marlin/issues/14694 
Ask in the discord and please close this question.

frank-pian, 2020-05-26:
> 
> 
> Please read #14694
> Ask in the discord and please close this question.

sorry，i fixed my question.
I think it may be related to this part of the code, but I don't know how to modify it.
stepper.h
```
#ifdef CPU_32_BIT
  /**
   * Duration of START_TIMED_PULSE
   *
   * ...as measured on an LPC1768 with a scope and converted to cycles.
   * Not applicable to other 32-bit processors, but as long as others
   * take longer, pulses will be longer. For example the SKR Pro
   * (stm32f407zgt6) requires ~60 cyles.
   */
  #define TIMER_READ_ADD_AND_STORE_CYCLES 34UL

  // The base ISR takes 792 cycles
  #define ISR_BASE_CYCLES  792UL

  // Linear advance base time is 64 cycles
  #if ENABLED(LIN_ADVANCE)
    #define ISR_LA_BASE_CYCLES 64UL
  #else
    #define ISR_LA_BASE_CYCLES 0UL
  #endif

  // S curve interpolation adds 40 cycles
  #if ENABLED(S_CURVE_ACCELERATION)
    #define ISR_S_CURVE_CYCLES 40UL
  #else
    #define ISR_S_CURVE_CYCLES 0UL
  #endif

  // Stepper Loop base cycles
  #define ISR_LOOP_BASE_CYCLES 4UL

  // To start the step pulse, in the worst case takes
  #define ISR_START_STEPPER_CYCLES 13UL

  // And each stepper (start + stop pulse) takes in worst case
  #define ISR_STEPPER_CYCLES 16UL
```

sjasonsmith, 2020-05-26:
> I think it may be related to this part of the code

Why do you think this code in particular would be part of the problem?

Those numbers are used to predict a maximum stepper rate, but I don’t think they contribute to a minimum speed.

p3p, 2020-05-26:
Minimum speed, as far as I remember, is dictated by the resolution and frequency of the timers used for stepping, not sure how this board is configured but if, for example, you have a 16bit timer running at 1MHz it cant trigger a step less than once every 65ms. Marlin may have added a safegaurd minimum since then though?

sjasonsmith, 2020-05-26:
It is definitely using a 16 bit timer. Only timers 2 and 5 are 32-bit. Those are currently used for other functions in the variant.h file, but could probably be switched around. To give a higher resolution step timer.

frank-pian, 2020-05-27:
> Minimum speed, as far as I remember, is dictated by the resolution and frequency of the timers used for stepping, not sure how this board is configured but if, for example, you have a 16bit timer [...]

thanks for your help! I think you are right. But I haven't found the real reason yet.

frank-pian, 2020-05-27:
> 
> 
> It is definitely using a 16 bit timer. Only timers 2 and 5 are 32-bit. Those are currently used for other functions in the variant.h file, but could probably be switched around. To give a higher resolution step timer.

If it is caused by the timer, why does the z-axis work normally? In addition, if it is not the delta machine, the X/Y axis speed is normal.

frank-pian, 2020-06-05:
#14532

boelle, 2020-06-21:
@frank-pian still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle