MarlinFirmware/Marlin #18110: [BUG] BLTouch does not stop Z axis if endstop is connected (Bugfix-2.0.x)

Exalik, 2020-05-25:
### Bug Description

The BLTouch does not stop the Z axis stepper motor on homing if Z endstop is connected.

### My Configurations

[Marlin Bugfix-2.0.x Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4678237/Marlin.Bugfix-2.0.x.Configuration.zip)

### Steps to Reproduce

1. Launch Homing XYZ
2. Look the z stepper motor to push on the bed
3. Turn off power

**Expected behavior:** The z stepper motor should stop

**Actual behavior:** The z stepper motor does not stop

#### Additional Information

ruggb, 2020-05-28:
I think u need one of these enabled - probably the first one and then you need to define your offsets

/**
 * A Fix-Mounted Probe either doesn't deploy or needs manual deployment.
 *   (e.g., an inductive probe or a nozzle-based probe-switch.)
 */
//#define FIX_MOUNTED_PROBE

/**
 * Use the nozzle as the probe, as with a conductive
 * nozzle system or a piezo-electric smart effector.
 */
//#define NOZZLE_AS_PROBE

offsets
//#define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 } //defined else where

swoopyfpv, 2020-06-06:
^^ Did the mentioned part work?

i have the same problem with a clone (3d touch).
The probe works fine from the LCD Menu (deploy/stow/self test)

If i go to auto bed leveling the printer will do bed and x. Then go to the middle, release the pin from the bltouch, but if the pin get's pulled up it will re release the pin without stopping the y axes from moving... i then need to cut the power to prevent the printer crashing into the bed

ruggb, 2020-06-06:
well, without your config files you can expect to trash it out by yourself.

swoopyfpv, 2020-06-06:
> well, without your config files you can expect to trash it out by yourself.

thanks for the quick answer!

sorry for the undetailed info ... thought this might be a beginners mistake
I'll attach the config.h as a text file.

thats what i changed from the default cr10s files (including some settings for the bmg extruder, which works fine)

-----------------

- deactivated endstops & 
#define Z_MIN_PROBE_ENDSTOP_INVERTING false 
(also tried on true and false, didn't change anything)

- deactivated #define PROBE_MANUALLY
- activated #define BLTOUCH (including the requirements BABYSTEPPING, BABYSTEP_ZPROBE_OFFSET, Z_SAFE_HOMING)
- set offset (only X,Y)
- deactivated #define MIN_SOFTWARE_ENDSTOP_Z

as the deploy/stow function is working i can assume the problem is either with the bltouch, or software. this is the second touch sensor, i crashed the first sensor on the bed because of the same problem... so i assume it's a software problem. also tried the marlin 1.1.9 ... same problem

if i reflash to manual and connect the normal switch z-endstop it's working fine.

thanks in advance

[configuration.h-bl-touch-textfile.txt](https://github.com/MarlinFirmware/Marlin/files/4741089/configuration.h-bl-touch-textfile.txt)

ruggb, 2020-06-06:
given that most people are not having an issue with this in 1.1.9 or 2.0 and that the setup of the config files is not will defined RE probes, I would assume your config is incorrect.
1st=I will assume you are doing all this testing after you do a G28, correct?
since this
//#define NO_MOTION_BEFORE_HOMING // Inhibit movement until all axes have been homed
is undefined. If not your system has no way of knowing where Z0 is.

>>deactivated endstops &
>>#define Z_MIN_PROBE_ENDSTOP_INVERTING false
>>(also tried on true and false, didn't change anything)

this is not defined true or false so I don't know what you are telling me.
//#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.

all the babystepping stuff is in the adv file that you did not include

>>set offset (only X,Y)
Your Touch probe has a Z offset. Without this, you cannot find Z0 without another endstop.
Since you have this enabled
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
I'm thinking you are trying to use the probe as an endstop but without enough info for Marlin to figure out where Z0 is.

>>deactivated #define MIN_SOFTWARE_ENDSTOP_Z
not sure why you want to do that, but mine are default so I can't relate, but I am guessing it may have something to do with your issue causing the bed crash. Endstops are ONLY enabled when you are doing a G28 (homing) unless you set endstops to be enabled all the time in ADV
//#define ENDSTOPS_ALWAYS_ON_DEFAULT
With this you are taking away Marlins knowledge of where Z0 is. But with everything else it may not know anyway.

I would advise you to get it working without the probe. Then add the probe as a probe only. THEN, if this is what you want, change things to make the probe function as an endstop also.

swoopyfpv, 2020-06-07:
thanks for the detailed info.

i'm assuming g28 is the same as "auto home" in marlin, right?!

auto homing isn't successful because the bltouch activation isn't recognized, so the printhead would crash into the bed... i than have to cut power.
thats because all the info/tutorials(cr10s bltouch) i looked up, installed the bltouch on the z endstop so the oem endstop's plug is used by the bltouch.

ruggb, 2020-06-07:
G28 is home all, G28 Z would be home Z only
If BL is connected to Z min endstop pin it should activate, but it is not at the exact Z as the nozzle. That is why you need a Z offset. It must activate b4 the nozzle hits the bed so Z0 is actually going to be positive. The offset then ( I think) will be negative. But I keep getting confused by this so proceed with caution.
There is a command to test pins & endstops - M43 & M119 = use them
Take it one step at a time.

swoopyfpv, 2020-06-07:
> G28 is home all, G28 Z would be home Z only
> If BL is connected to Z min endstop pin it should activate, but it is not at the exact Z as the nozzle. That is why you need a Z offset. It must activate [...]

thanks again!

checked everything, endstop wasn't recognized, needed to swap the enstop pins. now it works fine.

sjasonsmith, 2020-06-20:
@Exalik are you still having issues that you think are a bug? It seems all the recent activity involved issues from @swoopyfpv and his issues are now resolved.

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle