MarlinFirmware/Marlin #18111: [BUG] SKR Pro 1.1 w/2209s sensorless X/Y - M119 not reporting Z-endstop triggering

rhukster, 2020-05-25:
### Bug Description

I'm building a custom CoreXY with an SKR Pro 1.1. 

This issue is that I have X/Y  configured with sensorless homing  with 2209(v1.2) drivers with diag pin intact, and my Z0, Z2 (E1) and Z3 (E2) drivers, also 2209s with diag pins de-soldered.  The endstop LEDs next to the motor connectors on the board are completely off for X/Y/E0, and are dim for Z1, Z2, Z3.  

When i trigger my mechanical OMRON endstop by hand (which is connected to Z_MIN endstop on the board), the Z1 board LED turns off.  So it seems the board itself is registering the endstop being triggered. However, while doing this, the `M119` command always reports the end stop is `open`:

```
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open // not activated by hand (z1 LED light on)
Recv: ok
[...]
Send: M119
Recv: Reporting endstop status
Recv: x_min: open
Recv: y_min: open
Recv: z_min: open // activated by hand (z1 LED light off)
Recv: ok
```

### Diagnostic Attempts

I've tried pulling all the drivers, and also removing all stepper cables, so only the 1 single Z-endstop cable is attached, and the problem still remains.

I've also tried using 2209s with the diag pin intact, but these don't even activate the board endstop LEDs at all.

I've also tried reverting all my triple-z + probe changes and keeping the bare minimum SKR Pro 1.1 and 2209 driver configs intact, but that didn't seem to make any difference either.

I'm using the very latest Marlin `2.x-bugfix` branch that I merge daily so that is 100% fully up to date as of right now.

BTW, sensorless homing on X and Y is working just fine with `G28 X Y`, no issues there at all.

### My Configurations

[Configuration-files.zip](https://github.com/MarlinFirmware/Marlin/files/4678213/Configuration-files.zip)

### Steps to Reproduce

Setup an SKR Pro 1.1 with configred for X/Y/E/Z/Z2/Z3 with 2209s, using latest `bugfix-2.x` and connect a mechanical endstop to Z_MIN connector.  Try toggling the endstop manually.

**Expected behavior:**

* Z endstop led will toggle on and off 
* `M119` will report `open` or `TRIGGERED`

**Actual behavior:**

* Z endstop led will toggle on and off 
* `M119` always reports `open`

#### Additional Information

n/a

JAdams12, 2020-05-25:
I had this same issue on my SKR 1.4, ended up being that the endstop ports have pull up resistors or something that was causing my Z probe to end up in either always triggered or always open, depending on whether i reversed the logic or not. 

It might be worth using this as your endstop pullup settings on config.h 

`// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  //#define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  //#define ENDSTOPPULLUP_ZMIN
  //#define ENDSTOPPULLUP_ZMIN_PROBE
#endif`

That way your sensorless homing X and Y have the required pullups but your probe then may work. 

If that doesn't work (it didn't for me on the 1.4), they maybe theres another port you can use on your board for your Z endstop and just change it so you use probe for homing and change it to the correct pin under the homing section.

rhukster, 2020-05-26:
@JAdams12 i tried your suggestion for manually setting the ENDSTOPPULLUP (although I have to use XMIN/YMIN for sensorless homing), but it made no difference.

JAdams12, 2020-05-26:
@rhukster Im guessing that your Z min plug has a resistor always on then like my 1.4 turbo does. 

You might need to find an open pin that you can use for the signal part, in the pins.h for your board try to find one that you won’t use for something else and uncomment it and replace that pin (should be P followed by numbers) with your Z Min one, then plug the signal part of your probe into that new pin on the board. 

For example on mine I had to plug it into the probe pin spot, so I went into pins.h for my board and replaced the Z_min pin with the one from the probe input, that way the firmware things it’s using Z min, but the signal is just being read from a different port. 

I tried to look at the schematic of the 1.1 pro but I couldn’t find anything that was like immediately apparent as being an option. Maybe one of the AUX pins has an input you can steal?

rhukster, 2020-05-26:
Well, I got it sorted without any special configuration, and was totally a "not seeing the trees for the forest" kind of thing. 

I didn't pay attention to the pins on the endstop connector on the board and I thought that my mechanical OMRON endstop was wired correctly from the factory with V+ and Signal with ground in the middle (missing), but as it's only two wire type, it was Ground + Signal, so I moved the Ground pin to the middle of the connector and it just started magically working.

I also had wired my EZABL pro wire the same as the OMRON cable, so wired them both wrong! Sorry to waste your time!

closed 2020-05-26 by rhukster