MarlinFirmware/Marlin #18113: [BUG] [bugfix-2.0.x] [SWITCHING_NOZZLE ] Continuously changing the nozzle causes the height to increase

scott0122, 2020-05-25:
### Bug Description
Continuously changing the nozzle causes the height to increase

https://www.youtube.com/watch?v=OMkm0p_0DBQ

### My Configurations
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4678929/Configuration.zip)

coolcheat, 2020-05-26:
I can confirm this happens to me as well on latest bugfix-2.0.x

I don't have a mechanical switch like @scott0122 showed, but am using a single nozzle with dual extruders.

I suspect this has to do with SWITCHING_TOOLHEAD_Z_HOP, as marlin reports the updated position (+2 in my case).

Thanks.

c3D-Dan, 2020-05-28:
I have a similar hotend.  Doesnt happen for me. 2.0.5 bugfix.

sjasonsmith, 2020-06-20:
@scott0122 or @coolcheat, if this is still an issue, can you provide updated configuration files for the latest bugfix-2.0.x, and provide exact steps to reproduce/observe the problem?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

scott0122, 2020-06-30:
@sjasonsmith 
Configuration files are in my first post. 
Power on -> Switch nozzle(T0, T1, T0, T1...) -> z height continuously increased. 
Power on -> Auto Home(G28) -> Switch nozzle(T0, T1, T0, T1...) -> z height continuously increased.

closed 2020-08-05 by github-actions[bot]