MarlinFirmware/Marlin #18114: [BUG] X and Y don’t home

Luke3924, 2020-05-26:
Hello,

I recently purchased a Bigtreetech Skr 1.3 and a Bltouch sensor. I have everything installed and programmed in Marlin to the best of my ability. 

Now, the X and Y axis move correctly with manual controls and the sensors seam to be correct when I checked it with a M119. However, when I try to home the machine, it seams to skip the X and Y homing and moves directly to the Z homing.

Any help would be much appreciated!

swilkens, 2020-05-26:
Please post your configuration files, as the template asks of you.

After a firmware flash, try to 'M502' followed by a 'M500' to store default values to the eeprom. Then Try again.

Tannoo, 2020-05-26:
"[BUG]" should be changed to "[HELP]".

Luke3924, 2020-05-26:
Thanks for the advise. Sorry I didn’t know [Help] was a thing since I was given the option buggfix or feature request. 

I already did the M502 and M500 before I made this help request. It got rid of the eeprom version error but then I got the index error. 

I will post my configuration files first chance I get. I’m on my iPhone at the moment so I couldn’t attach the file.

Finally, sorry for closing and reopening the issue. I’m very new to github and I clicked the wrong button.

Thanks again for the help!

Luke3924, 2020-05-26:
So, I just reuploaded my marlin code and, magically, it just works. I did fix the index issue by initializing the eeprom and I had to switch an axis direction and it works now. 

However I do have one new issue. When it homes the Z axis it only drives one of the two on the downward stroke of the homing stepper motors. What I mean is both work when it raises the Z to 10mm before it probes, but when it lowers the probe to the surface it only does Z and not Z2.

For reference, I set the E1 port to Z2 in order to get dual Z axis.

closed 2020-05-26 by Luke3924