MarlinFirmware/Marlin #18117: [BUG] SKR Mini stop mid print

adams79, 2020-05-26:
### Bug Description

After replacing my mp select mini board with an e3 mini skr board sometimes during a print the board completely freeze with heater on and fan on. Tried both via SD and octoprint.
Replaced the board, same problem. Additionally when I run my Z-axis motor in StealthChop I've a sort of "constant" step skipping. Tried raising current but no changes. How can I track the source of problem? I'm going to change PSU as well, but I'm running out of ideas. The driver monitoring is disabled. After replacing the board I was able to print for about 5/6 hours without problems but then the problem resurfaced. Installed latest build available

Thank you

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

Start a printing from sd tr octoprint

**Expected behavior:**

The print should complete

**Actual behavior:** [What actually happens]

The print stops randomly
[Archivio.zip](https://github.com/MarlinFirmware/Marlin/files/4680502/Archivio.zip)

swilkens, 2020-05-26:
You have:

> #define DISABLE_X true
> #define DISABLE_Y true
> #define DISABLE_Z true

This disables the stepper motors when it's not in active use, this can cause some accuracy issues (step skipping?).

You are also not using the most recent bugfix, try the most recent - including the latest configuration files.

I also notice you are using this board for a Malyan M200 - I assume you found these motor currents for the Malyan M200 in the original firmware?

adams79, 2020-05-26:
Hi swilkens,
ok I'm going to test with latest bugfix and config. yes the currents are from the original Maylan board. 
Also I will try to not disable the motors, however the step skips on Z-axis is not related to this. Once I activate stealth chop on this axis seems that there is a "fixed" amount of skips since object are always lower of about 20%. The weird thing is that by moving the z-axis with direct command the movement is right, but during print seems to skip about 20% of steps... I'm thinking of switching the z-axis motor

adams79, 2020-05-26:
Ok, tried with latest build and config, stops after about 30 minute... trying to enable SD logging to check if some error is logged in the console...

kain0m, 2020-05-30:
Check this one out:
https://github.com/MarlinFirmware/Marlin/issues/15337
It is related to SKR 1.3 (not mini), but the issue is the same. Random freeze of the print, heaters on.

My solution was a different SD card.
Also, try reformatting the SD card(s) with FAT32.
I have also read that the supplied Bigtree tech Micro SD card is poor quality in some cases.

adams79, 2020-05-31:
This seems to not be my problem since it happens also by printing from octoprint without sd inserted

swilkens, 2020-06-01:
Since the Malyan M200 by default does not have a fan in the electronics case, could it be that the drivers are overheating? In that case I would expect one driver to stop, rather than all of them.

Try to enable TMC driver monitoring and see if the driver reports an overtemp when it stalls. Do you still have serial communication when this happens or does the board fully lock up?

Which environment are you using the build the firmware? 256K? USB composite?

cooldudie2, 2020-06-01:
i can confirm the exact same errors, as well as a lot of MP SELECT mini users using the Skr mini e3 1.2 mod i uploaded to thingiverse. (cooldudie2)  all the users i have been in contact with have experienced the same freeze up issue using the firmware i have written, as well as their own compiled firmware based on bugfix 2.0.x branch and newest marlin 2.0.5.3 branch. 

After replacing my mp select mini board with an e3 mini skr board sometimes during a print the board completely freeze with heater on and fan on. 

Tried both via SD and octoprint.
Replaced the board, same problem i have tried different suppliers and same problem occurs. 

someitmes it freezes sometimes it finishes prints just fine.

note the jumper cap is removed from the board as its set  in firmware and not required when using uart mode some people have not taken the cap off by default but confirm the problem still persists.

i have tried a powerful and more reliable 12v 750watts ATX monitored PSU and noticed no change in voltage drop when print fails, power fluctuation of max 0.2 volts either side of 12volts when in use. 

* i can confirm the PSU is not at fault.  problem still persists.

*Driver monitoring doesn't report any overheat prewarn triggers.

*linear advance and hybrid theshold are turned off to rule those out completely.

*i have also tried setting the extruder to spreadcycle mode to rule out the problem where stealthchop triggers a lock up when moving to fast in one direction and then the other as suggested by others on the forum.

*256k version and 512k version show the same fault regardless of environment used. 

*board fully locks up and doesnt respond to usb commands via pronterface or from the malyan m200 screen attached. 

 i will report back if the problem re occurs after some more prints. 

 

Steps to Reproduce
Start a printing from sd tr octoprint

Expected behavior:

The print should complete

Actual behavior: [What actually happens]

The print stops randomly

cooldudie2, 2020-06-01:
i have also just considered that the screen is drawing 3.3v logic power from the swd header on the board for power, i wondering if this could be causing the issue of lockup feature. whats the swd header for and if you draw power from it could it effect the other electronics

adams79, 2020-06-02:
I’ve currently swapped the z-axis motor with a Nema 17 however I think that this will not solve the problem. Also I have a spare skr 1.4 turbo (with tmc2209) that I bought for another printer, I can make a try with this board too

adams79, 2020-06-02:
> Since the Malyan M200 by default does not have a fan in the electronics case, could it be that the drivers are overheating? In that case I would expect one driver to stop, rather than all of them. [...]

I have a v2 that has a case fan (that I’ve also replaced with a noctua)... tried to print without side panel and the heat sink on drivers are cool

minosg, 2020-06-03:
From a similar thread investigating the same issue on another project.

https://github.com/KevinOConnor/klipper/issues/196

This correspondence with TMC is being attached which explains the lock-up issue

>  I think, the reason for disabling of the TMC2208 driver could be
> 
>     a hard stop of the motor in stealthChop (Step frequency goes from a higher value, e.g. > 0.5 RPS to 0)
> 
>     an abrupt change of motor velocity (Step frequency goes from a higher value to a low value within a single step).
> 
> When using stealthChop, please always make sure, that you use velocity ramping. A hard stop will cut away motor back-EMF at once. As stealthChop is a voltage based chopper, it cannot respond to this at once, like spreadCycle. The result is an overcurrent, and the motor driver goes to overcurrent switch off, until it becomes disabled / enabled again.
> 
> To resolve the problem, please use at least a tiny velocity ramping, when hard stopping the motor, e.g. within a few / a few ten microsteps.

To my understanding this issue could be addressed by:
- velocity ramping. ( Not certain how Linear advance will respect that )
- Setting the hybrid threshold to a value, that when the extrude changes direction suddenly it falls to spread cycle.
- Playing with the MINIMUM_STEPPER_POST_DIR_DELAY/MINIMUM_STEPPER_PRE_DIR_DELAY and overriding the driver default of 20nS

Does marlin have a velocity ramping configuration parameter?

adams79, 2020-06-03:
Confirmed that the problem still persists after swapping the Z-axis motor with a standard Nema 17. Since other SKR mini users are not experiencing this it should be related to same specific component of this printer (this is the reason for which I've swapped the motor). I'm thinking to make a test with the display completely disconnected (I use Octopi) since @cooldudie2 is not sure that is ok to get current from SWD. Regarding @minosg reported issue in my understanding if this happens for a single driver it should be stopped only this and not the whole board.. is this correct?

adams79, 2020-06-03:
@minosg additionally note that at least in my case steppers are on (you can check this by trying to move the head or the bed with hand) so drivers should not be stopped

adams79, 2020-06-03:
cannot be yet sure, but have printed for about 6 hours with display disconnected and no issue so far. will keep it disconnected, if someone can make the same try it will be useful to have multiple test cases... (you will need to print from pc or octoprint)

minosg, 2020-06-05:
@adams79 The Marlin lockup of the steppers is configured and you can set it in the conf files( see DEFAULT_STEPPER_DEACTIVE_TIME and DISABLE_X/Y)

The TMC driver lockup is a totally different issue, which is triggering a failsafe component on the motor driver IC when it monitors a current spike outside of the expected thresholds. As an oversimplified explanation, when you invert the direction the momentum of the motor will generate an inverse current which depending on phase can add to the driving one and push it over the threshold. When that happens the driver chip will lock up and you need to power cycle it to unlock it. TMC engineers in the thread I posted, recommend either:

- Gradually phase out direction changes
- Add a delay between direction changes for the inverse pulse to dissipate

Following up my suggestion, I did some research and experiments.

In your conf_adv you have set up the motor currents as

```
  #if AXIS_DRIVER_TYPE_X(TMC26X)
    #define X_MAX_CURRENT     1000  // (mA)
    #define X_SENSE_RESISTOR    91  // (mOhms)
    #define X_MICROSTEPS        16  // Number of microsteps
  #endif
```

This could be an issue for a couple of reasons

- MP Select mini comes with the [JK42HS34-1334 ](http://www.jkongmotor.com/Product/42HS-NEMA17-2Phase-Stepper-Motor-1-8%C2%B0.html ) stepper motors, which are rated at 1.33A peak current. This converted to RMS which the stepper driver requires it would be 0.940A. 
- The X_SENSE_RESISTOR  seems a bit too large, compared to what is set-up for the ender which is what this board was created as.

Mind that most Enders runs on 2A rated motors and at 24V which make the current calculations different. This would explain why most mini users see that issue but Ender users do not.

I have set the X_MAX_CURRENT  to a safe threshold of 500mA and the MINIMUM_STEPPER_POST_DIR_DELAY/MINIMUM_STEPPER_PRE_DIR_DELAY to a large arbitrary value of 1500 and it has yet to lock-up yet. I would recommend you do the same and try to see if that addresses the issue

adams79, 2020-06-05:
hi @minosg you should check config for tmc2209 and current is set to 450mah (380 for the Xaxis motor). Regarding the MINIMUM_STEPPER_POST_DIR_DELAY this appears only related to the the driver itself so this problem should happens for all users of this board... However please let us know is you have no other freeze. Unfortunately I've to confirm that with display connected and powered with a sperate line I still experienced a freeze. The only case with no freeze for almost 10 ours has been with display completely disconnected...

minosg, 2020-06-05:
@adams79 apologies I was looking at the wrong section of the attached config. 

As for the the problem appearing to every user, that should not be the case. BTT SKR e3 board has the r26 and r27 resistors set to .11Ω which according to chapter 8 of the TMC2209 datasheet is for sensing a motor of 1.7A and higher. While this setting is usefull for sensing the end point break, it is indicative that Select Mini is using a considerably lower power motor than the board was tested on. If that is operating at its threshold a lot of bizarre things can happen.

My testing is showings  that this delay does matter, and gets to make the issue less frequent. You can also isolate the extruder motor as the suspect, because if you do not use filament and run it as a dry run it should never freeze.

The other good question is why is the watchdog not kicking in and resetting the board?

adams79, 2020-06-05:
I’m just trying to swap the board with an skr 1.4 turbo that I got for
another printer, it has 4 tmc 2209... I’m curious to check if the same
problem will arise (however it will require some work to mount the board in
the little case)...

minosg, 2020-06-08:
To answer my own question and to offer a **workaround** on the freezing part, the reason is that the watchdg is not enabled on this cpu HAL

https://github.com/MarlinFirmware/Marlin/issues/18226

**If you uncomment this line the board will now reset when the crash occurs which is more safe than crashing with the heaters on.**

As to why that stop happens, I am still investigating it since it has been reported for months, and there is no apparent solution.

What I have confirmed so far.

- It is not directly related with stealthchop. It gets worse by it but it can trigger on builds with Spread cycle on motors only.
- It is somehow related with the extruder motor. Running prints without filament always complete.
- It triggers in both prints from SD and Host enabled
- It can occur in both 512 and 256KB builds.
- Setting the extuder jerk to 2.0 or lower can increase the frequency.

When it happens the board is lost. You have lost UART debugging, sd logging etc

This means that most likely we are seeing an interrupt race condition, or disabled interrupts. But on the same time the heaters are running in set temperature, so the pwm loop for the heater is working, which makes it less likely to be a hardfault.

adams79, 2020-06-08:
HI @minosg,
thank you for the update. In the meantime I've managed to mount the skr 1.4 turbo board and start printing, until now (about 4 hours) I've had no freeze. I was not able to connect the Maylan lcd onto this board (tried to connect onto the tft serial but unfortunately on this serial I see the standard console output and not the encoded one that should come from the maylan ext ui... can you help?)

adams79, 2020-06-08:
@minosg I can confirm that is related to the extruder, I see that printing at high speed with high layer height and fast retract the freeze occurs very often.

minosg, 2020-06-09:
> HI @minosg,
> thank you for the update. In the meantime I've managed to mount the skr 1.4 turbo board and start printing, until now (about 4 hours) I've had no freeze. I was not able to connect the [...]

That is a separate issue but I would be looking at the serial declaration. The Malyanlcd serial is set in
[extui_malyan_lcd.cpp](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/lcd/extui_malyan_lcd.cpp)

You need to make sure the Serial is available and properly set in the appropriate pins, ie for BTT Skr Mini you can find it on [pins_BTT_SKR_MINI_E3.h](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_E3.h)

Last but not least you need to make sure the host serial is properly selected in config.h

minosg, 2020-06-09:
Also does the SKR turbo have hardware or software serial? I have come to notice that SKR mini 1.2 is using software serial to talk to the drivers, and I am not quite confident that the pulse timings are withing acceptable threshold. 

But could a broken serial channel on the TMC driver commns break the firmware? I cannot see how.

adams79, 2020-06-09:
@minosg after more than 12 hours printing on the skr 1.4 board I experienced no freeze with the mini. Note that I use the same drivers (TMC 2209) on UART, so seems to me that the problem is not directly related to the drivers or the motors...

taragor, 2020-06-10:
I've just run into the issue exept the Z skipping.
I'm using the SKR mini V2.0 (STM32F1, TMC2209) on an ender 3. However I have printed without any issues for probably 30 hours so far on 50mm/s. 
Yet the printer freezes randomly with heaters on as soon as i try printing at 80mm/s (Had one freeze after 1h and one after 5h, same gcode). 
Anyway I'm not sure that the issue comes from sudden direction changes since both prints failed at curves/circles in walls (cura wall speed: 40mm/s)

minosg, 2020-06-10:
@taragor according to my experiments, it is more related to the acceleration than the maximum printing speed. You can trigger it quite consistently if you use any high retraction model ( ie Flexi Articulated Gecko Keychain ) and set:

- Maximum E-Jerk 2.0 or less
-Maximum Feed-rate E 100mm/s or more
- Maximum Accel E to 6000+
- Maximum Accell E when extruding to 1200+

As long as you have filament in the printer it will get frozen in the next 5 prints even when printing at 40 or 35mm/s speed.

Other things we have confirmed so far is that it still happens when you:
- Try older release tags of marlin.
- Reduce the TMC serial speed to 115200. Mini e3 is using software serial which is not ideal.
- Enable DISABLE_MULTI_STEPPING
- Try with a higher watt power supply ( 220W )
- Work with a higher voltage threshold 24V instead of 12V ( that is what you just confirmed )
- Disable steathchop and run the drivers in Spreadcycle mode ( less frequent but it happens )
- Disable Linear Advance, Hybrid threshold altogether
- Set minimum square pulse to 2, and 4
- Enable Square Wave Stepping
- Enabled velocity ramping by increasing the ihold_delay using tmc_adv()

Basically every workaround on tickets relating to the same issue proposed in the last few months has been tested, and even though it makes it better, it will not fix this issue.

The question now is weather the issue we are facing is: 

- A hard to reproduce long standing, creeping marlin issue
- A board design issue, if it only happens on SKR mini. BTT have omitted the recommended coil ripple protection capacitors in the driver outputs ( 470pf/100V). But as I see from this bug report [17161 ](https://github.com/MarlinFirmware/Marlin/issues/17161)something similar is happening in LPC1716X
- A physical limitation. If you try to do this the CPU cannot keep up for the fp math, but then it should either caught up in sanity checks or graciously crash rather than halt with heaters on.

I am pretty much at a loss.

sawaguna, 2020-06-10:
I have the same problem with my SKR 1.3
It's really random. 
It can works for days without issues, and then it can freeze during a print, with heaters on.

No idea where the problem is.
I had it happening with both SD and USB (Repetier-Server)

First time I had this problem, it was a file with 30% gyroid infill.
Each time I was attempting to print it, it would freeze the printer (being via SD or USB).
I just changed the infill type to rectangular, and it worked.

So I thought it was maybe a cache issue.
But I really don't know, and it's probably not related

taragor, 2020-06-10:
@minosg 
Wait, do you mean it will litteraly crash for the next 5 print, regardless of the file?

> As long as you have filament in the printer it will get frozen in the next 5 prints even when printing at 40 or 35mm/s speed.

If so I think I witnessed that. After my first failed print I had some issues restarting it:
I was printing live from octoprint and octoprint lost connection during G29:
It heats the bed and then does G28 followed by G29 (as specified in my start Gcode), however it stops sending temperature status and Octoprint will just time out. G29 however completes normally
I didn't think much of that since I occasionally had that issue before so I'm not even sure if it's connected to this.
I had this a few times before but this time it happened a few times in a row. I can't recall how often exactly but since i tried powercycling both the printer and the raspberry I guess it was 3-5 times.

minosg, 2020-06-10:
> @minosg
> Wait, do you mean it will litteraly crash for the next 5 print, regardless of the file?
> 
> > As long as you have filament in the printer it will get frozen in the next 5 prints even when printing at 40 or 35mm/s speed.

I believe you need a high retraction model. Articulated models are perfect for that. The reason that his issue has been creeping for over a year is that is quite random and hard to reproduce.

What I meant above is that if you use that model, and those  settings in prusaslicer  it will make it more apparent. 

**My gecko gcode, will crash the firmware in maximum 5 prints, regardless of the setting combinations I described above.** 

But if you run it without filament in the extruder, it will never crash.

taragor, 2020-06-10:
Do you know if it is possible to debug Marlin using JTAG? If so it might give a clue whether it's the m3 just halting or marlin waiting for some kind of event/data.

minosg, 2020-06-10:
Yes you can. BTT SKR mini is using an ST chip, which is supporting SWD debugging. The cheapest way to go around it is to buy an ST-Link which is supported by platformio. 

It becomes a bit more complicated for select mini users since they use that header to power the 3.3 display, which will need to be externally powered during the debug session.

The other issue is that there is no DebugMonitor for the STM32F1 implemented, like the LPC1768 so you need to implement something similar to use the unwinder and unwmemaccess

So it is possible, but being an open source project, people have time constrains. 

**What we need to determine if that is a proccessor/HAL family related problem or weather it affects all Marlin, and is just hidden by the fact that other platforms have a working watchdog ( So the board is resetting instead of being stuck with heaters on )**

taragor, 2020-06-14:
I've had another two failed prints. Both of them failed while printing curved perimeters.
Also I tried some more geckos with different settings and they all worked. 
Since all my fails were on curves I'll try to disable S-Curve acceleration or use junction deviation and see if that makes any difference.

adams79, 2020-06-17:
After many days using the SKR 1.4 turbo instead (with the same kind of TMC 2209 drivers) I can assure that it never freeze. Note that I'm using mostly the same configuration. I'm still not able to connect the display so I finally swapped it with a TFT24. As I mentioned below all my tries with the LCD disconnected completed with e3 mini also, so I suggest to make some try in this way (you will need an octoprint or a pc to send the print to the board)

minosg, 2020-06-17:
Just an update on this issue, having been testing for it in the last week.

When it happens, the Bed and Hotend PWM is working as intended, which means both the interrupts, and the PWM are functioning.  Also the DIAG pin on each of the drivers appears to be low, and the Index pin pulsing as expected.

Considering that the issue becomes more visible with setting high acceleration settings in M204 and under the assumption that it could not be the consumer's fault ( tcm_stepper) I started looking into the producer (planner.cpp). 

If you disable the Malyan_LCD and print using SD card commands (M21 & M24) I have yet to have a freeze in two consecutive days of printing the test file. This could also be the case for #18315

BTT have pushed a change like that on their display driver which could indicate a possible workaround 
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/5777f41b5f8c1a41410a6874614499c91ac78fa2

Could the people affected please share:
- What display are you using
- What happens if you run it in headless mode by disabling that display

adams79, 2020-06-17:
@minosg In my case using the Malyan display caused the issue.. Unfortunately I've not swapped both board and display so I don't know If I solved the problem by using a different display (TFT24) or board...

minosg, 2020-06-17:
You don't have the original board to try?

adams79, 2020-06-17:
No unfortunately I've returned it.

taragor, 2020-06-20:
@minosg I've tested with the display disabled in Marlin, having the TFT35 in BTT mode, therefore just running as an serial GCODE terminal, like octoprint or pronterface.
I've just had another freeze with the following setup: 
-TFT35 connected using the TFT connector (Serial 2 in Marlin)
-Raspberry connected via USB (Serial -1) and was printing live of Octoprint 
-Display disabled in Marlin, ribbon cable to EXP3 on the screen disconnected
-Object was sliced in cura, speed 80mm/s, despite that pretty much default settings.

minosg, 2020-06-20:
@taragor pleaee disable displays on firmware, not just disconnect them and print using sd commands. You can do that with pronterface. The issue i am seeing are serial interrupts firing into each other. Printing though octoprint or even enabling host keep alive will trigger it.

taragor, 2020-06-20:
@minosg I have disabled `#define CR10_STOCKDISPLAY` and some other things according to sanity check (i.e. M73). I'm currently printing with octoprint disconnected, and only the TFT35 in BTT mode (for my understanding that should be the same as using pronterface, please correct me if I'm wrong here). I'm currently printing live of the TFT, using a USB drive in the TFTs USB Slot, but will try printing of the printers onboard SD once that finishes or fails.

minosg, 2020-06-20:
I think that printing from sd is m21, m24 commands but pronterface has an sd button to automate it. And yes that will make sure the only interrupts firing are the tsteppet ones. Btt mode is still a serial

boelle, 2020-06-23:
Anyone that has the board mentioned:

Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

minosg, 2020-06-24:
@boelle I have the BTT SKR mini e3 which is the one mentioned. I also have a cheetah board, using the same STM32F1 chipset, and I have ordered another one. All of them are using this chipset and TCM uart's in serial mode and exhibit the same issue

I can confirm that the issue is still there on latest bugfix and the latest TMC_Stepper library ( v0.7 )

adams79, 2020-06-24:
I confirm that I'm seeing that on SKR E3 Mini with latest bug fix but not on SKR 1.4 (LPC chipset)

minosg, 2020-06-24:
### Theory 3:

I have been testing this board using a debugger. The original theory that the LR Is corrupted seems less likely and I believe it is more of an interrupt firing issue. 

I believe it was introduced by this pr https://github.com/MarlinFirmware/Marlin/pull/14030 , but need deeper understanding of the stepper class.

### What happens when the printer freezes.

A freeze will become evident when obviously the printer is no longer moving. The logic is NOT stuck in exception mode, and the interrupt will be returning to the main thread. The code snippet that is of interest is this part

https://github.com/MarlinFirmware/Marlin/blob/2.0.x/Marlin/src/module/planner.h

```
FORCE_INLINE static uint8_t moves_free() { return BLOCK_BUFFER_SIZE -1 - movesplanned(); }
FORCE_INLINE static uint8_t movesplanned() { return BLOCK_MOD(block_buffer_head - block_buffer_tail); }
FORCE_INLINE static block_t* get_next_free_block(uint8_t &next_buffer_head, const uint8_t count=1) {

  // Wait until there are enough slots free
  while (moves_free() < count) { idle(); }

  // Return the first available block
  next_buffer_head = next_block_index(block_buffer_head);
  return &block_buffer[block_buffer_head];
}
```
Which causes the UI update loop to trigger, and the lcd uart to keep on firing. The problem appears to be that this **while loop is not breaking** .

Chasing the logic the block_tail is only updated in the following method loop which appears is no longer firing

Stepper::pulse_phase_isr()-> discard_current_block -> release_current_block
Stepper::block_phase_isr()-> discard_current_block -> release_current_block

The interrupt configuration on this platform is the following

```
TMC_STepper SoftwareSerial Timer4
STEP_TIMER_NUM             Timer5
PULSE_TIMER_NUM            Timer5
TEMP_TIMER_NUM             Timer2
SERVO0_TIMER_NUM           Timer8

NVIC_SetPriorityGrouping(0x3);
#define STEP_TIMER_IRQ_PRIO 1
#define TEMP_TIMER_IRQ_PRIO 2
```
Which appear to be enabled during the time of the freeze

```
NVIC-ISER0 0x501001c0
NVIC-ISER1 0x00044160
```

And the interrupt priorities appear to be in order

```
IPR7_N0 ->  0x20
IPR12_N2-> 0x10
IPR9_N1 -> 0xd0
```

I have also checked timer5 and the CC register as well as the count register and the timer appears to be counting as expected.

### Further notes

This platform appears to not be using instruction and memory [barriers ](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489c/CIHGHHIE.html)for interrupts. I do not know if there is a design choice behind that, but the pull request I quoted is enabling and disabling timers using CPSID, CPSIE instructions, with no guarantees that they are actually in the state that we want them to.

I am not certain on the priority grouping setting. Considering that I cannot see any use of sub-priorities in Marlin Core, are they being used in Mapple or should it be set to 0x4 anyway? Is that code a snippet of when we were using Autopreload with the timers instead of resetting them?

When the board freezes most interrupts but the lcd and the temperature stop firing. This is also very bizarre behavior which I am trying to track down. **_This is also a very happy coincidence, since if the temperature interrupts dies the hotend wil cause meltdown/fire so its a considerable safety risk_**. I have disabled interrupt grouping and set the uasart1 ( lcd) interrupt to have the lowest priority (15)  and yet it keeps on firing, which indicates there is no interrupt preemption or starvation happening.

What I suspect happens is because of too many interrupts firing , ( TMC set in UART modes individually and not chained in a single line ) something in the stepper.cpp logic mini scheduler is falling out of its timing bounds. The problem is that when the issue happens is way before it becomes evident in the logic. so a longer trace is needed which is not supported by st-link.

To be continued.

minosg, 2020-06-24:
@adams79  are you using malyan_lcd on SKR1.4? What mode are you using the steppers on?

adams79, 2020-06-24:
@minosg no I'm not able to use the Malyan LCD, I'm using a TFT24 connected BOTH in serial and LCD12865 emulation mode. I'm using all steppers in stealthchop, the drivers are tmc2209. Print dozens of hours with no problem with this setup

drakaz, 2020-06-29:
Hello,

Same issue with a SKR mini v2.0, board completely freeze randomly with heater on and fan on, printing from Octoprint.

lexter12, 2020-07-01:
Same issue with SKR Mini V2, but can't print more than 5 minutes without freeze - pretty consistent, never managed to print over 10 minutes without hitting the freeze.

Depz42, 2020-07-03:
SKR Mini V2, downloaded a fresh copy of Marlin bug fix on 6/29 and the freezing issue persists with no rhyme or reason. I had a failed 20 minute print in the first few layers, successful 8 hour print, failed 30 minute print about halfway through. I've been able to save one of the prints because it tells you which layer it was on, but still a pretty serious bug.

minosg, 2020-07-03:
I have posted a possible workaround, which involves patching the libmaple library.

https://github.com/MarlinFirmware/Marlin/issues/18358#issuecomment-652584451

Can you please try it and let us know in #18358 if it has resolved your issues?

Depz42, 2020-07-03:
I uploaded the change and testing a print now. I tried printing the same thing 3 times and it failed twice, so it's not 100% reproducible. I'll monitor the next few prints and keep you updated.

Depz42, 2020-07-03:
Just ran a 1:45 hour print with no issues.

minosg, 2020-07-03:
Sd card print or host (octoprint?). Anyhow if cause was what I described in case 5 you shouldn't get any more prints.

Remember you may need to reapply the patch if you upgrade the platfromio

Depz42, 2020-07-03:
Octoprint on a raspberry pi 3B+. Thanks for the quick response and your time fixing this.

closed 2020-08-09 by github-actions[bot]

droneando, 2020-08-17:
> I have posted a possible workaround, which involves patching the libmaple library.
> 
> [#18358 (comment)](https://github.com/MarlinFirmware/Marlin/issues/18358#issuecomment-652584451)
> 
> Can you please try it and let us know in #18358 if it has resolved your issues?

> I have posted a possible workaround, which involves patching the libmaple library.
> 
> [#18358 (comment)](https://github.com/MarlinFirmware/Marlin/issues/18358#issuecomment-652584451)
> 
> Can you please try it and let us know in #18358 if it has resolved your issues?

Hi:
This workaround works for me! And also locks issues related with M500 use in terminal are fixed!
Thank you very much!