MarlinFirmware/Marlin #18119: [BUG]  Marlin 2.0.5.3 Auto Bed Leveling

celticht32, 2020-05-26:
I have a BLTouch v3 and have flashed Marlin 2.0.5.3 onto my Creality CR-10 S4... I followed every change guide I could find on the interwebs... and compiled and flashed to the CR-10. I got the cmos error.. Re-saved my settings at the default screen with no changes on the controller.. I then selected to do the auto bed leveling..

The gantry goes to the home center and deploys the BLTouch as normal.. then goes to the first probe point of 9 performs that correctly then goes to the second point deploys the BLTouch goes down a few then stops without touching the bed then raises up and then stops and the auto bed leveling stops without continuing..

celticht32, 2020-05-26:
[creality-cr10s4-bltouch3.zip](https://github.com/MarlinFirmware/Marlin/files/4683094/creality-cr10s4-bltouch3.zip)

c3D-Dan, 2020-05-27:
Check your bed is within 2mm from corner to corner.  If it's not, weird things happen. Run G29 P1 T to see how close you are.

psavva, 2020-05-27:
Is this a duplicate of https://github.com/MarlinFirmware/Marlin/issues/18033

celticht32, 2020-05-27:
it actually could be but mine I have 9 points defined and it stops on the second one.... 
the display says 9 points  and like I said goes to the center home point... then point 1 and then goes to the second one and stops and goes to the main marlin menu....

psavva, 2020-05-28:
> Download Marlin bugfix-2.0.x to test with the latest code.
> Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
> Issue M111 S247 to enable maximum logging.
> Issue G28
> Issue G29 V4
> Attach the log output into a .TXT file

See if you find Probing Failed towards the end of the log.  If so, seems that it's the same bug
..

I enabled UBL from @Roxy-3D  as it doesn't suffer from the same issue.  I'm sticking with UBL because of how well it performs. Absolutely awesome is all I can say

celticht32, 2020-05-28:
I will test it.. I saw you changing to UBL fixes the problem so I changed that and reflashing now...  will test in the morning

celticht32, 2020-05-30:
This is the same as issue above... doing UBL fixes the issue per above... so bug is https://github.com/MarlinFirmware/Marlin/issues/18033

celticht32, 2020-05-30:
also tested per above and get same log entry...

closed 2020-05-30 by celticht32