MarlinFirmware/Marlin #18121: [BUG] Display and Buttons unresponsive reprap_discount_full_graphic_smart_controller LPC1768 MKS SGEN_L

hamster65, 2020-05-26:
Hi,

my display often does not react to button or encoder actions during print (it is fine when paused or not printing). I guess it depends on the gcode / on the movements. 

Sometimes, I am not able to do anything for... let's say almost the whole first layer. Sometimes, it blocks only for 1 second.
I also observed that this does not happen every time. With the same gcode, it might work fine or not.

It seems as if inputs are somehow processed. When I turn the knob in the menu, nothing happens. When I turn again, the cursor eventually moves down 2 entries.
On the other hand, during baby stepping, the Z axis does not react, so those inputs are not processed. When the display finally reacts, Z might move a long way....

Especially in the menu, the display sometimes gets stuck while updating, text is garbled until I move the encoder again. 

In general, I have to trigger an update by pushing or rotating the encoder, otherwise it does not update.

Changing serial buffers and flow control had no impact. I also went back to 16 substeps with no effect.

hamster65, 2020-05-26:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/4683585/Marlin.zip)

bill-orange, 2020-05-26:
Boy, my gut reaction is that you have a hardware problem not a software problem.  Try rerouting you cables.  Also, while I have never experienced this, others have reported on the Facebook forum that touching a stepper motor can corrupt the display.  This is likely the result of bad or absent grounding.  Double check that you have good grounding on your mains and that hot and neural are not reversed.  Give every thing a good "wiggle test".  

You may find that the Facebook forum for Marlin is better able to help.

borland1, 2020-05-26:
Have you tried increasing (double/triple) the encoder rate multiplier setting values in file Configuration_adv.h under "Additional Features"?

```
// Change values more rapidly when the encoder is rotated faster
#define ENCODER_RATE_MULTIPLIER
#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #define ENCODER_10X_STEPS_PER_SEC   30  // (steps/s) Encoder rate for 10x speed
  #define ENCODER_100X_STEPS_PER_SEC  80  // (steps/s) Encoder rate for 100x speed
#endif
```

hamster65, 2020-05-26:
I am sure my wiring  grounding is ok. It is a corexy, the steppers and cables are miles away from the display cable. I am also experienced with electronics.

The encoder multiplier is only used when you rotate fast, but my problems happen also when I want to increase or decrease a value by on, or even when I press the encoder button. No multiplier there :).

borland1, 2020-05-26:
You could be right.  I had thought it might be related to this encoder problem....

https://github.com/MarlinFirmware/Marlin/issues/16298

hamster65, 2020-05-27:
I don't know if interrupts are used for the encoder. 

I think this happens when the printer is doing curves. It seems straight and diagonal lines are fine, on curves the main screen freezes. Coordinates are updated in irregular intervals between 1 and 20 seconds. Speed does not seem to matter, it happens on the first layer (20mm/s) and also on faster layers. The first layer is probably worse but that might be just my impression.

I also observed stuttering or small pauses on curves.

Perhaps a Cura setting? But I don't think I have changed anything special there.

hamster65, 2020-05-29:
For the record:
No issues on my last print. It might be related to the gcode but I also changed the following:
- lowered x y e substeps from 64 to 32
- Post and Pre dir delay from 100 to 50 µs 
- disabled square wave stepping
- enabled integrated baby stepping
I'll enable square wave stepping for my next print.

sjasonsmith, 2020-06-20:
@hamster65 are your issues resolved now? Can this issue be closed?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2020-06-23 by boelle