MarlinFirmware/Marlin #18122: [BUG] error: 'class CDCSerial' has no member named 'dropped' on enable SERIAL_STATS_*

drackp2m, 2020-05-26:
### Bug Description

Basically, by enabling the `SERIAL_STATS_MAX_RX_QUEUED` and / or `SERIAL_STATS_DROPPED_RX` option in Configuration_adv.h I get the following error in compilation.

```
Compiling .pio/build/LPC1769/src/src/gcode/control/M211.cpp.o
In file included from Marlin/src/gcode/control/../../inc/MarlinConfig.h:46,
                 from Marlin/src/gcode/control/../gcode.h:286,
                 from Marlin/src/gcode/control/M111.cpp:23:
Marlin/src/gcode/control/M111.cpp: In static member function 'static void GcodeSuite::M111()':
Marlin/src/gcode/control/M111.cpp:68:56: error: 'class CDCSerial' has no member named 'dropped'
   68 |         SERIAL_ECHOPAIR("\nDropped bytes: ", MYSERIAL0.dropped());
      |                                                        ^~~~~~~
Marlin/src/gcode/control/../../inc/../core/serial.h:105:57: note: in definition of macro '_SEP_2'
  105 | #define _SEP_2(PRE,V)     serial_echopair_PGM(PSTR(PRE),V)
      |                                                         ^
Marlin/src/gcode/control/../../inc/../core/serial.h:103:27: note: in expansion of macro '__SEP_N'
  103 | #define _SEP_N(N,V...)    __SEP_N(N,V)
      |                           ^~~~~~~
Marlin/src/gcode/control/../../inc/../core/serial.h:129:31: note: in expansion of macro '_SEP_N'
  129 | #define SERIAL_ECHOPAIR(V...) _SEP_N(NUM_ARGS(V),V)
      |                               ^~~~~~
Marlin/src/gcode/control/M111.cpp:68:9: note: in expansion of macro 'SERIAL_ECHOPAIR'
   68 |         SERIAL_ECHOPAIR("\nDropped bytes: ", MYSERIAL0.dropped());
      |         ^~~~~~~~~~~~~~~
Marlin/src/gcode/control/M111.cpp:72:60: error: 'class CDCSerial' has no member named 'rxMaxEnqueued'
   72 |         SERIAL_ECHOPAIR("\nMax RX Queue Size: ", MYSERIAL0.rxMaxEnqueued());
      |                                                            ^~~~~~~~~~~~~
Marlin/src/gcode/control/../../inc/../core/serial.h:105:57: note: in definition of macro '_SEP_2'
  105 | #define _SEP_2(PRE,V)     serial_echopair_PGM(PSTR(PRE),V)
      |                                                         ^
Marlin/src/gcode/control/../../inc/../core/serial.h:103:27: note: in expansion of macro '__SEP_N'
  103 | #define _SEP_N(N,V...)    __SEP_N(N,V)
      |                           ^~~~~~~
Marlin/src/gcode/control/../../inc/../core/serial.h:129:31: note: in expansion of macro '_SEP_N'
  129 | #define SERIAL_ECHOPAIR(V...) _SEP_N(NUM_ARGS(V),V)
      |                               ^~~~~~
Marlin/src/gcode/control/M111.cpp:72:9: note: in expansion of macro 'SERIAL_ECHOPAIR'
   72 |         SERIAL_ECHOPAIR("\nMax RX Queue Size: ", MYSERIAL0.rxMaxEnqueued());
      |         ^~~~~~~~~~~~~~~
*** [.pio/build/LPC1769/src/src/gcode/control/M111.cpp.o] Error 1
```

#### Additional Information

It may just be an uncaught error, like when I try to enable the `M575` command by activating `BAUD_RATE_GCODE` and it shows me a clean error saying my board doesn't support it.

Which by the way, is a `BOARD_BTT_SKR_V1_4_TURBO`.

Greetings and thank you very much for the great work you do. ;)

sjasonsmith, 2020-05-26:
Hi @drackp2m, could you please attach your configuration files?

drackp2m, 2020-05-27:
This is my setup for another issue that I opened a very few days ago (and they fixed quickly). If you want me to upload the current version, I can, but nothing has changed that I think should be relevant.

[Config](https://github.com/MarlinFirmware/Marlin/files/4634774/Configuration_2.0.5.2_15-05-20.zip).

sjasonsmith, 2020-05-29:
> It may just be an uncaught error

It looks like you were right about this. The serial classes (both USB and HardwareSerial) don't support the calls needed for these capabilities. I've posted a PR to add sanity checks for these for LPC, STM32, and STM32F1 HALs.

drackp2m, 2020-06-04:
Perfect, I downloaded the changes of last bugfix-2.0.x and a much more understandable error already appears when I activate either option.

Thank you very much @sjasonsmith for your lines of code.

closed 2020-06-04 by drackp2m