MarlinFirmware/Marlin #18123: [FR] Make https://marlinfw.org/docs/configuration/configuration.html readable

Tofoillote, 2020-05-26:
### Bug Description

on screen display of https://marlinfw.org/docs/configuration/configuration.html

is absolute mess going insanely wide !

### My Configurations

up to date firefow or chrome

### Steps to Reproduce

Have a look and shfit+wheelmouse scroll ...

#### Additional Information

Sorry for anouncing it this way, but it make me nuts, I really need that doc !

Tofoillote, 2020-05-26:
[image]

thisiskeithb, 2020-05-26:
This was already reported in the [MarlinDocumentation repo](https://github.com/MarlinFirmware/MarlinDocumentation/issues/306).

[Here's a link to the readable configuration page](https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_configuration/configuration.md).

rflulling, 2020-05-27:
wow, some looks like "My First HTML" in there. Over board on the nested cells, and does not have the boundaries set right.

I am glad to see needed updates. However this one didn't quite get off the ground.

sjasonsmith, 2020-06-20:
I'll go ahead and close this, because it looks like it was fixed. I remember seeing the problem when this was first reported, and it is all displaying cleanly for me now.

closed 2020-06-20 by sjasonsmith