MarlinFirmware/Marlin #18125: [BUG] G30 and G38 don't move or do anything?

Patronics, 2020-05-26:
### Bug Description

 I'm having an issue with Marlin 2.0.5.3, where I'm trying to configure it for a homemade CNC machine (intended primarily for etching PCBs), where the Z-axis probing commands (G30, and G38.2) seem to do nothing? 

They are recognized commands (it doesn't complain about not understanding them), and I'm pretty sure I have all the right configuration options set, and m119 correctly returns open when not contacting the bed, and triggered when it is. 

I'm just really puzzled as to why the machine just sits there when I ask it to probe the bed, but happily does all other movements, homing, etc, just fine. Any ideas what's going wrong? I can't really see any possibilities of it being anything other than a bug in marlin, unless there's some really obscure configuration option I'm missing. 

### My Configurations

[configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/4684861/configuration_files.zip)

### Steps to Reproduce

1. Install provided configuration files on a RAMPS-based CNC machine (with both a z-max endstop and z-min conductive tip as probe)

2. Send it a G30 command or G38.2 from your host software of choice.

**Expected behavior:** [What you expect to happen]

The Z axis moves down until the probe hits the bed

**Actual behavior:** [What actually happens]

The machine just sits there as if nothing had happened. (but does accept the command, responds ok, unlike unrecognized commands).

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Patronics, 2020-05-27:
I figured out the issue (at least for G30, G38.2 still does nothing?) It was with the x coordinate being below the threshold of MIN_PROBE_EDGE in the configuration.h file!
The firmware should really give some sort of feedback that that's the problem though, rather than just pretending everything proceeded as normal.

sjasonsmith, 2020-09-11:
When debugging probing in the past I have noticed that "not reachable" probe points are not reported well. There is certainly room for improvement in this area, unless something has changed since I last worked with it.