MarlinFirmware/Marlin #18126: [BUG] (short description)

kadangriebeler, 2020-05-26:
### Whine TMC2130

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

swilkens, 2020-05-27:
We need configuration files, printer model and control board information to be able to help you - as the template asks you to provide.

This sounds like the driver is in spreadcycle mode

kadangriebeler, 2020-05-28:
I'm sorry for the missing info and for my rubbish english...
I have a generic graber i3 from a local kit seller. It has mega2560 + ramps 1.4. I'm running the latest bug_fix 2.0.
During the printing, the movement noise is barely audible, but when the motors are stop came a fine whine.
Which are the files and how i upload them?

Thanks for the patien!

psavva, 2020-05-28:
Please upload your configuration and configuration_adv files

AmiSMB, 2020-05-28:
Not sure what your configuration is but you will get a whining no matter what if you are running at 12v.

sjasonsmith, 2020-06-20:
Closing due to required and requested configuration information not being provided.

closed 2020-06-20 by sjasonsmith