MarlinFirmware/Marlin #18127: [BUG] bugfix-2.0 fails to compile for string-related definitions on Linux for 32-bit platforms

ajosmer, 2020-05-27:
### Bug Description

After experiencing random crashing issues with the stable 2.0.5 branch, I tried compiling the bugfix-2.0.x branch for my Archim1 tonight.  This resulted in a long list of errors concerning redeclarations, all coming from string definitions in /src/MarlinCore.h and /src/MarlinCore.cpp, and how they differ from a definition in /src/HAL/HAL.h.  This appears to affect multiple 32-bit platforms, not just the Archim1 I'm building for, including the DUE itself and STM32.  The problem exists with a default configuration files on the 32-bit platforms I tested.  The problem does not exist on 8-bit AVR (tested the same dummy project on ATmega2560 + RAMPS).

This appears to be an issue with the way the PROGMEM array from the stable 2.0.5 branch was converted to a macro from the /src/HAL/HAL.h file and instantiated in /src/MarlinCore.cpp (array in old version is line 184, and macros in new version start at line 194 of /src/MarlinCore.cpp).  This smells a little like running through all 3 recursive passes of the precompiler without getting all the macros replaced to me, but I don't have the experience or knowledge to either back up that theory, or figure out how to up the number of recursive passes in the makefile.

### My Configurations

Control board:  Ultimachine Archim1 (SAM3X8E/DUE based)
Compiling system:  Arduino 1.8.12 on Linux Mint 19.1
Compiler: arm-none-eabi-gcc version 4.8.3-2014q1

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/4685706/Config.zip)

### Steps to Reproduce

1. Download recent buxfix-2.0.x branch (same behavior in versions from May 22, 2020 and May 26,2020)
2. Open in Arduino 1.8.12 on recent Linux (specifically Ubuntu-based, probably not the issue)
3. Configure for affected 32-bit platform, is not configuration file dependent (other than motherboard name)
    Board:  BOARD_ARCHIM1
    Board manager file:  https://raw.githubusercontent.com/ultimachine/ArduinoAddons/master/package_ultimachine_index.json
4.  "Verify" or Ctrl+R
5.  Produces the attached list of errors for string redeclarations.

**Expected behavior:** [What you expect to happen]

Compile with no errors, as the main 2.0.5 branch does with functionally identical configuration files.

**Actual behavior:** [What actually happens]

[Bugfix-2.0 May 26 2020 Errors.txt](https://github.com/MarlinFirmware/Marlin/files/4685707/Bugfix-2.0.May.26.2020.Errors.txt)

#### Additional Information

Tests performed:
1)  Build a project from the May 22, 2020 version with the attached configuration files for Archim1.  Produced the attached errors.
2)  Rewrite configuration files on May 26, 2020 version and build for Archim1.  Same errors.
3)  Build a clean copy of May 26, 2020 version for Mega2560 +  RAMPS.  Succeeds, no errors.
4)  Build a clean copy of May 26, 2020 version for Archim1.  Produces previous errors.
5)  Build the clean project for DUE instead of Archim1.  Produces previous errors.
6)  Build the clean project for Arduino Grand Central M4 (SAMD51 board).  Fails with different errors about bootloader, did not follow up.
7)  Build the clean project for STM32F103RE with target Maple.  Similar errors as with Archim1.
8)  Rewrite /src/HAL/HAL.h to remove 'constexpr' from:
        #define PGMSTR(NAM,STR) constexpr char NAM[] = STR
    Produces similar errors.

ajosmer, 2020-05-27:
Update:  I get the same errors when using PlatformIO with Sublime 3.2.2.

ellensp, 2020-05-27:
Firstly you have have this in your configuration.h
```
  //120V 250W silicone heater into 4mm borosilicate (MendelMax 1.5+)
  //from FOPDT model - kp=.39 Tp=405 Tdead=66, Tc set to 79.2, aggressive factor of .15 (vs .1, 1, 10)
  #define DEFAULT_bedKp 10.00
  #define DEFAULT_bedKi .023
  #define DEFAULT_bedKd 305.4
```

and
 
```
  //Custom
  #define DEFAULT_bedKp 41.41
  #define DEFAULT_bedKi 3.62
  #define DEFAULT_bedKd 315.81
```
and you have 
```
// Ultimaker
  #define DEFAULT_Kp 22.2
  #define DEFAULT_Ki 1.08
  #define DEFAULT_Kd 114
```
with 

```
  //Custom
  #define DEFAULT_Kp 53.59
  #define DEFAULT_Ki 6.44
  #define DEFAULT_Kd 111.47
```

This generates thousands of redefinition issues.

ajosmer, 2020-05-27:
I just missed that those weren't commented out in all my setting copying.  Those will generate preprocessor warnings, but not compiler errors.  The same compiler errors are present with a totally unmodified configuration file (with the exception of changing the motherboard).

ellensp, 2020-05-27:
"But other than the copious warning, on linux (ubuntu) using vscode and platformio it compiles fine."

Also if you update pins_ARCHIM1.h Timers section with 
```
//
// Timers
//

#undef STEP_TIMER_NUM
#define STEP_TIMER_NUM                         3
#undef HAL_STEP_TIMER_ISR 
#define HAL_STEP_TIMER_ISR()  void TC3_Handler()
```
It stops all the HAL_STEP_TIMER_ISR and STEP_TIMER_NUM redefinition warnings and makes everything more readable.

ajosmer, 2020-05-27:
I guess I'll try it on VSCode tomorrow, but there isn't an Archim board profile for Platformio, so I'm not sure if it'll produce a usable binary anyway.  Any idea what version of arm-none-eabi-gcc is running on your system?

ellensp, 2020-05-27:
Use the real marlin. it has the following. 

#
# Archim SAM
#
[env:DUE_archim]
platform      = atmelsam
board         = due
src_filter    = ${common.default_src_filter} +<src/HAL/DUE>
build_flags   = ${common.build_flags}
  -DARDUINO_SAM_ARCHIM -DARDUINO_ARCH_SAM -D__SAM3X8E__ -DUSBCON
extra_scripts = Marlin/src/HAL/DUE/upload_extra_script.py

Al you need to set is the following in platformio.ini 
default_envs = DUE_archim

ajosmer, 2020-05-27:
And thanks for the debugging tip for the pins file, that ought to make this a nicer head-bashing experience in the mean time.

ajosmer, 2020-05-27:
Okay, I'll give that a shot.  By "real Marlin" do you mean the stable branch as opposed to the bugfix branch?  Or bugfix as opposed to the 3-year-old version Ultimachine supposedly tuned for the Archim?

ellensp, 2020-05-27:
from the marlin github, I was testing with bugfix (but only to the level of compiling) not the 3 years old version on Ultimachine (who deigned this board originally)

ellensp, 2020-05-27:
toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)

ajosmer, 2020-05-27:
VSCode with PlatformIO did build successfully.  I had to do a lot of digging to figure out how to get it to upload, but in addition to setting default_envs to DUE_archim, you also have to edit the board in the environment section to be "dueUSB" instead of "due".  I now have the project built and uploaded to the Archim.

That puts the source of the bug somewhere that's different between the VSCode PlatformIO environment and the Sublime Text 3 PlatformIO environment.  Probably in the toolchain or compiler options.  Using VSCode works okay for me right now, but the issue should still be addressed unless Sublime Text 3 and Arduino are to be officially deprecated for 32-bit board use.

ajosmer, 2020-05-28:
Okay, I've got it, this is a weird one.  Whatever was changed between stable 2.0.5 and the recent bugfix-2.0.x creates errors when compiled with arm-none-eabi-gcc version 4.8.3-2014q1, which is the toolchain specified by the Arduino IDE for the SAM3X8E architecture.  These errors are not produced on more recent versions of the arm-non-eabi-gcc toolchain, such as the 7-2017q4, which was already in my Arduino tools, probably for some other installed ARM architecture.

HOWEVER, the Ultimachine-supplied ArduinoAddons for Archim contain an Arduino core which is not compatible with the newer toolchain.  The hack is as follows:

1) ~Changing the compiler toolchain path~
1a) Using the Arduino boards manager, install (if you haven't already) the most recent Due board (1.6.12 at the time of this writing).
1b) Open ~/.arduino15/packages/arduino/hardware/sam/1.6.12/platform.txt [NOTE: 1.6.12 will be your Due board library version, should be the only directory in sam/)
1c) Change:
compiler.path={runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path}/bin/
to:
compiler.path={runtime.tools.arm-none-eabi-gcc-7-2017q4.path}/bin/

2) ~Updating the Archim Arduino core~
2a)  Rename the ~/.arduino15/packages/ultimachine/hardware/sam/1.6.9-d/cores/arduino directory to "arduino.bak" in case something gets seriously broken.
2b)  Copy ~/.arduino15/packages/arduino/hardware/sam/1.6.12/cores/arduino in place of the folder you just renamed.

3) ~Removing a deprecated linker flag~
3a)  Open ~/.arduino15/packages/ultimachine/hardware/sam/1.6.9-d/platform.txt
3b)  Find linker command, titled "## Combine gc-sections, archives, and objects" (the title is line 80 in mine, the command to edit is line 81)
3c)  Delete the "-gcc" command option from the end of the line.  This creates a linker error on the newer arm-none-eabi-gcc toolchain because it interprets it as the new "-g" flag with "cc" as the level of debugging output, which is not valid.  Probably add a comment below it that you changed this in case you need to revert for some reason.

WHY OH WHY WOULD I GO TO THE TROUBLE???

Because Ultimachine has added a lot of very unique, uh, "features" to the Archim1 board, in particular using the DACs built into the SAM3X8E to control the DRV8825 stepper driver current, and PlatformIO doesn't know anything about it because that mostly came around after Ultimachine moved on from the Archim1 to focus on the Archim2 (which has different stepper drivers that work completely differently).  Without the ArduinoAddons from Ultimachine, the current control doesn't work, meaning you'll get one really loud motor and a bunch of motors that don't have enough torque to move themselves.  Essentially, the problem is two-fold:

1) Ultimachine has not supported the Archim1 with meaningful code in quite some time, and provide no documentation concerning Marlin 2.0 whatsoever, much less PlatformIO.
2) The Arduino IDE mandates an old compiler for the SAM3X8E which does not support some of the code recently used in the Marlin bugfix-2.0.x branch.

I'm going to close this bug since the problem has been identified, and for all intents and purposes, is exterior to Marlin.  However, it's looking like some of the more esoteric or older boards are going to essentially be unsupported lacking any meaningful continued contribution from their manufacturers as Marlin continues to grow.

closed 2020-05-28 by ajosmer