MarlinFirmware/Marlin #18129: Support for TFT and touch screen - need feedback and testing

jmz52, 2020-05-27:
TFT screens with optional touch panel can be used to control Marlin
Low-level IO uses DMA for background data transfer to TFT screen to minimize impact on Marlin's performance.

Current low-level IO implementation is for HAL STM32 and MCU STM32F1 only.
SPI implementation lacks STM32F4 initialization code.
FSMC implementation STM32F4 initialization code and (temporary) hardcoded to Zx specific.
Implementation for STM32F4 and FSMC on VE chips will follow.

Implemented Marlin_UI class for 320x240 screen.
Can be used with or without rotary encoder.
Support UBL or advanced pause is not implemented yet.

Tested on MKS Robin v2.4 with TFT v2.0 (ST7789v controller)

Earlier build were tested on EZT-T1 and  SKR Mini v1.1 with 128x128 (st7735 controller) but UI 128x128 became outdated due to changes made during touch support develoment.
UI 128x128 implementation will be uploaded once code is updated.

SPI IO required dedicated hardware SPI interface wich can't be shared with SD card. Should be OK to use same SPI for TFT screen and touch panel.

jmz52, 2020-05-27:
Bootscreen
[image]

jmz52, 2020-05-27:
**Status screen**
Do not mind grey color of hot ends - there are no sensors connected.
Hotends, heated bed , heated chamber, fan, feed rate, flow rate are clickable and will lead to corresponding edit screens
Settings icon will lead to main menu.
SD card will lead to "print from sd" menu when sd is mounted and printer is idle.

[image]

jmz52, 2020-05-27:
**Menu screens**
With page up, back and page down clickable navigation icons.
Click to menu item to select it.
Click to selected icon to activate it.
Double click logic is used to reduce errors when dealing with smaller screens.

[image]
[image]

jmz52, 2020-05-27:
**Edit screens**
With decrease, confirm, increase **and slider** controls.
Axis move screen does not have slider due to specific encoder handling.

[image]
[image]

jmz52, 2020-05-27:
**Confirm screen**
With cancel and confirm controls.
Selected control is highlighted and other one is dimmed, so you can know with one is selected if you use encoder instead of touch panel.
[image]

jmz52, 2020-05-27:
**Kill screen**
[image]

jmz52, 2020-05-27:
Related issues
https://github.com/MarlinFirmware/Marlin/issues/16787
https://github.com/MarlinFirmware/Marlin/issues/17844

jmz52, 2020-05-27:
@hobiseven, would you mind some code adoption and testing on Alfawise Ux0 boards?

hobiseven, 2020-05-27:
Gosh... Looks like we do the same thing, different flavors. I have a Little LGVL glue ported and running on Alfawise... 
See what we are about to test : 

https://www.lesimprimantes3d.fr/forum/uploads/monthly_2020_05/IMG_1702.jpg.2878a786f88dc99978ec135913587ea4.jpg

I invite you to read the code I built upon tag 2.5.0.3. 
This is work in progress. I use extui API. 

I send you an invite to my privare repo..

hobiseven, 2020-05-27:
@jmz52 
https://www.lesimprimantes3d.fr/forum/uploads/monthly_2020_05/IMG_1691.jpg.c92379445e9526c18adb49b1478a7ba2.jpg

LVGL is very nice to use, and the adaptation to get DMAs running in the background is really tiny.

hobiseven, 2020-05-27:
@jmz52  something which LVGL allows : dynamic displays : See the small moovie. I get the code from one of the forum member tonight . Please check the small video below : 

https://www.lesimprimantes3d.fr/forum/applications/core/interface/file/attachment.php?id=92278

thisiskeithb, 2020-05-27:
Did you see MKS’ LVGL port? https://github.com/MarlinFirmware/Marlin/pull/18071

hobiseven, 2020-05-27:
NOOOPEEE. So we all on our side do work on the same things!

thisiskeithb, 2020-05-27:
Options are nice 🙂

mattdog01, 2020-05-27:
This is beautiful.  Absolutely love it.
However, are you going to make it work for TFT v1.1?
In the current nightly build the "touch button" don't work on my TFT v1.1.  They are there but no action.  Will this upgrade fix that?

jmz52, 2020-05-27:
@mattdog01 TFT v1.1 needs different calibration data.
Check you Configuration.h for code below to see if you are using right numbers.
Default are for TVT v2.0

New code should work on TFT v1.1, but you need to change calibration data in
https://github.com/jmz52/Marlin/blob/tft_and_touch_screen/Marlin/src/HAL/STM32/tft/xpt2046.h

And make sure you are using build environment from
https://github.com/jmz52/Configurations/tree/mks_robin_hal_stm32/config/examples/Mks/Robin_HAL_STM32
Check [extra_config.ini](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Mks/Robin) for details

```
//
// ADS7843/XPT2046 ADC Touchscreen such as ILI9341 2.8
//
#define TOUCH_BUTTONS
#if ENABLED(TOUCH_BUTTONS)
  #define BUTTON_DELAY_EDIT  75 // (ms) Button repeat delay for edit screens
  #define BUTTON_DELAY_MENU 100 // (ms) Button repeat delay for menus

  /* MKS Robin TFT v2.0 */
  #define XPT2046_X_CALIBRATION    12013
  #define XPT2046_Y_CALIBRATION    -8711
  #define XPT2046_X_OFFSET           -32
  #define XPT2046_Y_OFFSET           256

  /* MKS Robin TFT v1.1 */
  //#define XPT2046_X_CALIBRATION -11792
  //#define XPT2046_Y_CALIBRATION   8947
  //#define XPT2046_X_OFFSET         342
  //#define XPT2046_Y_OFFSET         -19
#endif```

hobiseven, 2020-05-27:
@mattdog01 looks like your touch settings are inverted in y. So touch zones are on the opposite side of the screen. We actually have also 3 touch screen flavors on alfawise.

hobiseven, 2020-05-27:
@jmz52  I like your print screen. I was more thinking about more arrows for move and less text menus....

MS1987, 2020-05-28:
@jmz52 @hobiseven , can I have your email? I want to develop the touch screen together.
My email: lms228@163.com

hobiseven, 2020-05-28:
Folks, 
I probably do not have the expertise and level you guys have... I share my repo with you also @MS1987 

I will include a piece of LVGL code ( for the images / screens in the coming days, and push it. The goal for me was to make a POC of what the CPU can handle, without destroying the print performance. I started to build a small state machine , and picket icons googling here and there. @sensei73 asked to copy Alfawise U30 Pro GUI and he started to put together LVGL code to do that. 

As lvgl only updates what is nedded, we already do less DMA than with DOGM if I am correct.

looxonline, 2020-05-28:
Keen to see this ported to the LP1768/9 HALS. Will open it up substantially. Very nice to see marlin supporting TFT'S natively instead of via gcode as a proxy.

hobiseven, 2020-05-28:
Let me ask my application manager is he can assign someone to this. I am from NXP.

looxonline, 2020-05-28:
> Let me ask one of my application manager if he can assign someone to this. I am from NXP.

That would be incredible. If the SPI mode is used it could open the functionality up to many thousands who use the SKR series of motherboards via the EXP2 header.

hobiseven, 2020-05-28:
Well, you have a couple of things here :
LVGL is hardware independent, and beside the speed , all LVGL code could work on ANY TFT. The right LVGL hal has to be done. and this is far from rocket science. This includes FSMC for ST, SPI for a bunch or other boards, with speed limitations , as well as parallel TFT interfaces, for instance, on LPC1788. 
Hence, it looks to me that LVGL is the proper vehicule going forward. @thinkyhead do you have any preference on the graphics lib?

hobiseven, 2020-05-28:
> **Bootscreen**
> [image]

Hi @jmz52 

How did you store that picture in the chip? Is that a jpg decompiled on the fly? or a SD card file?

mattdog01, 2020-05-28:
Hello @jmz52 
I have a functional TFT v1.1 again. Thank you! 
My Nightly Build downloads didn't have src/HAL/STM32/tft so I added the TFT folder to STM32. Then I was able to update /xpt2046.h with my screens calibration numbers. You had taught me about the calibration difference for v1.1 for the Configuration.h file and it has worked great for a few of the Marlin updates and then it stopped working. I had to use a previously compiled ROBIN.bin to restore printer function.  
Prior to your help yesterday I had to revert back to "Marlin-MKS-Robin-ILI9328", that you provided me a while ago, just to be able to change settings on my printer and have it work.
So now I am really looking forward to setting up for your awesome GUI.  
Did your reply give me enough info to get the new TFT function working on my printer?  I tried to follow through it (but it was late in the day) and I had compiling errors that prevented a successful build.  I will try again today.
Thanks so much for your work on this.  It is going to be great to have the TFT experience again but under Marlin.  I'm loving Marlin for machine functionality but I have missed the bling of the MKS firmware.  Take care.

jmz52, 2020-05-29:
I think some explanation is on order.
Originally I've started TFT work for EZT-T1 delta printer.
It has mainboard is based on STM32F103RCT6 MCU and 128x128 TFT screen based on st7735 controller.
MCU has 128KB flash only meaning Marlin with delta math and HAL_STM32F1 will never fit into it.
In order to squeeze firmware into such limited flash I've had to switch to HAL_STM32 and abandon u8glib completely.
Fonts and icons were reused, but one does not need an whole library to draw 1bpp images.
Instead of expanding existing 12864 graphical interface I've made a new implementation of MarlinUI class dedicated for color screen of particular resolution (128x128).
This approach is less flexible that extui, but is has an undeniable benefit of already developed menu system.
I am no expert in Marlin's internals, so it would be a tremendous amount of work for me to develop new menu structure.
Beside I am not sure how much flexibility is needed for tiny 128x128 screen.

With MarlinUI class implementation, once you have a working status screen you need to implement about 5 drawing functions to get a fully functional menu.
It is not be as pretty as UI used on MKS boards or BTT smart screen controllers, but it is way better that x2 upscaled 12864 interface we have on Robin boards for some time now.

Final result was rather satisfying - I was able to fit delta math, SD support, eeprom emulation and code for TFT screen into 128KB.
[image]

With new UI working it was a matter of several night to implements new IO subsystem for FSMC and get new UI to MKS Robin TFT. Then I could start working on new display resolution and touch support.
[image]

I am no familiar with LVGL yet, but if it is lightweight enough to not impact print quality it might be a good foundation for new UI. My only concern here is that it might be too flash consuming for low-end MCUs.

For now I am going to focus on SDIO issues and FSMC support for STM32F103VE and STM32F4 MCUs. This low-level functionality will be needed for either UI implementation.

jmz52, 2020-05-29:
> How did you store that picture in the chip? Is that a jpg decompiled on the fly? or a SD card file?
> 320_240_2bytes/RGB565 is 153Kb, Jpeg is about 25Kb... hence my question

@hobiseven this is raw 16-bit TFT image. There is an external SPI flash on MKS Robin board but I am not using it (yet)
https://github.com/jmz52/Marlin/blob/tft_and_touch_screen/Marlin/src/lcd/tft/images/bootscreen_320x240x16.h
Other images are in 4 bpp greyscale format colored with dynamically generated palette.
1 bpp and 2 bpp images are supported as well.

sensei73, 2020-05-31:
@jmz52 Interesting, if we have access to the external SPI flash, we could use it to load images as menu. This technique is used on the U30 Pro, less ram consumed.

sensei73, 2020-05-31:
@jmz52 By the way, what is the size of this flash?

hobiseven, 2020-05-31:
@jmz52 Indeed we also have a pretty large Spi flash connected  on spi2. We could store data there too if needed. So far the basic lvgl lib plus a few menus takes 40kb of prog mem, on the ram side you need preferably at least 2 buffers of 10 full lines> we have for the data+ lvgl buffers a bit more than 50% of chip sram used and we can allocate 24kb to lvgl memory for graphics objects. A basic tile menu such as the one in the above video takes 6kb ram. 

Regarding using the spi memory, question is how to load it? We have reused alfawise bootloader and we can load prog mem but I am not sure about spi... we would need either to change alfawise bootloader with one that can load prog mem plus spi mem... or maybe we could check what the current bootloader does with the large longer3d.ui file, as it might store it in the spi... 

Do you have a piece of code able to dump an spi mem through serial, or write it into the sdcard??

odaruc, 2020-06-01:
Just wondering - any compatibility with MKS TFT32 V4.0? It looks like it is equipped with an STM32F107

I since replaced with Bigtreetech TFT35 running unified firmware. Works much better but still some rough spots before it could fully replace a stock screen. So I have two of the MKS TFT32 that are out of work and looking for a project.

jmz52, 2020-06-01:
@odaruc that depends on MCU used in MKS TFT and TFT screen interface.
I've check connection type for BTT TFT screens and here is what I've found:

- BTT TFT 24 has STM32F105RC MCU with no FSMC support. TFT screen uses 16 bits FSMC interface which is emulation via GPIO. There are some trick how DMA can be used for 16 bits GPIO data transfer, but I have not check if these tricks are applicable to schematic of this screen. If DMA transfer will njot be possible, then screen IO will take  way to much of MCU's time to be anything more that just proof of concept and UI demonstration board.
- BTT TFT 35 has STM32F205VC MCU and TFT screen is connected to hardware FSMC interface. It can run TFT UI code, so with custom expansion board you can make a control board from this smart screen.

Do you know if schematics for MKS TFT32 V4.0 is available somewhere?

hobiseven, 2020-06-01:
@jmz52 Indeed, you are correct. However, some TFT can also cope with SPI. There is an example with ESP32, a faster CPU, 250Mhz, vs 72 for STM32F103VE, running a TFT with LVGL, through DMA/SPI. might be applicable for some setups. 
But definitely, a true 16 bit // FSMC is much much faster. 

As far as I understood LVGL, the updates are made only on the needed basis, ie, you change an icon, only that icon gets updated, and not the full screen, saving quite some bandwidth. 

I still need to integrate the tile example from @sensei73 into my POC. Actually, the LVGL port of MKS is using a similar pages state machine than my embryon state machine I started to build. I requested them to provide me visuals of their LVGL GUI. Did not replied so far.

hobiseven, 2020-06-01:
@jmz52 gosh, I realize your issue. You have  STM32F103RCT6 MCU and I have STM32F103VET6 ..;
We have the luxury of 512Kb flash and 64Kb ram. 
Indeed, we are right now already over 256Kb...even with only a few tiles. That is a big issue for LVGL on your CPU.

looxonline, 2020-06-01:
> @jmz52 gosh, I realize your issue. You have STM32F103RCT6 MCU and I have STM32F103VET6 ..;
> We have the luxury of 512Kb flash and 64Kb ram.
> Indeed, we are right now already over 256Kb...even with only a few tiles. That is a big issue for LVGL on your CPU.

Actuall the STM32F103RCT6 contains 512k flash and there are no fuses blocking it off. I am assuming it just ships untested but in many cases it does work. If you can tell the linker to map non essential elements such as graphics to the upper end of flash then it would be very usable. The 48kB of RAM may be a bit more concerning and I have not seen any reports indicating that the STM32F103RCT6 can take advantage of extra RAM that is mapped but not in the spec sheet.

jmz52, 2020-06-01:
My mistake - EZT-T1 board is based on STM32F103RBT6 (128KB flash, 20KB RAM), not STM32F103RCT6.
STM32F103RCT6 is an MCU in SKR Mini v1.1 I've tested code on.

jmz52, 2020-06-01:
About SPI.
From there I stand using SPI storage for images (and may be fonts too) is a No2 priority.
No1 is to make IO works on SPI and FSMC for F1 and F4 MCUs.

Once the code is works on all boards it will be time to make UI looks pretty - better images, alternative menu structure for TFT screens, STL preview.
External SPI flash will be a great help here but I would suggest to make it optional. Most SPI screens have no SPI flash, so it is better to have an option to run with reduced graphics that can fit into progmem.

SPI flash update should be Marlin's responsibility - we can't expect random bootloader to be able to handle Marlin's images. I see two ways to update flash images from SD card - during TFT init phase and via dedicated g-code command.

looxonline, 2020-06-01:
> About SPI.
> From there I stand using SPI storage for images (and may be fonts too) is a No2 priority.
> No1 is to make IO works on SPI and FSMC for F1 and F4 MCUs.
>
> Once the code is works on all boards [...]

Absolutely agreed that Marlin should be responsible for the SPI storage update. Requiring bootloaders to do this would make this out of reach for most users as they would not know how to re-flash. Additionally, it would require either manufacturers to add this to their bootloaders or to make their bootloaders open source. Neither of which I feel is going to happen.

hobiseven, 2020-06-01:
@jmz52 i actually have done the dma fsmc stuff for f103. I do not have spi screens to test with. And I now looks at possible gui speed impacts although I have dma running in the background which is easier for the cpu than what we had with dogm.

xC0000005, 2020-06-07:
Doesn't the SD FAT library support SPI flash? We could treat the images as files, update becomes a "copy."

vivian-ng, 2020-06-08:
@hobiseven Can you share the approach for implementing LVGL? For example, what are the files that can be common across boards, and what are those parts which need to be specifically implemented in the HAL? I am hoping to get something up for ESP32 to use ILI9341 display and using LVGL seems to offer quite some potential for cross-board compatibility with lower-level requirements handled by each HAL.

makerbase-mks, 2020-06-08:
@hobiseven I can show serval images about our using of LVGL:
[image]
[image]
[image]
[image]

hobiseven, 2020-06-08:
> @hobiseven Can you share the approach for implementing LVGL? For example, what are the files that can be common across boards, and what are those parts which need to be specifically implemented in [...]

Hi. All is already written for ESP32 / LVGL. The port is done, although it needs to be integrated within Marlin. 3 main functions to be addressed, well described in LVGL documentation : 
1- DMA + Hardware SPI to have a background flush of the display buffers on the frame buffer within IL9341. It has to be DMA and it has to be SPI hardware. 
2 - Touchscreen event handler , using a second SPI channel
3 - timer to be added along the step and temp timers

and if you want to debug, a 4th function, to dump messages on the serial line.

hobiseven, 2020-06-08:
> @hobiseven I can show serval images about our using of LVGL:
> [image]
> [image]
> [image]
> [image]

Looks good.

What is the final code size? How much RAM did you allocate to LVGL? 
Have you monitored CPU load?

makerbase-mks, 2020-06-09:
@hobiseven ,My MCU has  64K RAM, 512k FLASH.  And just now we use about 60% of RAM, 50% of FLASH. For LVGL, we have allocate 20k .

hobiseven, 2020-06-09:
@makerbase-mks  > Sounds like an stm32f103ZE or VE . I got RAM 50%, including 2x10 line buffers for the DMA, and we can allocate about 24Kb for LVGL RAM,. For flash, as soon as you fit pictures, it goes up, but I am at about 260K. So pretty similar. I make a similar state machine, but way simpler... I will copy yours. 

Question is now : When scrolling, have you noticed any speed issue with the printer? There might be the infamous layer shifts... But as we run 32 bit CPUs, we may not bump in the issue. 

Well, good work you did! Now, we can all have fun making custom skins for the printers ( themes, or whatever you want to call a custom look).

ETE-Design, 2020-06-12:
@jmz52 Would this Work on the MKS TFT 28 / 32?

jmz52, 2020-06-12:
@ETE-Design, MKS TFT or MKS Robin TFT?
MKS TFT is smart screen with it's own MCU and firmware. It is connected to main board via COM port. This code is not applicable to MKS or BTT smart screens.
MKS Robin TFT does not have it's own MCU and depends on main board (MKS Robin). These screens are supported.

dandantsui, 2020-06-12:
@jmz52 I use a MKS_Robin on my printer. Happy to test out if you need.

enry68, 2020-06-14:
Compliments !!! I whish to try to port your job on the Anycubic Trigorilla Pro board with colours TFT. What about it ? Do you think is it possible ? 
Thank you and again compliments for your job.
Enrico.

hobiseven, 2020-06-14:
As long as you have enough flash ( 64kb) ans ram (20ko+) and a fast spi with dma or better a 16 bit Port with dma that will work. Note for 8bit avr that will be a challenge... works so far on stm32f103 . Due should be fine so as lpc1768.

enry68, 2020-06-14:
Trigorilla Pro has 32bit processor ...

hobiseven, 2020-06-14:
Then you need to adapt the Hal and get 3 basic functions ported: display flush callback, touch callback ( might be exactly the same as the one implemented) and the custom timer for lvgl. That s all! Lvgl documentation has a good porting guide all is explainned

enry68, 2020-06-14:
> 
> 
> Then you need to adapt the Hal and get 3 basic functions ported: display flush callback, touch callback ( might be exactly the same as the one implemented) and the custom timer for lvgl. That s all! Lvgl documentation has a good porting guide all is explainned

>Thank you very much!

jmz52, 2020-06-14:
@enry68 jus to make sure there is no misunderstanding.
There are two different implementation of TFT UI.

Backported MKS UI for 480x320 TFT screens - https://github.com/MarlinFirmware/Marlin/pull/18071
Developed for STM32F1xx MCUs (HAL STM32F1) Based on LVGL library.

Implementation of native Marlin menu system for 320x240 TFT screens - https://github.com/MarlinFirmware/Marlin/pull/18130. Developed for new HAL STM32 and tested on STM32F1 and STM32F4 MCUs. Does node use external libraries in current implementation.

If I recall correctly the Trigorilla Pro board is based on STM32F103Zx MCU with TFT screen connected to FSMC interface. Do you have pinmap for your board?

enry68, 2020-06-14:
> @enry68 jus to make sure there is no misunderstanding.
> There are two different implementation of TFT UI.
>
> Backported MKS UI for 480x320 TFT screens - #18071
> Developed for STM32F1xx MCUs (HAL [...]

Ciao, and thank you very very much! Yes I've extracted the pinout from the schematics and compared it with MKS Robin (pro). I've made and excell file with this table. Can you help me for the implementation ? Is it possible to attach here the excel file ?

I try anyway to insert here:

TRIGORILLA DISPLAY PINOUT

  |  
-- | --
1 | 3.3V
2 | N.C.
3 | N.C.
4 | N.C.
5 | GND
6 | PD15
7 | PD14
8 | PD0
9 | PD1
10 | PD7
11 | PD11
12 | PD5
13 | PD4
14 | PD13
15 | PE7
16 | PE12
17 | PE13
18 | N.C.
19 | PE14
20 | PE15

21 | PD8
22 | PD9
23 | PD10
24 | PB6
25 | PA6
26 | PA7
27 | BOOT0
28 | PA5
29 | PF11
30 | PE8
31 | PE9
32 | PE10
33 | GND
34 | GND
35 | GND
36 | N.C.
37 | N.C.
38 | 3.3V
39 | 3.3V
40 | 3.3V

MKS ROBIN DISPLAY PINOUT

1 | 3.3V
-- | --
2 | GND
3 | PD14
4 | PD15
5 | PD0
6 | PD1
7 | PE7
8 | PE8
9 | PE9
10 | PE10
11 | PE11
12 | PE12
13 | PE13
14 | PE14
15 | PE15
16 | PD8
17 | PD9
18 | PD10
19 | PD7
20 | PD11

21 | PD5
-- | --
22 | PD4
23 | PC6
24 | PD13
25 | PB14
26 | X
27 | PB15
28 | PC5
29 | PB13
30 | VCC
31 | PA7
32 | GND

enry68, 2020-06-14:
[TrigorillaPro-MKSNano.xlsx](https://github.com/MarlinFirmware/Marlin/files/4776805/TrigorillaPro-MKSNano.xlsx)

Done!

jmz52, 2020-06-14:
@enry68 Is you board supported in Marlin? If not you need complete pinout to make a pins definition file - heaters, thermistors, endstops, stepper motors, UART(s).
You need to get Marlin running on your board before you add new code to use TFT screen.
Also do you know if your board's bootloader uses firmware encryption?

jmz52, 2020-06-14:
@dandantsui you can get latest code from https://github.com/jmz52/Marlin/tree/tft_and_touch_screen
and add following build environment to platformio.ini file
```
[env:mks_robin_stm32]
extends          = common_stm32
board            = genericSTM32F103ZE
board_build.core = stm32
board_build.variant = MARLIN_F103Zx
board_build.ldscript = ldscript.ld
board_build.offset = 0x7000
board_build.firmware = Robin.bin
build_flags       = ${common_stm32.build_flags}
 -DTRANSFER_CLOCK_DIV=8 -DHAVE_HWSERIAL3
 -DHAL_SRAM_MODULE_ENABLED -DHAL_SD_MODULE_ENABLED
 -DHAS_GRAPHICAL_TFT -DHAS_FSMC_TFT -DTFT_320x240 -DTOUCH_SCREEN
 -DULTIPANEL -DLCD_HEIGHT=6 -DNO_LCD_REINIT
build_unflags     = ${common_stm32.build_unflags} -DUSBCON -DUSBD_USE_CDC
extra_scripts     = pre:buildroot/share/PlatformIO/scripts/generic_create_variant.py
  buildroot/share/PlatformIO/scripts/stm32_bootloader.py
  buildroot/share/PlatformIO/scripts/mks_encrypt.py
lib_deps =
```
And you need to disable  `#define FSMC_GRAPHICAL_TFT` and `#define TOUCH_BUTTONS` in Configuration.h file

[84 more replies]

closed 2020-11-04 by thisiskeithb