MarlinFirmware/Marlin #18131: [FR] meaningful linear advance error messages

c3D-Dan, 2020-05-27:


psavva, 2020-05-28:
@Sebastianv650 any idea?

c3D-Dan, 2020-05-28:
At the very least, I can gather from a full days testing, this is an indication the extruder jerk isn't set high enough for Linear advance to work.  Further of note, when this error occurs, it can be roughly translated to mean linear advance isn't working and therefore expect inconsistent extrusion.  So far, solutions I've found is to lower K factor and increase extruder jerk.  With a long bowden, My K factor is .95, meaning my E jerk needs to be insanely high to get linear advance working.  Having such a high E jerk will no doubt also result in skipped E steps, which...will give me inconsistent extrusion.    
Thinkyhead, could you please chime in?

Sebastianv650, 2020-05-28:
This is a debug message, indicating that stepper pulses are (or need to be) fired at more than 10kHz. This is a thing, as that's the threshold for double stepping, meaning there is a very high risk of overloading the CPU.

The needed solution in this case is to lower the requested steps/s to reduce the CPU load. Lower K factor, an extruder with lower steps/mm, printing slower, there are some options. Using LA on bowden systems was always on the edge of impossible :-/

Note that this is true for the 16MHz boards, I'm not longer so much up to date. Therefore I can't tell if that still a valid point for the newer 32bit boards.