MarlinFirmware/Marlin #18133: [BUG] (short description)

richardklingler, 2020-05-27:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

richardklingler, 2020-05-27:
Seems the defined A0 line is taken low during the execution of M504...

Tested with:

#define DOGLCD_A0   P0_16

and with default setting:

#define DOGLCD_A0   P1_22

richardklingler, 2020-05-27:
Issuing the command (set LCD contrast)

M250 C200
or M501

brings the display back...

When issuing the M504 command...and read back M250 contrast value, it is modified....

After reset:

M250
LCD Contrast: 190

After M504:

M504
echo:EEPROM OK
ok
M250
LCD Contrast: 255

After M501:

echo:; LCD Contrast:
echo:  M250 C190

boelle, 2020-06-21:
@richardklingler still an issue?

boelle, 2020-06-23:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

richardklingler, 2020-08-03:
This is still happening with 2.0.6 bugfix...

thinkyhead, 2020-08-04:
@richardklingler — Enable `PINS_DEBUGGING` and see if `M43` reveals anything about pin usage.

boelle, 2020-08-04:
@richardklingler 

Does it also happen with [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) ?

richardklingler, 2020-08-04:
This is what M43 returns:

`M43
PIN: 0.00        <unused/unknown>                       Input  = 1                       
PIN: 0.01        <unused/unknown>                       Input  = 1                       
PIN: 0.02 (A 2)  TXD0                                   Input  = 1
PIN: 0.03 (A 3)  RXD0                                   Input  = 1
PIN: 0.04        TMC_SW_SCK                             Output = 1
PIN: 0.05        TMC_SW_MISO                            Input  = 1
PIN: 0.06        <unused/unknown>                       Output = 1                       
PIN: 0.07        <unused/unknown>                       Input  = 0                       
PIN: 0.08        <unused/unknown>                       Input  = 1                       
PIN: 0.09        <unused/unknown>                       Input  = 1                       
PIN: 0.10        <unused/unknown>                       Input  = 1                       
PIN: 0.11        E0_DIR_PIN                             protected 
PIN: 0.15        DOGLCD_SCK                             Output = 0
.                SCK_PIN                                Output = 0
PIN: 0.16        LCD_SDSS                               Input  = 1
PIN: 0.17        MISO_PIN                               Input  = 1
PIN: 0.18        DOGLCD_MOSI                            Output = 0
.                MOSI_PIN                               Output = 0
PIN: 0.19        Y_STEP_PIN                             protected 
PIN: 0.20        Y_DIR_PIN                              protected 
PIN: 0.21        Z_ENABLE_PIN                           protected 
PIN: 0.22        Z_STEP_PIN                             protected 
PIN: 0.23 (A23)  <unused/unknown>   Analog in =     0   Input  = 0                       
PIN: 0.24 (A24)  <unused/unknown>   Analog in =  1023   Input  = 0                       
PIN: 0.25 (A25)  <unused/unknown>   Analog in =  1023   Input  = 0                       
PIN: 0.26 (A26)  SPINDLE_LASER_ENA_PIN                  Output = 1
PIN: 0.27        <unused/unknown>                       Input  = 0                       
PIN: 0.28        BTN_ENC                                Input  = 1
PIN: 0.29        <unused/unknown>                       Input  = 1                       
PIN: 0.30        <unused/unknown>                       Input  = 0                       
PIN: 1.00        POWER_LOSS_PIN                         Input  = 1
PIN: 1.01        E1_CS_PIN                              protected 
.                Y2_CS_PIN                              protected 
.                Z2_SERIAL_TX_PIN                       protected 
.                Z2_SERIAL_RX_PIN                       protected 
.                E1_SERIAL_TX_PIN                       protected 
.                E1_SERIAL_RX_PIN                       protected 
PIN: 1.04        E0_SERIAL_TX_PIN                       Input  = 1
.                E0_SERIAL_RX_PIN                       Input  = 1
PIN: 1.08        Z_CS_PIN                               protected 
.                Z_SERIAL_TX_PIN                        protected 
.                Z_SERIAL_RX_PIN                        protected 
PIN: 1.09        Y_CS_PIN                               protected 
.                Y_SERIAL_TX_PIN                        protected 
.                Y_SERIAL_RX_PIN                        protected 
PIN: 1.10        X_CS_PIN                               protected 
.                X_SERIAL_TX_PIN                        protected 
.                X_SERIAL_RX_PIN                        protected 
PIN: 1.14        E1_DIR_PIN                             protected 
.                Y2_DIR_PIN                             protected 
PIN: 1.15        E1_STEP_PIN                            protected 
.                Y2_STEP_PIN                            protected 
PIN: 1.16        E1_ENABLE_PIN                          protected 
.                Y2_ENABLE_PIN                          protected 
PIN: 1.17        TMC_SW_MOSI                            Output = 0
PIN: 1.18        LCD_PINS_ENABLE                        Output = 0
PIN: 1.19        LCD_PINS_RS                            Output = 0
PIN: 1.20        LCD_PINS_D4                            Output = 1
PIN: 1.21        DOGLCD_CS                              Output = 1
.                LCD_PINS_D5                            Output = 1
PIN: 1.22        DOGLCD_A0                              Output = 1
.                LCD_PINS_D6                            Output = 1
PIN: 1.23        LCD_PINS_D7                            Output = 1
.                SDSS                                   Output = 1
.                SS_PIN                                 Output = 1
PIN: 1.24        NEOPIXEL_PIN                           Input  = 0
PIN: 1.25        FIL_RUNOUT2_PIN                        Analog in =     0
[... 8 lines cut ...]
PIN: 1.29        X_MIN_PIN                              protected 
.                X_DIAG_PIN                             protected 
.                X_STOP_PIN                             protected 
PIN: 1.30 (A62)  BEEPER_PIN                             Output = 0
PIN: 1.31 (A63)  <unused/unknown>   Analog in =   818   Input  = 0                       
PIN: 2.00        SERVO0_PIN                             Input  = 1
PIN: 2.01        X_ENABLE_PIN                           protected 
PIN: 2.02        X_STEP_PIN                             protected 
PIN: 2.03        FAN_PIN                                protected 
[... cut ...]
PIN: 2.11        Z_DIR_PIN                              protected 
PIN: 2.12        E0_ENABLE_PIN                          protected 
PIN: 2.13        E0_STEP_PIN                            protected 
PIN: 3.25        BTN_EN2                                Input  = 1
PIN: 3.26        BTN_EN1                                Input  = 1
PIN: 4.28        <unused/unknown>                       Input  = 0                       
PIN: 4.29        <unused/unknown>                       Input  = 1                       
`

richardklingler, 2020-08-04:
Interesting is that M504 changes the contrast value which gets returned by M250:

M250
LCD Contrast: 210
ok
M504
echo:EEPROM OK
ok
M250
LCD Contrast: 255
ok

richardklingler, 2020-09-04:
So no comment on this?

ellensp, 2020-09-23:
You configs updated to current bugfix for testing.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5266041/Configuration.zip)

ellensp, 2020-09-23:
Current bugfix this still happens
> M250
LCD Contrast: 195
> ok
> M504
echo:EEPROM OK
> ok
> M250
LCD Contrast: 255
> ok

ellensp, 2020-09-23:
added some debugging 
20:09:41.158 > WRITE EEPROM:600Data:1
20:09:41.158 > ^^ EEPROM_WRITE lcd_contrast195

20:10:27.470 > READ EEPROM:600Data:195
20:10:27.470 > ^^EEPROM_READ lcd_contrast

Its not a eeprom issue. Same location, same data...

ellensp, 2020-09-23:
Found it...   preparing PR

ellensp, 2020-09-23:
If you take a look at the above PR it fixes the contrast corruption for the "M250 M504 M250" test.
It should also fix your LCD issue. 
Please test and confirm.
Its only a tiny 5 line change and should be easy to apply apply to your older version of marlin.

ellensp, 2020-09-23:
Is now part of marlin.

closed 2020-09-23 by ellensp