MarlinFirmware/Marlin #18134: [BUG] enabling BABYSTEP_XY in coreXY machines will lead to error on babystep function

ShadowOfTheDamn, 2020-05-27:
### Bug Description

enabling BABYSTEP_XY in coreXY machines will lead to error on babystep function definition.

### My Configurations

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4692237/Configuration_adv.zip)

### Steps to Reproduce
 
this is a compile time error. just by enabling BABYSTEP_XY  in config_adv you get the compile error.

**Expected behavior:** coreXY babystepping work out of the box

**Actual behavior:** it wont compile

#### Additional Information

it seems by changing the lines 2813 in ../src/module/stepper.cpp
from: 
  #elif IS_CORE
to:
#endif
  #if IS_CORE
you allow the code for having the babystep_core function to actually get defined which will cause the code to compiles. but I haven't got the time to test it on my printer.

ShadowOfTheDamn, 2020-05-28:
uploaded the edited firmware.
- babystepping is added to the menus.
- changing the values is possible during print for X and Y and Z.
- babystepping of X and Y axis does nothing (no movements)
- Z is working as before.

ruggb, 2020-05-28:
as far as I know XY babystepping does not and has never worked for Core XY machine. It used to be default enabled which prevented Z babystepping from working. I have no problem with Z babystepping on my CoreXY

ShadowOfTheDamn, 2020-05-29:
It is useful to have babystepping in XY directions because:
1- during print you may want to increase tension in your core-XY belts which may result in shifts in x or y direction which then can be compensated with babystepping.
2- I have read your previous comments @ruggb and you actually don't want marlin to have this feature for some reason and you just say no one will use it. so please don't start to ruin this post and remove your comments.

ruggb, 2020-05-29:
@ShadowOfTheDamn > 
> It is useful to have babystepping in XY directions because:
> 1- during print you may want to increase tension in your core-XY belts which may result in shifts in x or y direction which then can be [...]

Really! I really don't care if Marlin has the XY BS feature. It is like all the rest of the features I do not use. And if you can mess with your printer belts while it is printing then come up with a good print by adjusting XY, your better than I can imagine.
My issue was that the contributors had no clue why Z BS did not work with CoreXY printers and kept swearing to me that it does not work for CoreXY. I finally found someone on the net who had it working and told me how to make mine work by disabling XY BS. When I told them that, instead of checking it out they just argued with me that Z BS doesn't work with my printer. Apparently, mine isn't a CoreXY, right? It took them 2 years to figure out XY BS should be default disabled because it was causing Z BS not to work.

gouache, 2020-06-17:
I fonund some bugs and try to fix regarding BABYSTEP_XY in coreXY.
there's some duplicated corexy kinematics in babystep.cpp and stepper.cpp
I removed that and made working perfectly now. but I've tested only for coreXY.
so more testing is needed for another core system.
anyhow I'll make some pull request later. please review at that time.

thinkyhead, 2020-06-18:
In spite of the unwelcome comments from @ruggb disparaging everyone, there is now a patch so please give it a test.

boelle, 2020-06-21:
@ShadowOfTheDamn still an issue?

ShadowOfTheDamn, 2020-06-21:
Hi every one,
Sorry for the late reply, was busy repairing my PC and dont have the platformIO on my system yet.  thanks a lot guys. Will check the PR this weekend. will post about it.
cheers

boelle, 2020-06-23:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands.

ShadowOfTheDamn, 2020-06-27:
Used the new bugfix build and XY baby-stepping is working correctly.
Cheers

closed 2020-06-27 by ShadowOfTheDamn