MarlinFirmware/Marlin #18135: UBL G29 J Fail

inventabuild, 2020-05-28:
I run the below start code and the first and second "G29 J" probe points work fine (front left and right edge of the bed), but then the third probe point occurs in exactly the same spot as the second probe point (front right edge of the bed) instead of occuring at the center of the back edge of the bed.

G29 A; UBL activate UBL
G29 L1; UBL load mesh from slot 1
G29 J; UBL 3 point level

Does anyone have any idea why "G29 J" probes points 2 and 3 in the same spot leading to a failed plane tilt for UBL?  config.h/config_adv.h attached.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4693075/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/4693077/Configuration_adv.zip)

psavva, 2020-05-28:
@Roxy-3D any ideas?

Lord-Quake, 2020-05-28:
I'm using 2.0.5.3 with UBL. I haven't looked at you configs yet but are you homing your printer before your level with G29?

Here is my starting code:
```
G28                  ; Home extruder
G29 L1             ; Load mesh from slot 1
G29 J               ; Run 3 point level calibration
```

inventabuild, 2020-05-29:
Below is my entire start code (relevant code is bolded) and here's a vid showing points 2 and 3 being tapped at nearly the same spot (https://youtu.be/6698ECupQYQ).  The 3 tap points should be FRONT LEFT X40 Y7, FRONT RIGHT X253 Y7 and BACK CENTER X107 Y250.  What's interesting is that the LCD screen on my printer shows the hot end moved to the back center of the bed (X107, Y250), but in reality it stayed at about the same point at the front right corner of the bed (253, 7).

I've been doing testing and found out if I put a G4 S1 (1 SEC PAUSE) followed by an M400 (FINISH MOVES) after the G29 J then (so for) the G29 J completes with the third point happening at the back of the bed.  Have you or anyone ever heard of such accessory code being required to keep the gcode running properly after some "G" or "M" command or do you think my printer or wiring is really messed up somewhere?

START CODE
M73 P0 ; clear GLCD progress bar
M75 ; start GLCD timer
G26 ; clear potential 'probe fail' condition
M107 ; disable fans
G90 ; absolute positioning
M82 ; set extruder to absolute mode
G92 E0 ; set extruder position to 0
M140 S60 ; start heating bed no wait
M109 S140 ; wait for extruder to reach wiping temp
M190 S60 ; start heating bed wait
**G28 ; home extruder
M400 l wait for moves to finish
M204 S100 ; set probing acceleration
G29 A; UBL activate UBL
G29 L1; UBL load mesh from slot 1
G29 J; UBL 3 point level
M204 S500 ; restore standard acceleration
G4 S1 ; pause
M400 ; wait for moves to finish**
M117 Heating... ; progress indicator message on LCD
M104 S210 ; set ext temp don't wait
M140 S60 ; start heating bed no wait
M109 S210 ; set ext temp wait
M190 S60 ; set bed temp wait and wait
M104 S[extruder0_temperature] ; set ext temp don't wait
M140 S[bed0_temperature] ; start heating bed no wait
M109 S[extruder0_temperature] ; set ext temp wait
M190 S[bed0_temperature] ; start heating bed no wait
G1 Z2 E0 F75 ; prime tiny bit of filament into the nozzle
M117 TAZ 6 Printing... ; progress indicator message on LCD
M83 ; set extruder to relative mode

Lord-Quake, 2020-05-29:
I took your code in bold and pasted it a test file. My printer behaved as expected.

inventabuild, 2020-05-29:
Thank you very much for testing!  The code also works for me (so far) w/ the following two  lines of code (which are also in my start code above); however when I take out these two lines of code then 3 point leveling fails as described above.  Do you (or anyone) know if this kind of waiting / pausing code is sometimes necessary to put after a G code command (like G29) to prevent the printer from getting tripped up?
**G4 S1 ; pause
M400 ; wait for moves to finish**

Lord-Quake, 2020-05-29:
In my opinion both codes are not necessary. However even with them in place you should not be having the issue.
Right now I can't see this being a Marlin fault but rather a hardware or mechanical problem of some kind.

inventabuild, 2020-05-29:
Thanks, I agree.  At this point it points to a problem w/ the printer if it's not executing gcode as expected.

closed 2020-05-29 by inventabuild