MarlinFirmware/Marlin #18136: [BUG] (Leveling Fade causes gradual increases in layer height)

snekonaplane, 2020-05-28:
### Bug Description

When leveling fade is enabled, each increase in layer height beyond the set fade height has an additional z movement. This eventually causes the print to fail.

### My Configurations

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4693199/Configuration.zip)

### Steps to Reproduce

1. Set a fade height of 5 with start g code in slicer
2. Run print
3. Allow print to reach a layer height beyond the fade height

**Expected behavior:** [What you expect to happen]

The bed level will be fully faded and the z compensation will cease. The z height will increment by the set layer height.

**Actual behavior:** [What actually happens]

The bed level fades and z compensation ceases, but there is an additional z movement with each layer increment.

#### Additional Information

[image]

ellensp, 2020-05-28:
"CONFIGURATION_H_VERSION 010109" 
marlin 1.1.9 is no longer under development

Please try this feature in current bugfix 2.0.x to ascertain if problem still exists in currently maintained marlin.

sjasonsmith, 2020-06-20:
@snekonaplane, are you able to test with Marlin 2.x as @ellensp suggested? You realistically are not likely to ever see a fix for something like this in the 1.x codebase.

shitcreek, 2020-06-22:
This appears like a major issue which should have been caught and fixed by now if it's an actual bug.
That said, Marlin 1.x is no longer supported. Please use 2.x instead.

closed 2020-06-22 by shitcreek