MarlinFirmware/Marlin #18142: [BUG] SINGLENOZZLE M106 fan control without SINGLENOZZLE_STANDBY_FAN

Brassn, 2020-05-28:
### Bug Description

When having SINGLENOZZLE enabled but leaving SINGLENOZZLE_STANDBY_FAN disabled the command M106 S255 does no longer change the cooling fan speed if another tool than T0 is active.  
M106 defaults to the active tool if the P parameter is omitted, however in this scenario fan 0 must be used.

Previously this was fixed by this line 
https://github.com/MarlinFirmware/Marlin/blob/e4d8336175ee09b28d23220f446d07451ca3de43/Marlin/src/module/temperature.cpp#L200
but since the (otherwise much welcomed) changes from commit 89b17b54637200838e52ae39a9e4b23f3c2f8d35 it is no longer active in this scenario.

### suggested fix

These changes to temperature.cpp should do the trick:

```diff
    #if ENABLED(SINGLENOZZLE_STANDBY_FAN)
      if (target != active_extruder) {
        if (target < EXTRUDERS) singlenozzle_fan_speed[target] = speed;
        return;
      }
-      target = 0; // Always use fan index 0 with SINGLENOZZLE
    #endif
+
+    #if ENABLED(SINGLENOZZLE)
+      target = 0; // Always use fan index 0 with SINGLENOZZLE
+    #endif
```

boelle, 2020-06-21:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Brassn, 2020-06-21:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/4809305/Configuration.zip)
Sorry for deviating from the template

boelle, 2020-06-29:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Brassn, 2020-06-30:
I updated to bugfix-2.0.x as of state 34b84d3e63ae313261e736389c60f8883701a70e but unfortunately the issue remains.

Here is how I tested:
First I made sure my fix is no more in place

```
C:\Users\myself\Repositorys\Marlin>git diff --name-only upstream/bugfix-2.0.x..bugfix-2.0.x
Marlin/Configuration.h
Marlin/Configuration_adv.h
Marlin/src/pins/ramps/pins_MKS_GEN_13.h
```

Only minor and unrelated changes to the pin layout were performed aside the configuration. But you may also have a look at them [in my fork](https://github.com/Brassn/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/ramps/pins_MKS_GEN_13.h) if you prefer to.

Next connecting to Marlin via Pronterface:

```
Marlin bugfix-2.0.x
echo: Last Updated: 2020-06-30 | Author: (Brassn)
echo:Compiled: Jun 30 2020
echo: Free Memory: 1593  PlannerBufferBytes: 1216
echo:V81 stored settings retrieved (1058 bytes; crc 58438)
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
echo:SD card ok
Printer is now online.
>>> M106 S255
SENDING:M106 S255
M106 P0 S255
```
At this point the fan begins to spin, now switching to tool 1 to repeat the process.
```
>>> M106 S0
SENDING:M106 S0
M106 P0 S0
>>> T1
SENDING:T1
echo:Active Extruder: 1
>>> M106 S255
SENDING:M106 S255
```
Nothing happens at this point, neither a response nor the fan spinning.

After reapplying my changes as stated in the initial post [(or alternatively here)](https://github.com/Brassn/Marlin/commit/c5b36b9a43b027cc4345952e574a4889e860866f) the response is as follows:

```
>>> T1
SENDING:T1
echo:Active Extruder: 1
>>> M106 S255
SENDING:M106 S255
M106 P0 S255
>>> M106 S0
SENDING:M106 S0
M106 P0 S0
```

The fan begins and stops to spin accordingly.

thinkyhead, 2020-07-20:
Probably a simple bug. If it's obvious I'll patch it very soon. If I cannot find the bug then I may ask you for some additional testing and feedback.

firepower9966, 2020-08-06:
I also seeing this, no fan when printing single nozzle with dual extruders. even though my Gcode contains M106 S255 at layer 1
my configuration.h single nozzle is 
// For Cyclops or any "multi-extruder" that shares a single nozzle.
#define SINGLENOZZLE

// Save and restore temperature and fan speed on tool-change.
// Set standby for the unselected tool with M104/106/109 T...
#if ENABLED(SINGLENOZZLE)
  #define SINGLENOZZLE_STANDBY_TEMP
  #define SINGLENOZZLE_STANDBY_FAN
#endif

thordarsen, 2020-08-26:
I've had this same problem since I updated to after the mentioned PR
From My Octoprint terminal
`Send: T0`
`Recv: echo:Active Extruder: 0`
`Recv: ok`
`Send: M106 S255`
`Recv: M106 P0 S255`
`Recv: ok`
Fan turns on
Switch tool
`Send: T1`
`Recv: echo:Active Extruder: 1`
`Recv: ok`
Cut Fan Speed to 50%;
`Send: M106 S127`
`Recv: M106 P1 S127`
`Recv: ok`
Fan does not change, but in the lcd menu 'Fan Speed 2 is at 50%'
Note that the echo from Marlin specifies fan P1, which should not happen in a single nozzle setup, the expected behavior (I think) is that there is only one Part cooling fan in a single nozzle setup

Switch bact to Tool 0
`Send: T0`
`Recv: echo:Active Extruder: 0`
`Recv: ok`
`Send: M106 S127`
`Recv: M106 P0 S127`
`Recv: ok`
Now the fan is at 50%

Looking at PR [#17712] (url) https://github.com/MarlinFirmware/Marlin/commit/89b17b54637200838e52ae39a9e4b23f3c2f8d35
It seems like everywhere in the functional code where `#if ENABLED(SINGLENOZZLE)` existed, it got replaced by `#if ENABLED(SINGLENOZZLE_STANDBY_TEMP)` or `#if ENABLED(SINGLENOZZLE_STANDBY_FAN)`

except for this block (repeated in a few places)
`#if ENABLED(SINGLENOZZLE)`
`  #include "../../module/tool_change.h"`
`#endif`

I think I know what the I know what the SINGLENOZZLE part of the the PR was doing but it seems to have been overaggressive in the replacements of the flags. Notably this section of Temperature.cpp which only sets the target fan to 0 if the SINGLENOZZLE_STANDBY_FAN flag is set instead of anytime the SINGLENOZZLE flag is set.

`    #if ENABLED(SINGLENOZZLE_STANDBY_FAN)`
`      if (target != active_extruder) {`
`        if (target < EXTRUDERS) singlenozzle_fan_speed[target] = speed;`
`        return;`
`      }`
`      target = 0; // Always use fan index 0 with SINGLENOZZLE`
`    #endif`

Config
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5127138/Configuration.zip)

thordarsen, 2020-08-26:
I performed a quick test changing temperature.cpp to: (change in bold)
`    #if ENABLED(SINGLENOZZLE_STANDBY_FAN)`
`      if (target != active_extruder) {`
`        if (target < EXTRUDERS) singlenozzle_fan_speed[target] = speed;`
`        return;`
`      }`
**`    #endif`
`    #if ENABLED(SINGLENOZZLE)`**
`      target = 0; // Always use fan index 0 with SINGLENOZZLE`
`    #endif`

This worked as expected. 

Now to figure out a PR.

boelle, 2020-09-15:
this one seems fixed with the pr

closing

we can reopen if not fixed

closed 2020-09-15 by boelle